openbmc/openbmc #1242: ipmid: time related errors at host boot time

shenki, 2017-03-03:
I booted my plametto to petitboot and noticed this in the journal:

`v1.99.3-32-g05b7eba` (05b7eba)

```
Mar 03 00:12:02 palmetto ipmid[782]: ERROR:[0xFF] handling NetFn:[0xA], Cmd:[0x49]
Mar 03 00:12:02 palmetto dbus[678]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.52" (uid=0 pid=801 comm="/usr/sbin/timemanager ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.Failed" requested_reply="0" destination=":1.47" (uid=0 pid=782 comm="/usr/sbin/ipmid ")
Mar 03 00:12:18 palmetto systemd[1]: Starting Cleanup of Temporary Directories...
Mar 03 00:12:18 palmetto systemd[1]: Started Cleanup of Temporary Directories.
Mar 03 00:23:03 palmetto timemanager[801]: Request to SET Host time
Mar 03 00:23:03 palmetto timemanager[801]: Curr_Mode: NTPCurr_Owner: BMChost_offset: 0
Mar 03 00:23:03 palmetto dbus[678]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.52" (uid=0 pid=801 comm="/usr/sbin/timemanager ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.Failed" requested_reply="0" destination=":1.47" (uid=0 pid=782 comm="/usr/sbin/ipmid ")
Mar 03 00:23:03 palmetto ipmid[782]: ERROR:[0xFF] handling NetFn:[0xA], Cmd:[0x49]
Mar 03 00:34:04 palmetto timemanager[801]: Request to SET Host time
Mar 03 00:34:04 palmetto timemanager[801]: Curr_Mode: NTPCurr_Owner: BMChost_offset: 0
Mar 03 00:34:04 palmetto dbus[678]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.52" (uid=0 pid=801 comm="/usr/sbin/timemanager ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.Failed" requested_reply="0" destination=":1.47" (uid=0 pid=782 comm="/usr/sbin/ipmid ")
Mar 03 00:34:04 palmetto ipmid[782]: ERROR:[0xFF] handling NetFn:[0xA], Cmd:[0x49]
Mar 03 00:45:07 palmetto timemanager[801]: Request to SET Host time
Mar 03 00:45:07 palmetto timemanager[801]: Curr_Mode: NTPCurr_Owner: BMChost_offset: 0
Mar 03 00:45:07 palmetto dbus[678]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.52" (uid=0 pid=801 comm="/usr/sbin/timemanager ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.Failed" requested_reply="0" destination=":1.47" (uid=0 pid=782 comm="/usr/sbin/ipmid ")
Mar 03 00:45:07 palmetto ipmid[782]: ERROR:[0xFF] handling NetFn:[0xA], Cmd:[0x49]
Mar 03 00:56:10 palmetto timemanager[801]: Request to SET Host time
Mar 03 00:56:10 palmetto timemanager[801]: Curr_Mode: NTPCurr_Owner: BMChost_offset: 0
Mar 03 00:56:10 palmetto dbus[678]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.52" (uid=0 pid=801 comm="/usr/sbin/timemanager ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.Failed" requested_reply="0" destination=":1.47" (uid=0 pid=782 comm="/usr/sbin/ipmid ")
Mar 03 00:56:10 palmetto ipmid[782]: ERROR:[0xFF] handling NetFn:[0xA], Cmd:[0x49]
```

ratagupt, 2017-03-07:
What is the defect here?Seems on plamettto  current mode is set to NTP and time manager is throwing the error,that time can not be set which is WAD.
@tomjoseph83 :- Can you comment on this?

tomjoseph83, 2017-03-16:
This is not a bug, if the current owner is NTP, then Set SEL Time is not allowed. Please close this.

geissonator, 2017-03-20:
Do we need a bug on the host for trying to set the SEL time when we're in NTP mode?

tomjoseph83, 2017-05-05:
@geissonator @vishwabmc Do we need to open a issue with hostboot for this?

mdmillerii, 2017-05-09:
@tomjoseph83 how would hostboot know the platform policy made the clock read-only?   Seems to me like we should just make the result a 1-line debug ignore.

tomjoseph83, 2017-05-11:
@mdmillerii Agree with you that the host needs to be agnostic about the BMC time manager policy and just ignore the Set SEL time request.

tomjoseph83, 2017-07-04:
@mine260309 would be taking care of this in the timemanager re-factoring, to avoid flooding the journal with Set SEL time request not getting honoured because of BMC time manager policy.

geissonator, 2017-10-17:
Lei, ready to close out?

mine260309, 2017-10-20:
Yeah, I think this can be closed.

closed 2017-11-03 by rfrandse