openbmc/openbmc #1243: ipmid: mac address errors

shenki, 2017-03-03:
I saw this when booting the host on my palmetto:

`v1.99.3-32-g05b7eba` (05b7eba)

```
Mar 03 00:11:17 palmetto ipmid[782]: sscanf Failed in extracting mac address.
Mar 03 00:11:17 palmetto ipmid[782]: getHostNetworkData failed for get_sys_boot_options.
Mar 03 00:11:17 palmetto ipmid[782]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Mar 03 00:11:17 palmetto ipmid[782]: sscanf Failed in extracting mac address.
Mar 03 00:11:17 palmetto ipmid[782]: getHostNetworkData failed for get_sys_boot_options.
Mar 03 00:11:17 palmetto ipmid[782]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Mar 03 00:11:17 palmetto ipmid[782]: sscanf Failed in extracting mac address.
Mar 03 00:11:17 palmetto ipmid[782]: getHostNetworkData failed for get_sys_boot_options.
Mar 03 00:11:17 palmetto ipmid[782]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Mar 03 00:11:17 palmetto ipmid[782]: sscanf Failed in extracting mac address.
Mar 03 00:11:17 palmetto ipmid[782]: getHostNetworkData failed for get_sys_boot_options.
Mar 03 00:11:17 palmetto ipmid[782]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Mar 03 00:11:17 palmetto ipmid[782]: No registered handlers for NetFn:[0x3A], Cmd:[0x1A] trying Wilcard implementation
Mar 03 00:11:17 palmetto ipmid[782]: No Registered handlers for NetFn:[0x3A],Cmd:[0xFF]
Mar 03 00:11:17 palmetto ipmid[782]: ERROR:[0xC1] handling NetFn:[0x3A], Cmd:[0x1A]
```

tomjoseph83, 2017-03-16:
This error "sscanf Failed in extracting mac address." is thrown when the MAC address is not in the expected format. Could you mention the palmetto where this was tried?

geissonator, 2017-03-20:
So you need a fw_printenv from the palmetto showing this issue Tom?

tomjoseph83, 2017-05-05:
@geissonator Can we get this closed, since there is no debug to do further.

dkodihal, 2017-07-07:
Ratan is going to fix this as part of some the network manager refactor.

dkodihal, 2017-07-07:
Same as #556 as per @ratagupt.

closed 2017-07-07 by dkodihal