openbmc/openbmc #1244: [Barreleye VERSION_ID=""v1.99.3-33"] State manager services are not running with latest openbmc build image

SrideviRamesh, 2017-03-03:
Issue:

State manager services on barreleye are not running with latest openbmc build image 

BMC Log Info:
```
 #ps -eaf | grep state
  767 root       0:00 /usr/sbin/netman_watch_dns /run/systemd/netif/state
 1328 root       0:00 grep state

# obmcutil state
Traceback (most recent call last):
  File "/usr/sbin/obmcutil", line 196, in <module>
    obj = bus.get_object(bus_name, obj_path)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_nam                                                                                                                     e_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_servic                                                                                                                     e_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_                                                                                                                     blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The na                                                                                                                     me org.openbmc.managers.System was not provided by any .service files

```

system Info:

cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-33"
VERSION_ID="v1.99.3-33-gdb86d27"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-33"
BUILD_ID="v1.99.3"

please find attached FFDC, serial port console, journal, dmesg logs

[20170303014758130221_Myffdc.zip](https://github.com/openbmc/openbmc/files/816109/20170303014758130221_Myffdc.zip)
[barreleye_dmesg_logs.txt](https://github.com/openbmc/openbmc/files/816111/barreleye_dmesg_logs.txt)
[barreleye_serial_port_logs.txt](https://github.com/openbmc/openbmc/files/816112/barreleye_serial_port_logs.txt)
[barreleye_journal_logs.txt](https://github.com/openbmc/openbmc/files/816115/barreleye_journal_logs.txt)

gkeishin, 2017-03-03:
```
root@witherspoon:~# systemctl status org.openbmc.managers.System.service | cat
● org.openbmc.managers.System.service - Phoshpor System Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.managers.System.service; enabled; vendor preset: enabled)
   Active: activating (start) since Wed 2017-02-22 06:57:40 UTC; 3s ago
 Main PID: 3203 (python)
   CGroup: /system.slice/org.openbmc.managers.System.service
           └─3203 python /usr/sbin/system_manager.py

Feb 22 06:57:40 witherspoon systemd[1]: org.openbmc.managers.System.service: Service hold-off time over, scheduling restart.
Feb 22 06:57:40 witherspoon systemd[1]: Stopped Phoshpor System Manager.
Feb 22 06:57:40 witherspoon systemd[1]: Starting Phoshpor System Manager...
root@witherspoon:~#

root@witherspoon:~# systemctl status org.openbmc.Sensors.service | cat
● org.openbmc.Sensors.service - Phosphor Sensor Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.Sensors.service; enabled; vendor preset: enabled)
   Active: activating (start) since Wed 2017-02-22 06:57:05 UTC; 3s ago
 Main PID: 3175 (python)
   CGroup: /system.slice/org.openbmc.Sensors.service
           └─3175 python /usr/sbin/sensor_manager2.py

Feb 22 06:57:05 witherspoon systemd[1]: org.openbmc.Sensors.service: Service hold-off time over, scheduling restart.
Feb 22 06:57:05 witherspoon systemd[1]: Stopped Phosphor Sensor Manager.
Feb 22 06:57:05 witherspoon systemd[1]: Starting Phosphor Sensor Manager...
root@witherspoon:~#
```

gkeishin, 2017-03-03:
This is for witherspoon and barreleye.. so its pervasive and broken..

geissonator, 2017-03-03:
```
Feb 01 02:36:39 barreleye system_manager.py[747]: Traceback (most recent call last):
Feb 01 02:36:39 barreleye system_manager.py[747]:   File "/usr/sbin/system_manager.py", line 10, in <module>
Feb 01 02:36:39 barreleye system_manager.py[747]:     import obmc.enums
Feb 01 02:36:39 barreleye system_manager.py[747]: ImportError: No module named enums
...
Feb 01 02:36:48 barreleye sensor_manager2.py[789]: Traceback (most recent call last):
Feb 01 02:36:48 barreleye sensor_manager2.py[789]:   File "/usr/sbin/sensor_manager2.py", line 7, in <module>
Feb 01 02:36:48 barreleye sensor_manager2.py[789]:     import obmc.sensors
Feb 01 02:36:48 barreleye sensor_manager2.py[789]: ImportError: No module named sensors
```

geissonator, 2017-03-03:
Only change between working and not working was https://github.com/openbmc/openbmc/commit/db86d27a593f932cd2d4d77307801b3a8e00874f

geissonator, 2017-03-06:
Fixed here - https://gerrit.openbmc-project.xyz/#/c/2833/

closed 2017-03-06 by geissonator