openbmc/openbmc #1245: Memmory corruption [witherspoon]

gkeishin, 2017-03-03:
On BMC reset, we are seeing this errors.  Had a discussion with @amboar and it happened in generic kernel code not anything specific to the BMC.  We have seen this error twice in BMC reset but not frequently.

- [x]  Serial console log not sure if that would help 
[Memory_corruption.txt](https://github.com/openbmc/openbmc/files/816379/Memory_corruption.txt)

```
------------[ cut here ]------------
WARNING: CPU: 0 PID: 1121 at /var/lib/jenkins-slave/workspace/openbmc-build/distro/ubuntu/target/witherspoon/openbmc/build/tmp/work-shared/witherspoon/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
sysfs: cannot create duplicate filename '/devices/hub@00'
CPU: 0 PID: 1121 Comm: sh Not tainted 4.7.10-39eb5a44bfd49fe530f6b68c4d99ab2d8d391be0 #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<80110cbc>] (__warn+0xc4/0xf0)
[<80110cbc>] (__warn) from [<80110d2c>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d2c>] (warn_slowpath_fmt) from [<801f88dc>] (sysfs_warn_dup+0x50/0x70)
[<801f88dc>] (sysfs_warn_dup) from [<801f8970>] (sysfs_create_dir_ns+0x74/0x88)
[<801f8970>] (sysfs_create_dir_ns) from [<802a2940>] (kobject_add_internal+0x108/0x30c)
[<802a2940>] (kobject_add_internal) from [<802a2bc0>] (kobject_add+0x7c/0xb4)
[<802a2bc0>] (kobject_add) from [<802e5fec>] (device_add+0x118/0x53c)
[<802e5fec>] (device_add) from [<8034fd6c>] (fsi_master_register+0x434/0x8ac)
[<8034fd6c>] (fsi_master_register) from [<8035010c>] (fsi_master_register+0x7d4/0x8ac)
[<8035010c>] (fsi_master_register) from [<80350684>] (store_scan+0x7c/0x90)
[<80350684>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
[<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
[<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
[<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
[<801a36d4>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
---[ end trace 56188a36510b62b7 ]---
------------[ cut here ]------------
WARNING: CPU: 0 PID: 1121 at /var/lib/jenkins-slave/workspace/openbmc-build/distro/ubuntu/target/witherspoon/openbmc/build/tmp/work-shared/witherspoon/kernel-source/lib/kobject.c:240 kobject_add_internal+0x290/0x30c
kobject_add_internal failed for hub@00 with -EEXIST, don't try to register things with the same name in the same directory.
CPU: 0 PID: 1121 Comm: sh Tainted: G        W       4.7.10-39eb5a44bfd49fe530f6b68c4d99ab2d8d391be0 #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<80110cbc>] (__warn+0xc4/0xf0)
[<80110cbc>] (__warn) from [<80110d2c>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d2c>] (warn_slowpath_fmt) from [<802a2ac8>] (kobject_add_internal+0x290/0x30c)
[<802a2ac8>] (kobject_add_internal) from [<802a2bc0>] (kobject_add+0x7c/0xb4)
[<802a2bc0>] (kobject_add) from [<802e5fec>] (device_add+0x118/0x53c)
[<802e5fec>] (device_add) from [<8034fd6c>] (fsi_master_register+0x434/0x8ac)
[<8034fd6c>] (fsi_master_register) from [<8035010c>] (fsi_master_register+0x7d4/0x8ac)
[<8035010c>] (fsi_master_register) from [<80350684>] (store_scan+0x7c/0x90)
[<80350684>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
[<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
[<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
[<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
[<801a36d4>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
---[ end trace 56188a36510b62b8 ]---
```

gkeishin, 2017-03-03:
As per @shenki this is a FSI bug.  He'll take a dig into it on the error.

shenki, 2017-03-03:
I would expect this to be resolved by the changes pulled in by https://gerrit.openbmc-project.xyz/#/c/2824/

amboar, 2017-03-03:
I feel like this wasn't the stack trace I was shown on slack, this is clearly fsi.

gkeishin, 2017-03-31:
Can we close this now @eddiejames @cbostic

closed 2017-09-18 by gkeishin