openbmc/openbmc #1246: Status field is not present for all REST operation which fails [Barreleye]

rahulmah, 2017-03-03:
Status field is not present for all REST operation which fails. This issue is seen with all REST command where we expect error.

It is seen in recent builds.

Example: 
Here is the REST API request with invalid url. We can see that no status field is present in response. 
```bash-4.1$ curl -c cjar -b cjar -k https://9.3.23.30/xyz/openbmc_proj
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_proj"
  },
  "message": "404 Not Found"              <---------- status field is not present
}bash-4.1$```

If REST API works fine. we can see status field.
```
bash-4.1$ curl -c cjar -b cjar -k https://9.3.23.30/xyz/openbmc_project/
{
  "data": [
    "/xyz/openbmc_project/sensors",
    "/xyz/openbmc_project/inventory",
    "/xyz/openbmc_project/ObjectMapper",
    "/xyz/openbmc_project/software",
    "/xyz/openbmc_project/logging"
  ],
  "message": "200 OK",
  "status": "ok"                        <---------- status field is present
}```

BMC fw : 
```root@barreleye:~# cat /etc/version
20170302221532
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-32"
VERSION_ID="v1.99.3-32-g05b7eba"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-32"
BUILD_ID="v1.99.3"
root@barreleye:~#```

rahulmah, 2017-03-06:
@ratagupt @geissonator : Can you please take this forward and get it fixed? Due to this issue we are seeing many failure (close to 40 TCs) in CT.

ratagupt, 2017-03-06:
Seems the problem is with the negative test cases.need to check why status has been removed from the curl response,feel that it(status) should be set to ''error" in case of any error.
Anyways but message is giving the http status code.

closed 2017-03-07 by williamspatrick