openbmc/openbmc #1247: Barreleye unexpected host reboot after 11 hours of uptime

astallma, 2017-03-03:
System Type:  Barreleye

BMC version  (multiple, here's from z1407abmc):
   s# cat /etc/os-release
   ID=openbmc-phosphor
   NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
   VERSION=0.1.0 (jethro)
   VERSION_ID=v1.0.5-5-g6044148-dirty
   PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
   BUILD_ID=v1.0.5

Kernel level (z1407abmc):
   s# uname -a
   Linux barreleye 4.4.26-openbmc-4.4-20161021-1 #1 Fri Oct 21 10:32:53 UTC 2016 armv5tejl GNU/Linux

Scenario:

Host unexpectedly reboots after 11 hours of uptime.  IPMI tracing turned on from OpenBMC and there is no indication of a Host request for a reboot (see below).  No new SEL entries generated.   Minimal logging in the journal (see below).   Can recreate as needed.

Expected Results:

Host stays running!   I can't deny that this may be a host issue, but we need some help debugging here (various host subsystems have been actively debugging).    There should at least be some indication on the BMC of why the Host went down.   Was this reboot requested by the Host?  Was it associated with an error?  (It would be great of the journal could point to the SEL associated with the reboot).    In addition to debugging this scenario, we may need this information for future support.

Details:

Journal snippet:
                                                          
```
....
Feb 28 14:30:19 barreleye systemd[1]: host-ipmid.service: Main process exited, code=killed, status=9/KILL         
Feb 28 14:30:19 barreleye systemd[1]: host-ipmid.service: Unit entered failed state.
Feb 28 14:30:19 barreleye systemd[1]: host-ipmid.service: Failed with result 'signal'.
Feb 28 14:30:20 barreleye systemd[1]: host-ipmid.service: Service hold-off time over, scheduling restart. 
Feb 28 14:30:20 barreleye systemd[1]: systemd-hwdb-update.service: Cannot add dependency job, ignoring: Unit systemd-hwdb-update.service is masked.
Feb 28 14:30:20 barreleye systemd[1]: Stopped Phosphor OpenBMC IPMI daemon.                                         
Feb 28 14:30:20 barreleye systemd[1]: Started Phosphor OpenBMC IPMI daemon.           <<== Turning on IPMI tracing                                  
Feb 28 16:51:31 barreleye system_manager.py[966]: Rebooting                           <<== Reboot a couple of hours later!                           
Feb 28 16:51:31 barreleye system_manager.py[966]: Turn off power                                                    
Feb 28 16:51:31 barreleye system_manager.py[966]: Stopping watchdog                                                 
Feb 28 16:51:31 barreleye system_manager.py[966]: Running System State: HOST_POWERING_OFF                           
Feb 28 16:51:34 barreleye system_manager.py[966]: Running System State: HOST_POWERED_OFF                            
Feb 28 16:51:34 barreleye system_manager.py[966]: Deleting OCC device                                               
Feb 28 16:51:34 barreleye system_manager.py[966]: Turn on power and boot                                            
Feb 28 16:51:34 barreleye system_manager.py[966]: Running System State: HOST_POWERING_ON                            
Feb 28 16:51:34 barreleye system_manager.py[966]: Deleting OCC device                                               
Feb 28 16:51:34 barreleye system_manager.py[966]: Cannot read attributes: /sys/class/hwmon/hwmon7/temp2_input       
Feb 28 16:51:34 barreleye system_manager.py[966]: Cannot read attributes: /sys/class/hwmon/hwmon7/temp18_input  
....    
```

IPMI tracing from Console snippet:

```
root@barreleye:~# ps | grep ipmid                     <<== IPMI tracing turned on
 2624 root      4952 S    /usr/sbin/ipmid -d 0xff
 2627 root      4952 S    /usr/sbin/ipmid
 2629 root      2972 S    grep ipmid
root@barreleye:~# kill -9 2627
root@barreleye:/tmp# journalctl --no-pager > /tmp/log_before0228_reboot
root@barreleye:/tmp# 
IPMI Incoming: Seq 0x00, NetFn 0x06, CMD: 0x36          <<== Exchange caps as part of boot processing!
Handling Netfn:[0x6], Cmd:[0x36]
IPMI Response:
0x000000: 00 01 40 40 0a 01                               ..@@..          
IPMI Incoming: Seq 0x01, NetFn 0x06, CMD: 0x06 
0x000000: a0 7f                                           ..              
IPMI SET ACPI STATE Ignoring for now
IPMI Response:
0x000000: 00                                              .               
IPMI Incoming: Seq 0x02, NetFn 0x06, CMD: 0x24 
0x000000: 41 01 00 02 b0 04                               A.....          
WATCHDOG SET Timer:[0x4B0] 100ms intervals
IPMI Response:
0x000000: 00                                              .               
IPMI Incoming: Seq 0x03, NetFn 0x06, CMD: 0x22 
WATCHDOG RESET
IPMI Response:
0x000000: 00         
....                                     .      

```

geissonator, 2017-03-03:
I see this is the old stable release.  Nothing indicating a checkstop (that could cause an auto reboot).  Nothing from the host over IPMI.  Kind of looks like someone just called the dbus interface into skeleton/pychassisctl/chassis_control.py to do a reboot.  @anoo1 or @causten any thoughts on this?

anoo1, 2017-03-03:
I took a look at the skeleton code and doesn't seem reboot is called internally, so perhaps the host is calling to reboot? Could we get the LTC team to comment if this could be driven from the host due to some setting?

geissonator, 2017-03-06:
Running this in an ssh session will generate a ton of output but would tell us where the reboot is coming from:
dbus-monitor --system

astallma, 2017-03-07:
I've attached a file containing output from BMC ssh session after running "dbus-monitor --system".   The reboot occurs at the end and the trace includes the first few IPMI commands from early Hostboot steps.    I have data going back even further than what is included in this file, but it didn't look relevant, let me know if I should include additional trace data.

Here is where things seem to start getting interesting:

```
[z1407abmc_11hourreboot_bmctrace.txt](https://github.com/openbmc/openbmc/files/824677/z1407abmc_11hourreboot_bmctrace.txt)

signal sender=:1.11 -> dest=(null destination) serial=14353408 path=/org/openbmc/sensors/HSCC/Vout; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.openbmc.SensorValue"
   array [
      dict entry(
         string "value"
         variant             int32 12026
      )
   ]
   array [
   ]
signal sender=:1.32 -> dest=(null destination) serial=951 path=/org/openbmc/control/chassis0; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.openbmc.control.Chassis"
   array [
      dict entry(
         string "reboot"
         variant             int32 1
      )
   ]
   array [
   ]
signal sender=:1.18 -> dest=(null destination) serial=751 path=/org/openbmc/control/power0; interface=org.openbmc.Control; member=GotoSystemState
   string "HOST_POWERING_OFF"
signal sender=:1.8 -> dest=(null destination) serial=26860 path=/org/openbmc/managers/System; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.openbmc.managers.System"
   array [
      dict entry(
         string "current_state"
         variant             string "HOST_POWERING_OFF"
      )
   ]
   array [
   ]
```

Again, we are looking for help understanding what triggered this host reboot.

JoshDKing, 2017-03-21:
Spoke with @astallma and he said they were able to solve the issue.

astallma, 2017-03-22:
We resolved the root cause of the issue, but the host FW team believes the BMC logging could be improved for host reboot scenarios.   Having the cause of the reboot logged could help to debug scenarios where the host may not capture the necessary information.     

For this particular scenario, host firmware caused a checkstop which flips a gpio bit that the BMC monitors.  The BMC then specifically triggers the reboot.   This information would have assisted debug in this case and potentially future scenarios.    Other scenarios such as an host IPMI reboot request or a host reboot initiated by the BMC (in response to a user request) could be differentiated in the log.

(If this information already exists, please let me know how to find it... the information I was able to find in the journal was very minimal).

williamspatrick, 2017-03-22:
@astallma - we already have it improved in some cases in the 1.99.x code and have it planned for other errors. Nobody is planned to touch the 1.x release though. You're welcome to make a pull request. ;)

closed 2017-03-22 by williamspatrick

astallma, 2017-03-22:
Sounds great, looking forward to all of the new features in 1.99.x