openbmc/openbmc #1250: Kernel Segmentation fault during boot preparing for code update

micwalsh, 2017-03-03:
cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-2"
VERSION_ID="v1.99.3-2-g4f50ea6-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-2"
BUILD_ID="v1.99.3"

cat /etc/version     
20170228140204

[w15.170303.105423.bmc_console.txt](https://github.com/openbmc/openbmc/files/817514/w15.170303.105423.bmc_console.txt)

My log data:
https://tbaix2.rchland.ibm.com/afs/rchland.ibm.com/rel/common/non-share/fspusers/WWW/sysmgr/tailFile.php?file=/gsa/ausgsa/projects/a/autoipl/status/w15.170303.105404.status

geissonator, 2017-03-06:
Any ideas on this one @shenki or @amboar ?

shenki, 2017-03-06:
Can't tell from the information provided. The crash is due to a null pointer in the shutdown path.

Given we have not seen this before I would point the finger at recent driver work such as FSI.

amboar, 2017-03-08:
@shenki yeah, I think @gkeishin has filed an issue somewhere with FSI stack traces in dmesg prior to this one appearing on shutdown. Not a smoking gun, but suspicious.

gkeishin, 2017-03-08:
Today in latest build in BMC reset test
```
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/611/ 
- [x] VERSION_ID : v1.99.3-45-g8e6b232
```
```
[  OK  ] Reached target Final Step.
         Starting Reboot...
watchdog: watchdog0: watchdog did not stop!
systemd-shutdown[1]: Sending SIGTERM to remaining processes...
systemd-journald[448]: Received SIGTERM from PID 1 (systemd-shutdow).
systemd-shutdown[1]: Sending SIGKILL to remaining processes...
systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
systemd-shutdown[1]: Unmounting file systems.
systemd-shutdown[1]: Remounting '/' read-only with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work'.
systemd-shutdown[1]: Remounting '/' read-only with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work'.
systemd-shutdown[1]: All filesystems unmounted.
systemd-shutdown[1]: Deactivating swaps.
systemd-shutdown[1]: All swaps deactivated.
systemd-shutdown[1]: Detaching loop devices.
systemd-shutdown[1]: All loop devices detached.
systemd-shutdown[1]: Detaching DM devices.
systemd-shutdown[1]: All DM devices detached.
systemd-shutdown[1]: Successfully changed into root pivot.
systemd-shutdown[1]: Returning to initrd...
watchdog: watchdog0: watchdog did not stop!
shutdown: reboot --log-level 6 --log-target kmsg --log-color
+ sort -r
+ awk /oldroot|mnt/ { print $2 }
+ umount /oldroot/sys/kernel/debug
+ umount /oldroot/sys/fs/cgroup/systemd
+ umount /oldroot/sys/fs/cgroup
+ umount /oldroot/sys
+ umount /oldroot/proc
+ umount /oldroot/dev/shm
+ umount /oldroot/dev/pts
+ umount /oldroot/dev
+ umount /oldroot
+ umount /mnt
+ set +x
Remaining mounts:
tmpfs / tmpfs rw,nosuid,nodev,mode=755 0 0
dev /dev devtmpfs rw,relatime,size=215184k,nr_inodes=53796,mode=755 0 0
proc /proc proc rw,relatime 0 0
sys /sys sysfs rw,relatime 0 0
tmpfs /run tmpfs rw,nosuid,nodev,mode=755 0 0
------------[ cut here ]------------
WARNING: CPU: 0 PID: 3131 at /var/lib/jenkins-slave/workspace/openbmc-build/distro/ubuntu/target/witherspoon/openbmc/build/tmp/work-shared/witherspoon/kernel-source/lib/kobject.c:597 kobject_get+0x30/0x98
Unable to handle kernel paging request at virtual address 00008d74
pgd = 9da5c000
[00008d74] *pgd=9d014831, *pte=00000000, *ppte=00000000
Internal error: Oops: 17 [#1] ARM
CPU: 0 PID: 3131 Comm: reboot Tainted: G        W       4.7.10-2763f3eb775cf760b1c0f28d283da6df6c42f061 #1
Hardware name: ASpeed SoC
task: 9d592000 ti: 9dafe000 task.ti: 9dafe000
PC is at string+0x2c/0x74
LR is at 0xffffffff
pc : [<802a7028>]    lr : [<ffffffff>]    psr: 00000093
sp : 9daffd28  ip : 8072cef4  fp : 000003e0
r10: 9daffd54  r9 : 805b1a7f  r8 : 8072cb14
r7 : 8072cef4  r6 : 9daffe40  r5 : 8072cb20  r4 : 9daffe3c
r3 : 805b24de  r2 : 00008d74  r1 : 00000000  r0 : 8072cb20
Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 00c5387d  Table: 9da5c008  DAC: 00000051
Process reboot (pid: 3131, stack limit = 0x9dafe188)
Stack: (0x9daffd28 to 0x9db00000)
fd20:                   8072cb20 8072cef4 00008d74 802a8e80 8072cb20 ffffff04
fd40: ffff0a00 00002bf4 8070200c 0000000c 000000cc ffffff04 ffff0a00 1c0dbeb1
fd60: 807086e0 000003e0 00000000 ffffffff 00000009 802a2390 807086e0 00000000
fd80: 00000000 802a9558 00000000 8013401c 60000013 801014a4 8016e700 8070200c
[... 23 lines cut ...]
[<801343d0>] (vprintk_default) from [<80110c84>] (__warn+0x8c/0xf0)
[<80110c84>] (__warn) from [<80110d2c>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d2c>] (warn_slowpath_fmt) from [<802a2390>] (kobject_get+0x30/0x98)
[<802a2390>] (kobject_get) from [<802e4b6c>] (get_device+0x14/0x1c)
[<802e4b6c>] (get_device) from [<802e6728>] (device_shutdown+0x2c/0x120)
[<802e6728>] (device_shutdown) from [<80128470>] (kernel_restart+0xc/0x50)
[<80128470>] (kernel_restart) from [<80128720>] (SyS_reboot+0x124/0x1f4)
[<80128720>] (SyS_reboot) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
Code: e3a01000 31a02003 e151000e 0a000002 (e4d23001)
---[ end trace 762486e67f46ef5c ]---
Segmentation fault
Execute reboot -f if all unmounted ok, or exec /init
shutdown-sh#

U-Boot 2016.07 (Feb 22 2017 - 17:20:20 +0000)

DRAM:  496 MiB
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Hit any key to stop autoboot:  0
## Loading kernel from FIT Image at 20080000 ...
```

shenki, 2017-09-25:
Closing ,we have moved on from the 4.7 kernel.

closed 2017-09-25 by shenki