openbmc/openbmc #1251: avahi fails to start due to missing xml

shenki, 2017-03-06:
`v1.99.3-32-g05b7eba` (05b7eba) on palmetto

```
Mar 06 02:48:20 palmetto avahi-daemon[711]: ../../avahi-0.6.32/avahi-daemon/chroot.c: open() failed: No such file or directory
Mar 06 02:48:20 palmetto avahi-daemon[689]: Failed to open /usr/share/dbus-1/interfaces/org.freedesktop.Avahi.Server.xml: Invalid argument
Mar 06 02:48:20 palmetto avahi-daemon[689]: Failed to load introspection data.
```

geissonator, 2017-03-06:
Same signature as #985 @ratagupt

ratagupt, 2017-03-09:
In avahi-daemon/makefile.am there is introspectiondir variable name which is pointing to introspectiondir=$(datadir)/dbus-1/interfaces.

Earlier there was no dbus interfaces file instead of there was .introspect files.This change was introduced through this commit.
https://github.com/lathiat/avahi/commit/d77d1a036bc186878e804736f02cb38eef450e1d

Now in the recipie file in the do_install function.
the dbus interfaces folder is being deleted.
Seems we don't need to delete it.I made the changes over there and it worked fine for me.
```
do_install() {
	autotools_do_install

	# don't install /var/run when populating rootfs. Do it through volatile
	# /var/run of current version is empty, so just remove it.
	# if /var/run become non-empty in the future, need to install it via volatile
	rm -rf ${D}${localstatedir}/run
	rmdir --ignore-fail-on-non-empty ${D}${localstatedir}
	**rm -rf ${D}${datadir}/dbus-1/interfaces**
	test -d ${D}${datadir}/dbus-1 && rmdir --ignore-fail-on-non-empty ${D}${datadir}/dbus-1
	rm -rf ${D}${libdir}/avahi

	install -d ${D}${sysconfdir}/udhcpc.d
	install ${WORKDIR}/00avahi-autoipd ${D}${sysconfdir}/udhcpc.d
	install ${WORKDIR}/99avahi-autoipd ${D}${sysconfdir}/udhcpc.d
}
```

ratagupt, 2017-03-09:
Forgot to mention that these files are being referred by code at runtime.

closed 2019-10-27 by stale[bot]