openbmc/openbmc #1253: sensor_manager failure with new kernel

shenki, 2017-03-06:
I'm testing with a 4.10 based kernel and I get this at boot. Can someone who is familair with this code path give me some hints?

```
Mar 06 05:41:05 palmetto system_manager.py[726]: Running System State: HOST_BOOTED
Mar 06 05:41:05 palmetto sensor_manager2.py[696]: Installing OCC device
Mar 06 05:41:05 palmetto sensor_manager2.py[696]: sh: line 0: echo: write error: Invalid argument
Mar 06 05:41:05 palmetto kernel: occ-i2c 3-0050: hwmon_device_register() is deprecated. Please convert the driver to use hwmon_device_register_with_info().
Mar 06 05:41:05 palmetto kernel[698]: occ-i2c 3-0050: hwmon_device_register() is deprecated. Please convert the driver to use hwmon_device_register_with_info().
Mar 06 05:41:05 palmetto discover_system_state.py[806]: Traceback (most recent call last):
Mar 06 05:41:05 palmetto discover_system_state.py[806]:   File "/usr/sbin/discover_system_state.py", line 72, in <module>
Mar 06 05:41:05 palmetto discover_system_state.py[806]:     intf.setValue("Enabled")
Mar 06 05:41:05 palmetto discover_system_state.py[806]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
Mar 06 05:41:05 palmetto discover_system_state.py[806]:     **keywords)
Mar 06 05:41:05 palmetto discover_system_state.py[806]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Mar 06 05:41:05 palmetto discover_system_state.py[806]:     message, timeout)
Mar 06 05:41:05 palmetto discover_system_state.py[806]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method "setValue" with signature "s" on interface "org.openbmc.SensorValue" doesn't exist
Mar 06 05:41:05 palmetto systemd[1]: Started Phosphor Hwmon Poller.
Mar 06 05:41:05 palmetto systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40100-occ\x4050.service: Unit not needed anymore. Stopping.
Mar 06 05:41:05 palmetto systemd[1]: Stopping Phosphor Hwmon Poller...
Mar 06 05:41:05 palmetto systemd[1]: Stopped Phosphor Hwmon Poller.
```

shenki, 2017-03-06:
I was thrown off by sensor_manger2.py. The print is coming from [pyphosphor/obmc/sensors.py](https://github.com/openbmc/pyphosphor/blob/57e3bf965d731ac52f59497a6247a1d80d1bc0a0/obmc/sensors.py#L270):

```
    ##override setValue method
    @dbus.service.method(
        SensorValue.IFACE_NAME, in_signature='v', out_signature='')
    def setValue(self, value):
        if (value == "Enabled"):
            print "Installing OCC device"
            os.system("echo 1 > " + self.sysfs_attr)
        else:
            print "Deleting OCC device"
            os.system("echo 0 > " + self.sysfs_attr)
        SensorValue.setValue(self, value)
```

shenki, 2017-03-06:
This code looks like it will need attention when we move to @eddiejames's new OCC driver.

geissonator, 2017-03-06:
Is this 4.10 specific or just general lack of OCC support on Palmetto's?

shenki, 2017-03-06:
Plametto has OCC support. It might be 4.10 specific, but it's hard to tell from the logs.

This code needs some attention. Is it scheduled for rework?

eddiejames, 2017-03-07:
This is the old p8 occ driver. My new driver currently waiting on the lkml addresses this issue.

shenki, 2017-03-09:
This does not relate to the kernel driver at all. The userspace code needs rework.

Items that need fixing:

1. Replace uses of `os.system()` with `write()`
2. add logging for the file that is being written/read from so users can diagnose
3. Look into the discover_system_state traceback

closed 2019-04-27 by stale[bot]