openbmc/openbmc #1255: IPMID Error Setting BMC/HOST time [ Witherspoon ]

gkeishin, 2017-03-06:
```
root@witherspoon:~# cat /etc/os-release | grep VERSION_ID
VERSION_ID="v1.99.3-39-g2cac4f7"
root@witherspoon:~# cat /etc/version
20170306031742
```

This seems pervasive. I can see in Poweron's

```
Feb 22 06:46:09 witherspoon ipmid[919]: IPMI Handling GET-SEL-TIME
Feb 22 06:46:09 witherspoon ipmid[919]: Host Time read:[Wed Feb 22 06:34:32 2017 GMT] :: [1487745272]
Feb 22 06:46:09 witherspoon ipmid[919]: Handling Set-SEL-Time:[0xA], Cmd:[0x49]
Feb 22 06:46:09 witherspoon ipmid[919]: Data: 0x58BCEF71]
Feb 22 06:46:09 witherspoon ipmid[919]: Error [Permission denied] setting time
Feb 22 06:46:09 witherspoon ipmid[919]: IPMI Handling GET-SEL-TIME
Feb 22 06:57:45 witherspoon ipmid[919]: Host Time read:[Wed Feb 22 06:46:ERROR:[0xFF] handling NetFn:[0xA], Cmd:[0x49]
Feb 22 06:57:45 witherspoon dbus[763]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.66" (uid=0 pid=956 comm="/usr/sbin/timemanager ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.Failed" requested_reply="0" destination=":1.57" (uid=0 pid=919 comm="/usr/sbin/ipmid ")
```

- [x] Journal log  [journald_v1.99.3-39-g2cac4f7_20170306031742_Error_ipmid.txt](https://github.com/openbmc/openbmc/files/820453/journald_v1.99.3-39-g2cac4f7_20170306031742_Error_ipmid.txt)

gkeishin, 2017-03-06:
In our Time test as well see similar reason of failures
```
To JSON using : content={ "data": 
{ "description": "Internal Server Error", "exception": "DBusException('Current Mode is NTP',)", "traceback": [ "Traceback (most recent call last):", " File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle", " return route.call(**args)", " File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 550, in __call__", " return self.callback(*a, **kw)", " File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 607, in wrap", " return callback(*a, **kw)", " File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 747, in wrap", " return self.to_jsonp(callback(*a, **kw))", " File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper", " rv = callback(*a, **ka)", " File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 713, in wrap", " data = callback(*a, **kw)", " File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 653, in wrap", " return callback(*a, **kw)", " File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 678, in wrap", " return callback(*a, **kw)", " File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 101, in __call__", " return getattr(self, 'do_' + request.method.lower())(**kw)", " File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 235, in do_post", " return request.route_data['method'](*request.parameter_list)", " File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", " **keywords)", 
" File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", " message, timeout)", 
"DBusException: org.freedesktop.DBus.Error.Failed: Current Mode is NTP" ] 
}, "message": "500 Internal Server Error" } 
```

gkeishin, 2017-03-06:
@tomjoseph83  Can you take a look at it ?

ratagupt, 2017-03-06:
Hi George,
Here in your time test you can not set the time while the mode is "NTP".
Seems to me the problem is ..
1) IPMI was setting the time @06:46:09 
     that request reached to time manager @06:57:45 and the time manager replied saying that it can not set the time.
2) IPMI got the GET SEL TIME request after that.
    which was reached to time manger @06:57:45 and it returned the time.
So here need to take a look why there is a time difference?

Feb 22 06:46:09 witherspoon ipmid[919]: Handling Set-SEL-Time:[0xA], Cmd:[0x49] >>>>>>
Feb 22 06:46:09 witherspoon ipmid[919]: Data: 0x58BCEF71]
Feb 22 06:46:09 witherspoon ipmid[919]: Error [Permission denied] setting time
Feb 22 06:46:09 witherspoon ipmid[919]: IPMI Handling GET-SEL-TIME
Feb 22 06:57:45 witherspoon timemanager[956]: Request to SET Host time >>>>>>>>>>>
Feb 22 06:57:45 witherspoon timemanager[956]: Curr_Mode: NTPCurr_Owner: BMChost_offset: 0
Feb 22 06:57:45 witherspoon ipmid[919]: Host Time read:[Wed Feb 22 06:46:ERROR:[0xFF] handling NetFn:[0xA], Cmd:[0x49]
Feb 22 06:57:45 witherspoon dbus[763]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.66" (uid=0 pid=956 comm="/usr/sbin/timemanager ") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.Failed" requested_reply="0" destination=":1.57" (uid=0 pid=919 comm="/usr/sbin/ipmid ")
Feb 22 06:57:45 witherspoon timemanager[956]: Request to get HOST time
Feb 22 06:57:45 witherspoon timemanager[956]: Wed Feb 22 06:57:45 2017 GMT
Feb 22 06:57:45 witherspoon timemanager[956]:  Host_time_str: [ Wed Feb 22 06:57:45 2017 GMT ] host_time usec: [ 1487746665288717 ] host_offset: [ 0 ]

gkeishin, 2017-03-06:
Great. Thanks @ratagupt for the analysis..

gkeishin, 2017-03-23:
any update on this one ?

gkeishin, 2017-09-18:
Closing as no longer seen on latter builds

closed 2017-09-18 by gkeishin