openbmc/openbmc #1256: Interemittent org.freedesktop.DBus.Error.NoReply error on bmc reboots

geissonator, 2017-03-06:
Along as similar vein as #978 and #974 and #832 and #893 I've been seeing the symptom recently where we fail to get to BMC_READY due to the system_manager.py getting stuck.

All I did to hit this was reboot the BMC.  A "systemctl restart org.openbmc.managers.System.service" fixes the issue.

```
[barreleye]# systemctl status org.openbmc.managers.System.service | cat
\ufffd\ufffdorg.openbmc.managers.System.service - Phoshpor System Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.managers.System.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2017-02-01 02:36:34 UTC; 13min ago
 Main PID: 806 (python)
   CGroup: /system.slice/org.openbmc.managers.System.service
           \ufffd\ufffd\ufffd\ufffd\ufffd\ufffd806 python /usr/sbin/system_manager.py

Feb 01 02:37:51 barreleye system_manager.py[806]: Traceback (most recent call last):
Feb 01 02:37:51 barreleye system_manager.py[806]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
Feb 01 02:37:51 barreleye system_manager.py[806]:     *message.get_args_list()))
Feb 01 02:37:51 barreleye system_manager.py[806]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 91, in <lambda>
Feb 01 02:37:51 barreleye system_manager.py[806]:     path, retry, x))
Feb 01 02:37:51 barreleye system_manager.py[806]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 114, in get_object_error
Feb 01 02:37:51 barreleye system_manager.py[806]:     self.error_callback(e)
Feb 01 02:37:51 barreleye system_manager.py[806]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 54, in default_error
Feb 01 02:37:51 barreleye system_manager.py[806]:     raise e
Feb 01 02:37:51 barreleye system_manager.py[806]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
```

```
[barreleye]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-32"
VERSION_ID="v1.99.3-32-g05b7eba"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-32"
BUILD_ID="v1.99.3"
```

geissonator, 2017-03-20:
Very intermittent fail, hit another one today:
```
[barreleye]# systemctl status org.openbmc.managers.System.service | cat
\ufffd\ufffdorg.openbmc.managers.System.service - Phoshpor System Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.managers.System.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2017-02-01 02:36:50 UTC; 57min ago
 Main PID: 766 (python)
   CGroup: /system.slice/org.openbmc.managers.System.service
           \ufffd\ufffd\ufffd\ufffd\ufffd\ufffd766 python /usr/sbin/system_manager.py

Feb 01 02:37:15 barreleye system_manager.py[766]:     *message.get_args_list()))
Feb 01 02:37:15 barreleye system_manager.py[766]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 91, in <lambda>
Feb 01 02:37:15 barreleye system_manager.py[766]:     path, retry, x))
Feb 01 02:37:15 barreleye system_manager.py[766]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 114, in get_object_error
Feb 01 02:37:15 barreleye system_manager.py[766]:     self.error_callback(e)
Feb 01 02:37:15 barreleye system_manager.py[766]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 54, in default_error
Feb 01 02:37:15 barreleye system_manager.py[766]:     raise e
Feb 01 02:37:15 barreleye system_manager.py[766]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Feb 01 02:37:17 barreleye system_manager.py[766]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('USB_RESET', False)], [], 'FSI_DATA', 'FSI_CLK', 'FSI_ENABLE', 'CRONUS_SEL', [('BMC_THROTTLE', True), ('IDBTN', False)]]
Feb 01 02:37:19 barreleye system_manager.py[766]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('USB_RESET', False)], [], 'FSI_DATA', 'FSI_CLK', 'FSI_ENABLE', 'CRONUS_SEL', [('BMC_THROTTLE', True), ('IDBTN', False)]]
```
```
[barreleye]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-113"
VERSION_ID="v1.99.3-113-g70491bb"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-113"
BUILD_ID="v1.99.3"
```

williamspatrick, 2017-03-20:
Seems like a missing dependency of Wants/After

geissonator, 2017-03-22:
Some more data on the fail.  We seem to be hitting the failure while waiting for the specified services to show up in mapper:

```
Mar 21 09:23:18 barreleye system_manager.py[757]: SystemManager Goto System State: BMC_STARTING
Mar 21 09:23:18 barreleye system_manager.py[757]: Running System State: BMC_STARTING
Mar 21 09:23:25 barreleye system_manager.py[757]: New object: /org/openbmc/sensors/speed/fan5 (:1.29)
Mar 21 09:23:39 barreleye system_manager.py[757]: New object: /org/openbmc/control/power0 (:1.89)
Mar 21 09:23:39 barreleye system_manager.py[757]: New object: /org/openbmc/control/host0 (:1.92)
Mar 21 09:23:54 barreleye system_manager.py[757]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('USB_RESET', False)], [], 'FSI_DATA', 'FSI_CLK', 'FSI_ENABLE', 'CRONUS_SEL', [('BMC_THROTTLE', True), ('IDBTN', False)]]
Mar 21 09:23:55 barreleye system_manager.py[757]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('USB_RESET', False)], [], 'FSI_DATA', 'FSI_CLK', 'FSI_ENABLE', 'CRONUS_SEL', [('BMC_THROTTLE', True), ('IDBTN', False)]]
Mar 21 09:24:03 barreleye system_manager.py[757]: Traceback (most recent call last):
Mar 21 09:24:03 barreleye system_manager.py[757]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
Mar 21 09:24:03 barreleye system_manager.py[757]:     *message.get_args_list()))
Mar 21 09:24:03 barreleye system_manager.py[757]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 91, in <lambda>
Mar 21 09:24:03 barreleye system_manager.py[757]:     path, retry, x))
Mar 21 09:24:03 barreleye system_manager.py[757]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 114, in get_object_error
Mar 21 09:24:03 barreleye system_manager.py[757]:     self.error_callback(e)
Mar 21 09:24:03 barreleye system_manager.py[757]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 54, in default_error
Mar 21 09:24:03 barreleye system_manager.py[757]:     raise e
Mar 21 09:24:03 barreleye system_manager.py[757]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
```

Here's what it's waiting for:

```
	'BMC_STARTING' : {
23		'/org/openbmc/control/power0' : 0,
24		'/org/openbmc/control/host0' : 0,
25		'/org/openbmc/control/flash/bios' : 0,
26		'/org/openbmc/sensors/speed/fan5': 0,
27		'/org/openbmc/inventory/system/chassis/io_board' : 0,
28	},
```

3 of the services show up in the logs, but bios and io_board don't.

Which if you add in their traces shows this:
```
Mar 21 09:23:25 barreleye system_manager.py[757]: New object: /org/openbmc/sensors/speed/fan5 (:1.29)
Mar 21 09:23:26 barreleye systemd[1]: Starting Read system/chassis/io_board EEPROM...
Mar 21 09:23:39 barreleye system_manager.py[757]: New object: /org/openbmc/control/power0 (:1.89)
Mar 21 09:23:39 barreleye system_manager.py[757]: New object: /org/openbmc/control/host0 (:1.92)
Mar 21 09:23:54 barreleye system_manager.py[757]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('USB_RESET', False)], [], 'FSI_DATA', 'FSI_CLK', 'FSI_ENABLE', 'CRONUS_SEL', [('BMC_THROTTLE', True), ('IDBTN', False)]]
Mar 21 09:23:55 barreleye system_manager.py[757]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('USB_RESET', False)], [], 'FSI_DATA', 'FSI_CLK', 'FSI_ENABLE', 'CRONUS_SEL', [('BMC_THROTTLE', True), ('IDBTN', False)]]
Mar 21 09:23:56 barreleye mapper[863]: Error finding '/org/openbmc/control/flash/bios' service: Connection timed out
Mar 21 09:23:56 barreleye systemd[1]: obmc-read-eeprom@system-chassis-io_board.service: Main process exited, code=exited, status=1/FAILURE
Mar 21 09:23:56 barreleye systemd[1]: Failed to start Read system/chassis/io_board EEPROM.
Mar 21 09:23:56 barreleye systemd[1]: obmc-read-eeprom@system-chassis-io_board.service: Unit entered failed state.
Mar 21 09:23:56 barreleye systemd[1]: obmc-read-eeprom@system-chassis-io_board.service: Failed with result 'exit-code'.
Mar 21 09:24:01 barreleye systemd[1]: Starting Sync system/chassis/io_board...
Mar 21 09:24:03 barreleye system_manager.py[757]: Traceback (most recent call last):
Mar 21 09:24:03 barreleye system_manager.py[757]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
Mar 21 09:24:03 barreleye system_manager.py[757]:     *message.get_args_list()))
```

So I guess the main issue here has to do with these required services timing out and failing.

geissonator, 2017-03-22:
One weird thing is that system_manager.py is waiting on a mapper call to indicate obmc-read-eeprom@system-chassis-io_board.service has completed and it looks like obmc-read-eeprom@system-chassis-io_board.service is also waiting on a mapper call to get the system manager object from mapper at the same time.
```
Mar 21 09:23:26 barreleye systemd[1]: Starting Read system/chassis/io_board EEPROM...
Mar 21 09:23:56 barreleye phosphor-read-eeprom[849]: Failed to get system manager service:[Connection timed out]
Mar 21 09:23:56 barreleye phosphor-read-eeprom[849]: ERROR: cannot get defined fru area
```

geissonator, 2017-03-24:
Potential fix here - https://gerrit.openbmc-project.xyz/#/c/3265/

geissonator, 2017-03-24:
Did a quick test of the fix, didn't seem to help :(
```
[barreleye]# tracf | grep system_manager.py
Mar 15 19:36:12 barreleye system_manager.py[768]: Running System State: BASE_APPS
Mar 15 19:36:12 barreleye system_manager.py[768]: SystemManager Init Done
Mar 15 19:36:12 barreleye system_manager.py[768]: Running SystemManager
Mar 15 19:36:17 barreleye system_manager.py[768]: SystemManager Goto System State: BMC_STARTING
Mar 15 19:36:17 barreleye system_manager.py[768]: Running System State: BMC_STARTING
Mar 15 19:36:38 barreleye system_manager.py[768]: New object: /org/openbmc/sensors/speed/fan5 (:1.30)
Mar 15 19:36:59 barreleye system_manager.py[768]: Traceback (most recent call last):
Mar 15 19:36:59 barreleye system_manager.py[768]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
Mar 15 19:37:00 barreleye system_manager.py[768]:     *message.get_args_list()))
Mar 15 19:37:00 barreleye system_manager.py[768]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 91, in <lambda>
Mar 15 19:37:00 barreleye system_manager.py[768]:     path, retry, x))
Mar 15 19:37:00 barreleye system_manager.py[768]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 114, in get_object_error
Mar 15 19:37:00 barreleye system_manager.py[768]:     self.error_callback(e)
Mar 15 19:37:00 barreleye system_manager.py[768]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 54, in default_error
Mar 15 19:37:00 barreleye system_manager.py[768]:     raise e
Mar 15 19:37:00 barreleye system_manager.py[768]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Mar 15 19:37:07 barreleye system_manager.py[768]: New object: /org/openbmc/control/power0 (:1.77)
Mar 15 19:37:08 barreleye system_manager.py[768]: New object: /org/openbmc/control/host0 (:1.78)
Mar 15 19:37:11 barreleye system_manager.py[768]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('USB_RESET', False)], [], 'FSI_DATA', 'FSI_CLK', 'FSI_ENABLE', 'CRONUS_SEL', [('BMC_THROTTLE', True), ('IDBTN', False)]]
Mar 15 19:37:12 barreleye system_manager.py[768]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('USB_RESET', False)], [], 'FSI_DATA', 'FSI_CLK', 'FSI_ENABLE', 'CRONUS_SEL', [('BMC_THROTTLE', True), ('IDBTN', False)]]
```

geissonator, 2017-05-12:
This issue magically disappeared in recent firmware levels so closing it out.

closed 2017-05-12 by geissonator