openbmc/openbmc #1259: OpenFSI  on re-scan hub devices are not removed from /sys/devices

cbostic, 2017-03-06:
On re-scan the openFSI code will create new hub devices at /sys/devices/hub@00  and hub@01   (assuming 2 proc system).   Problem however is that any old hub devices from previous scan were not removed.  As a result we get a sysfs stack dump indicating duplicate name errors.

GHF, 2017-03-06:
Example dump from Zaius:

```
------------[ cut here ]------------
WARNING: CPU: 0 PID: 1303 at /usr/local/google/home/xow/obmc/linux/fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
sysfs: cannot create duplicate filename '/devices/hub@00'
CPU: 0 PID: 1303 Comm: sh Tainted: G        W       4.7.10-39eb5a44bfd49fe530f6b68c4d99ab2d8d391be0 #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<80110cbc>] (__warn+0xc4/0xf0)
[<80110cbc>] (__warn) from [<80110d2c>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d2c>] (warn_slowpath_fmt) from [<801f88dc>] (sysfs_warn_dup+0x50/0x70)
[<801f88dc>] (sysfs_warn_dup) from [<801f8970>] (sysfs_create_dir_ns+0x74/0x88)
[<801f8970>] (sysfs_create_dir_ns) from [<802a2940>] (kobject_add_internal+0x108/0x30c)
[<802a2940>] (kobject_add_internal) from [<802a2bc0>] (kobject_add+0x7c/0xb4)
[<802a2bc0>] (kobject_add) from [<802e5fec>] (device_add+0x118/0x53c)
[<802e5fec>] (device_add) from [<803504cc>] (fsi_master_register+0x438/0x8b0)
[<803504cc>] (fsi_master_register) from [<80350df4>] (store_scan+0x8c/0xb0)
[<80350df4>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
[<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
[<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
[<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
[<801a36d4>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
---[ end trace 6b436221b1a92423 ]---
------------[ cut here ]------------
WARNING: CPU: 0 PID: 1303 at /usr/local/google/home/xow/obmc/linux/lib/kobject.c:240 kobject_add_internal+0x290/0x30c
kobject_add_internal failed for hub@00 with -EEXIST, don't try to register things with the same name in the same directory.
CPU: 0 PID: 1303 Comm: sh Tainted: G        W       4.7.10-39eb5a44bfd49fe530f6b68c4d99ab2d8d391be0 #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<80110cbc>] (__warn+0xc4/0xf0)
[<80110cbc>] (__warn) from [<80110d2c>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d2c>] (warn_slowpath_fmt) from [<802a2ac8>] (kobject_add_internal+0x290/0x30c)
[<802a2ac8>] (kobject_add_internal) from [<802a2bc0>] (kobject_add+0x7c/0xb4)
[<802a2bc0>] (kobject_add) from [<802e5fec>] (device_add+0x118/0x53c)
[<802e5fec>] (device_add) from [<803504cc>] (fsi_master_register+0x438/0x8b0)
[<803504cc>] (fsi_master_register) from [<80350df4>] (store_scan+0x8c/0xb0)
[<80350df4>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
[<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
[<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
[<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
[<801a36d4>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
---[ end trace 6b436221b1a92424 ]---
```

cbostic, 2017-03-07:
Two patches address this issue:

http://patchwork.ozlabs.org/patch/735992/
The following depends on the previous:
http://patchwork.ozlabs.org/patch/736019/

rfrandse, 2017-03-27:
Patches Merged

closed 2017-03-27 by rfrandse

bradbishop, 2017-03-27:
I don't have the exact commit but this is supposed to be fixed as of at least openbmc/linux@168b8cb (probably earlier).