openbmc/openbmc #1261: FSI cannot re-bind to device

shenki, 2017-03-07:
With a newly booted zaius running `4.7.10-2763f3eb775cf760b1c0f28d283da6df6c42f061` ([v1.99.3-42-ge8c604b](e8c604b)) I performed an unbind and then attempted to bind the driver again:
```
root@zaius:/sys/devices/platform/fsi-master/driver# echo fsi-master > unbind 
root@zaius:/sys/devices/platform/fsi-master/driver# echo fsi-master > bind 
-sh: echo: write error: No such device
```
It failed with this warning:

```
WARNING: CPU: 0 PID: 856 at /var/lib/jenkins-slave/workspace/openbmc-build/distro/ubuntu/target/zaius/openbmc/build/tmp/work-shared/zaius/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
sysfs: cannot create duplicate filename '/devices/platform/fsi-master/scan'
CPU: 0 PID: 856 Comm: sh Not tainted 4.7.10-2763f3eb775cf760b1c0f28d283da6df6c42f061 #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<80110cbc>] (__warn+0xc4/0xf0)
[<80110cbc>] (__warn) from [<80110d2c>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d2c>] (warn_slowpath_fmt) from [<801f88dc>] (sysfs_warn_dup+0x50/0x70)
[<801f88dc>] (sysfs_warn_dup) from [<801f8520>] (sysfs_add_file_mode_ns+0xfc/0x184)
[<801f8520>] (sysfs_add_file_mode_ns) from [<801f85fc>] (sysfs_create_file_ns+0x38/0x40)
[<801f85fc>] (sysfs_create_file_ns) from [<802e9c9c>] (platform_drv_probe+0x38/0x6c)
[<802e9c9c>] (platform_drv_probe) from [<802e88c0>] (driver_probe_device+0x1b0/0x3d8)
[<802e88c0>] (driver_probe_device) from [<802e6edc>] (bind_store+0x98/0xe0)
[<802e6edc>] (bind_store) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
[<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
[<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
[<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
[<801a36d4>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
---[ end trace 6189bd2727d1a562 ]---
fsi-master-gpio: probe of fsi-master failed with error -17
```

@cbostic the driver needs to tear down the resources it creates when initialising the driver

cbostic, 2017-03-08:
Patch submitted to address this issue:

http://patchwork.ozlabs.org/patch/736698/

cbostic, 2017-03-08:
Attempted several unbind/bind/scan sequences without any errors.   Multiple unbinds in sequence, multiple binds in sequence do not generate any problems.   Attempted several scans in a row and mixed in various combos of unbind and bind without issue.

cbostic, 2017-10-16:
@rfrandse can you please close this out?  This was addressed earlier in the year.

closed 2019-04-27 by stale[bot]