openbmc/openbmc #1262: Palmetto failed to power-on host

adamliyi, 2017-03-07:
I tested openbmc using latest auto-build image:
https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=palmetto/lastBuild/artifact/images/palmetto/

The openbmc version is:

```
root@palmetto:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-42"
VERSION_ID="v1.99.3-42-ge8c604b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-42"
BUILD_ID="v1.99.3"
```

Run `obmcutil poweron` fails to power-on the host - watchdog will reset the booting.

```
Mar 07 06:47:35 palmetto systemd[1]: Started Start Power0.
Mar 07 06:47:35 palmetto system_manager.py[769]: Running System State: HOST_POWERING_ON
Mar 07 06:47:35 palmetto timemanager[854]:  PGOOD has changed..
Mar 07 06:47:37 palmetto kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Mar 07 06:47:37 palmetto kernel[705]: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Mar 07 06:47:37 palmetto system_manager.py[769]: Running System State: HOST_POWERED_ON
Mar 07 06:47:37 palmetto timemanager[854]:  User Settings have changed..
Mar 07 06:47:37 palmetto power_control.exe[819]: PowerControl: pgood: 1, setting reset USB_RESET to 1
Mar 07 06:47:37 palmetto systemd[1]: Started Wait for Power0 to turn on.
Mar 07 06:47:37 palmetto timemanager[854]:  PGOOD has changed..
Mar 07 06:47:37 palmetto timemanager[854]: Changing time settings is *deferred* now
Mar 07 06:47:37 palmetto systemd[1]: Reached target Power0 On.
Mar 07 06:47:37 palmetto systemd[1]: Reached target Power0 (On).
Mar 07 06:47:37 palmetto systemd[1]: Starting Start Host0...
Mar 07 06:47:37 palmetto systemd[1]: Starting PCIE Slot Presence Detection For Chassis0...
Mar 07 06:47:37 palmetto systemd[1]: Reached target Chassis0 (Power On).
Mar 07 06:47:37 palmetto phosphor-chassis-state-manager[823]: Recieved signal that power ON is complete
Mar 07 06:47:37 palmetto phosphor-chassis-state-manager[823]: Change to Chassis Power State
Mar 07 06:47:38 palmetto systemd[1]: Started PCIE Slot Presence Detection For Chassis0.
Mar 07 06:47:38 palmetto control_host.exe[808]: Booting host
Mar 07 06:47:38 palmetto systemd[1]: Started Start Host0.
Mar 07 06:47:39 palmetto systemd[1]: Reached target Start Host0.
Mar 07 06:47:39 palmetto systemd[1]: Reached target Host0 (Started).
Mar 07 06:47:39 palmetto systemd[1]: Starting Start Watchdog0...
Mar 07 06:47:39 palmetto host_watchdog.exe[706]: Setting watchdog poll interval to: 30000
Mar 07 06:47:39 palmetto control_host.exe[808]: Using primary side of the bios flash
Mar 07 06:47:39 palmetto host_watchdog.exe[706]: Starting watchdog with poll interval: 30000
Mar 07 06:47:39 palmetto systemd[1]: Started Start Watchdog0.
Mar 07 06:47:39 palmetto systemd[1]: Reached target Chassis0 (Start).
Mar 07 06:47:39 palmetto phosphor-host-state-manager[807]: Recieved signal that host is running
Mar 07 06:47:39 palmetto phosphor-host-state-manager[807]: Change to Host State
Mar 07 06:47:40 palmetto system_manager.py[769]: Running System State: HOST_BOOTING
Mar 07 06:47:40 palmetto host_watchdog.exe[706]: Setting watchdog poll interval to: 30000
Mar 07 06:47:40 palmetto host_watchdog.exe[706]: Starting watchdog with poll interval: 30000
Mar 07 06:48:10 palmetto host_watchdog.exe[706]: Polling watchdog: 0
Mar 07 06:48:10 palmetto chassis_control.py[681]: Watchdog Error, Going to quiesce
Mar 07 06:48:10 palmetto systemd[1]: Reached target Quiesce Target.
Mar 07 06:48:11 palmetto phosphor-host-state-manager[807]: Auto reboot enabled. Beginning reboot...
Mar 07 06:48:11 palmetto phosphor-host-state-manager[807]: Host State transaction request
Mar 07 06:48:11 palmetto systemd[1]: Stopped target Quiesce Target.
Mar 07 06:48:11 palmetto phosphor-host-state-manager[807]: Auto reboot enabled. Beginning reboot...
Mar 07 06:48:11 palmetto phosphor-host-state-manager[807]: Host State transaction request
Mar 07 06:48:11 palmetto systemd[1]: Starting Stop Watchdog0...
Mar 07 06:48:11 palmetto systemd[1]: Started Stop Power0.
Mar 07 06:48:11 palmetto systemd[1]: Stopped target Chassis0 (Start).
Mar 07 06:48:11 palmetto systemd[1]: Stopped Start Watchdog0.
Mar 07 06:48:11 palmetto systemd[1]: Stopped PCIE Slot Presence Detection For Chassis0.
Mar 07 06:48:11 palmetto systemd[1]: Stopped target Chassis0 (Power On).
Mar 07 06:48:11 palmetto host_watchdog.exe[706]: Stopping watchdog
Mar 07 06:48:11 palmetto phosphor-chassis-state-manager[823]: Recieved signal that power ON is complete
Mar 07 06:48:11 palmetto power_control.exe[819]: PowerControl: setting power up POWER_PIN to 1
Mar 07 06:48:11 palmetto phosphor-chassis-state-manager[823]: Change to Chassis Power State
Mar 07 06:48:12 palmetto system_manager.py[769]: Running System State: HOST_POWERING_OFF
Mar 07 06:48:12 palmetto system_manager.py[769]: Running System State: HOST_POWERED_OFF
Mar 07 06:48:12 palmetto sensor_manager2.py[703]: Deleting OCC device
Mar 07 06:48:12 palmetto power_control.exe[819]: PowerControl: pgood: 0, setting reset USB_RESET to 0
Mar 07 06:48:12 palmetto timemanager[854]:  PGOOD has changed..
Mar 07 06:48:12 palmetto timemanager[854]:  User Settings have changed..
Mar 07 06:48:12 palmetto timemanager[854]:  PGOOD has changed..
Mar 07 06:48:12 palmetto timemanager[854]: Changing time settings allowed now
Mar 07 06:48:11 palmetto systemd[1]: Starting Wait for Power0 to turn off...
Mar 07 06:48:11 palmetto systemd[1]: Stopped target Host0 (Started).
Mar 07 06:48:11 palmetto systemd[1]: Stopped target Start Host0.
Mar 07 06:48:11 palmetto systemd[1]: Stopped Start Host0.
Mar 07 06:48:11 palmetto systemd[1]: Stopped target Power0 (On).
Mar 07 06:48:11 palmetto systemd[1]: Stopped target Power0 On.
Mar 07 06:48:11 palmetto systemd[1]: Stopped Start Power0.
Mar 07 06:48:11 palmetto systemd[1]: Stopped Wait for Power0 to turn on.
Mar 07 06:48:11 palmetto systemd[1]: Started Stop Watchdog0.
Mar 07 06:48:12 palmetto phosphor-host-state-manager[807]: Recieved signal that host is running
Mar 07 06:48:12 palmetto phosphor-host-state-manager[807]: Change to Host State
Mar 07 06:48:12 palmetto systemd[1]: Started Wait for Power0 to turn off.
Mar 07 06:48:13 palmetto systemd[1]: Reached target Chassis0 (Power Off).
Mar 07 06:48:13 palmetto phosphor-chassis-state-manager[823]: Recieved signal that power OFF is complete
Mar 07 06:48:13 palmetto phosphor-chassis-state-manager[823]: Change to Chassis Power State
Mar 07 06:48:13 palmetto systemd[1]: Reached target Power0 Off.
Mar 07 06:48:13 palmetto systemd[1]: Reached target Chassis0 (Stop).
Mar 07 06:48:13 palmetto phosphor-host-state-manager[807]: Recieved signal that host is off
Mar 07 06:48:13 palmetto phosphor-host-state-manager[807]: Change to Host State
Mar 07 06:48:13 palmetto phosphor-host-state-manager[807]: Reached intermediate state, going to next
Mar 07 06:48:13 palmetto systemd[1]: Starting Wait for Power0 to turn on...
Mar 07 06:48:13 palmetto systemd[1]: Stopped target Chassis0 (Stop).
Mar 07 06:48:13 palmetto phosphor-host-state-manager[807]: Recieved signal that host is off
Mar 07 06:48:13 palmetto phosphor-host-state-manager[807]: Change to Host State
Mar 07 06:48:13 palmetto phosphor-host-state-manager[807]: Reached intermediate state, going to next
Mar 07 06:48:13 palmetto systemd[1]: Stopped Stop Watchdog0.
Mar 07 06:48:13 palmetto systemd[1]: Stopped target Chassis0 (Power Off).
Mar 07 06:48:13 palmetto phosphor-chassis-state-manager[823]: Recieved signal that power OFF is complete
Mar 07 06:48:13 palmetto phosphor-chassis-state-manager[823]: Change to Chassis Power State
Mar 07 06:48:13 palmetto systemd[1]: Stopped target Power0 Off.
Mar 07 06:48:13 palmetto systemd[1]: Stopped Wait for Power0 to turn off.
Mar 07 06:48:13 palmetto systemd[1]: Stopped Stop Power0.
Mar 07 06:48:13 palmetto systemd[1]: Starting Start Power0...
Mar 07 06:48:14 palmetto power_control.exe[819]: PowerControl: setting power up POWER_PIN to 0
Mar 07 06:48:14 palmetto systemd[1]: Started Start Power0.
Mar 07 06:48:14 palmetto kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Mar 07 06:48:14 palmetto kernel[705]: ftgmac100 1e660000.ethernet eth0: NCSI interface down
```

@gwshan is seeing the same issue.

gkeishin, 2017-03-07:
Apart from ftgmac100  issue seen https://github.com/openbmc/openbmc/issues/1064
There is this Host going for auto reboot reaching `Quiesce Mode` 
```
Mar 07 06:48:10 palmetto chassis_control.py[681]: Watchdog Error, Going to quiesce
Mar 07 06:48:10 palmetto systemd[1]: Reached target Quiesce Target.
```

shenki, 2017-03-09:
fwiw, I boot the host with the latest master on palmetto daily

adamliyi, 2017-03-09:
Just check the Palmetto (palm309) in the reservation system, which have the same issue - so I think this bug is valid.

```
root@palmetto:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-45"
VERSION_ID="v1.99.3-45-g8e6b232"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-45"
BUILD_ID="v1.99.3"
```

geissonator, 2017-03-09:
I've seen this a bit lately on our other systems.  Looks to me like PNOR is getting corrupted or something to the point where hostboot can't start.  Try re-flashing PNOR?  That fixes it for me.

gwshan, 2017-03-09:
@adamliyi helped build BMC image from "master" branch. The host doesn't boot after "obmcutil poweron". The host boots well after I reflashed the PNOR image.

gwshan, 2017-03-09:
@adamliyi I think this bug can be closed if we doesn't plan to track the corrupted PNOR issue by this.

yadron, 2017-03-09:
The same situation with Barreleye (cannot boot host), and I noted an error during update (when tried updating second time for the same latest master image):

```
root@barreleye:/run/initramfs# ./update 
update:
Updating bmc...
Erasing block: 128/128 (100%) 
Writing kb: 32768/32768 (100%) 
Verifying kb: 32336/32768 (98%) flashcp: verification mismatch at 0x1f94000
```

yadron, 2017-03-09:
For me it looks like some processes uses incorrectly flash memory and/or CPU (too often read? too often write? Could be some broken processes that being restarted in cycle).

After prev. msg, I've recompiled image and tried updating again (every time different errors during verification - see below-, so for me it looks like non-flash hardware problem, but software problem, and at last I think that some new python or dbus services or sensors readers makes CPU 100% busy from time to time for the latest master bundle and as result flashcp works badly and host cannot boot maybe because of this too or because of ipmid broken):

```
root@barreleye:/run/initramfs# ./update 
update:
Updating bmc...
Erasing block: 128/128 (100%) 
Writing kb: 32768/32768 (100%) 
Verifying kb: 32392/32768 (98%) flashcp: verification mismatch at 0x1fa2000

```
And one more attempt:

```
root@barreleye:/run/initramfs# ./update 
update:
Updating bmc...
Erasing block: 128/128 (100%) 
Writing kb: 32768/32768 (100%) 
Verifying kb: 32404/32768 (98%) flashcp: verification mismatch at 0x1fa5000
root@barreleye:/run/initramfs#
```

More attempts.. Every time verify does more than prev attampt, but fails on ~ +20 kb blocks..

I've tried killing "/usr/sbin/flash_bios.exe" process (just in case it does something with flash).

Updating again, the same error.

I've tried killing bmc_update.py then:

```
root@barreleye:/run/initramfs# cp /usr/sbin/bmc_update.py /usr/sbin/bmc_update_.py
root@barreleye:/run/initramfs# rm /usr/sbin/bmc_update.py                         
root@barreleye:/run/initramfs# ps|grep bmc_up
 5523 root       0:03 python /usr/sbin/bmc_update.py
 5564 root       0:00 grep bmc_up
root@barreleye:/run/initramfs# kill 5523
root@barreleye:/run/initramfs# ps|grep bmc_up
 5569 root       0:00 grep bmc_up
root@barreleye:/run/initramfs# ./update

```

and again.. Still verify error..

I've noted that for some reason during flash update or any other activity the system stalls sometimes (frozen for say 20 seconds without reason).. So I thought that this is not flash error, but some services makes system busy and as result flashcp works incorrectly because of this..

So tried "top" few times and still removing most active services:

```
root@barreleye:/run/initramfs# cp /usr/sbin/hwmon.py /usr/sbin/hwmon_.py  
root@barreleye:/run/initramfs# rm /usr/sbin/hwmon.py                      
root@barreleye:/run/initramfs# ps|grep hwmo
  715 root       0:01 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@40/rtc@68
  718 root       0:01 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@40/lm75@4a
  778 root       5:27 python /usr/sbin/hwmon.py
 5800 root       0:02 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@1c0/nct7904@2e
 5808 root       0:02 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@1c0/nct7904@2d
 5823 root       0:00 grep hwmo
root@barreleye:/run/initramfs# kill 778
root@barreleye:/run/initramfs# ps|grep hwmo
  715 root       0:01 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@40/rtc@68
  718 root       0:01 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@40/lm75@4a
 5800 root       0:02 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@1c0/nct7904@2e
 5808 root       0:02 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@1c0/nct7904@2d
 5836 root       0:00 grep hwmo
root@barreleye:/run/initramfs# ps|grep hwmo
[... 735 lines cut ...]
Failed to find Sensor Type 0x0
997: Default
Failed to issue method call: Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python2.7/site-packages/obmc/dbuslib/bindings.py", line 82, in Set
    self.validator(interface_name, property_name, new_value)
  File "/usr/sbin/settings_manager.py", line 236, in input_validator
    self.validate_list(shk['allowed'], value)
  File "/usr/sbin/settings_manager.py", line 176, in validate_list
    raise ValueError("Invalid input. Data not in allowed values")
ValueError: Invalid input. Data not in allowed values

and console-client working again:

root@barreleye:~# obmc-console-client 
 38.70957|ISTEP 21. 1
....

```

and **host booted well** using "v1.99.3-dirty".

As result, the version "v1.99.3-48" looks very-very-very buggy (it looks like it even broken FS so it looks like flash-hardware problems instead, why? Is it because of CPU 100% busy like I thought above?).
If it will not be fixed soon, then I could suggest moving back to "v1.99.3-dirty" and trying aging...

adamliyi, 2017-03-10:
@gwshan , @geissonator . Yes, I tried to re-flash PNOR, and Palmetto host can be booted correctly on my side too. I talked with FW people, it is said that the PNOR has the possibility to become corrupted - if the host is powered off during booting, since the PNOR will do Read and Write during booting.  Will close this issue.

closed 2017-03-10 by adamliyi