openbmc/openbmc #1263: SDK failing to build

geissonator, 2017-03-07:
https://openpower.xyz/job/openbmc-build-sdk/

More of the *.conf stuff that we hit in just building master a week or so ago.

```
13:33:45 ERROR: Logfile of failure stored in: /var/lib/jenkins-slave/workspace/openbmc-build-sdk/distro/ubuntu/target/barreleye/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/temp/log.do_compile.20876
13:33:45 Log data follows:
13:33:45 | DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
13:33:45 | DEBUG: Executing shell function do_compile
13:33:45 | NOTE: make -j16
13:33:45 | make: *** No rule to make target '/var/lib/jenkins-slave/workspace/openbmc-build-sdk/distro/ubuntu/target/barreleye/openbmc/build/tmp/sysroots/x86_64-linux/usr/share/phosphor-ipmi-host/*.conf', needed by 'ipmiwhitelist.cpp'.  Stop.
13:33:45 | WARNING: exit code 1 from a shell command.
13:33:45 | ERROR: oe_runmake failed
13:33:45 | ERROR: Function failed: do_compile (log file is located at /var/lib/jenkins-slave/workspace/openbmc-build-sdk/distro/ubuntu/target/barreleye/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/temp/log.do_compile.20876)
```

geissonator, 2017-04-02:
Did a fresh x86 SDK build this morning and hit this same thing.

My bad hack is to just copy phosphor-host-ipmid/host-ipmid-whitelist.conf into the directory it's complaining about and rebuild.

mine260309, 2017-04-18:
I hit on the same issue trying to build x86-64 SDK.

ratagupt, 2017-05-27:
Are we still hitting the same issue,I tried to build and have not hit.
@geissonator : can you confirm?

williamspatrick, 2017-05-27:
It will only fail with an SDK build from an sstate cache.

geissonator, 2017-05-30:
We have a jenkins job that keeps trying (and failing) if you need some info to look at @ratagupt  - https://openpower.xyz/job/openbmc-build-sdk/

amboar, 2017-07-06:
Issue has reoccurred on openpower.xyz. Last successful build was 16th of June.

geissonator, 2018-01-23:
I think the issue is that the provided .conf files are being provided by the openpower layer, and we can't really have a dependency from phosphor-host-ipmid on openpower.

From a working build:
```
ls ./tmp/sysroots/x86_64-linux/usr/share/phosphor-ipmi-host/
config  openpower-ipmi-oem-whitelist-native.conf  phosphor-ipmi-fru-whitelist-native.conf  sensor  sensor-yamls
```

Although that phosphor-ipmi-fru-whitelist-native.conf file is from the phosphor layer.  Hmmm.

gtmills, 2018-05-23:
@geissonator This fixed? Can we close?