openbmc/openbmc #1265: Add support for iio device callouts

bradbishop, 2017-03-07:
The iio-hwmon bridge allows phosphor-hwmon to be used with iio devices.  The current code will callout the iio-hwmon device.  Need to enhance the callout lookup to be the actual iio device.

bjwyman, 2017-05-16:
[] Create a configuration file for one of the IIO hwmon drivers used (bmp280).
[] Update udev rule algorithm to find iio hwmon instance via phandle value
[] Update call out algorithm to find path for iio hwmon instance

bradbishop, 2017-05-18:
for reference here is a udev event from the bmp280

```root@witherspoon:/sys/firmware/devicetree/base/iio-hwmon# udevadm info --path /devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-3/i2c-3/3-0077/iio:device0
P: /devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-3/i2c-3/3-0077/iio:device0
N: iio:device0
E: DEVNAME=/dev/iio:device0
E: DEVPATH=/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-3/i2c-3/3-0077/iio:device0
E: DEVTYPE=iio_device
E: MAJOR=252
E: MINOR=0
E: OF_COMPATIBLE_0=bosch,bmp280
E: OF_COMPATIBLE_N=1
**** E: OF_FULLNAME=/ahb/apb/i2c@1e78a000/i2c-bus@100/bmp280@77 ****
E: OF_NAME=bmp280
E: SUBSYSTEM=iio```

bjwyman, 2017-06-02:
https://gerrit.openbmc-project.xyz/4573 Update call out algorithm for iio hwmon instance
Resolves #1265 Add support for iio devices

rfrandse, 2017-06-03:
<https://gerrit.openbmc-project.xyz/4573> Update call out algorithm for iio hwmon instance
Resolves  #1265 Add support for iio devices

closed 2017-06-06 by williamspatrick