openbmc/openbmc #1266: Generate bmc errors on host checkstop and host watchdog failures

geissonator, 2017-03-07:


geissonator, 2017-04-03:
Per recent design meeting, can just create the new target and services for this function, no need to have the function refactored.

vishwabmc, 2017-04-05:
I discussed this with @devenrao and here were my thoughts.

* Create the Error yaml files
* Create the tiny application that would consume the yaml and then log an error when called into
* Include this app as part of the new obmc-gpio-asserted@.target.wants

devenrao, 2017-04-06:
- [ ] Create an error yaml and metadata file
- [ ] Create a service that consumes the error yaml file and log's an error when called into
- [ ] Link the service to obmc-gpio-asserted@.target.wants

rfrandse, 2017-05-11:
https://gerrit.openbmc-project.xyz/3937 Add dependency for openpower-debug-collector
Resolves #1266 Generate bmc errors on host checkstop and host watchdog failures

closed 2017-05-13 by williamspatrick