openbmc/openbmc #1271: Sprint 17.03.08 - Committed

rfrandse, 2017-03-07:


spinler, 2017-03-09:
Squad 2

dkodihal, 2017-03-09:
SQUAD1:

Deepak
1093: Ready for merge review.
1249: Code done. Will push for review by 03/10.
1275: Current Sprint - task break-down ready. Need to discuss details.

Tom
856: Work in progress
1212: Needs to be moved to current sprint. To be assigned to new members?

Ratan
781: Ready for merge review. Tom's portion is up for peer review.
1162: Ready for merge review.

Vishwa
Needs merge review on
https://gerrit.openbmc-project.xyz/#/c/2858/
https://gerrit.openbmc-project.xyz/#/c/2860/1
https://gerrit.openbmc-project.xyz/#/c/2727/

anoo1, 2017-03-09:
-------------------------------------------------------------------------------------------------
**Squad 3 -**
-------------------------------------------------------------------------------------------------
-------------------------------------------------------------------------------------------------
**Adriana:anoo1 -**
**17.02.08:**
- https://github.com/openbmc/openbmc/issues/1079 (program to create and update UBI volumes) - In review

**17.02.22:**
- https://github.com/openbmc/openbmc/issues/1048 (Create a UBIFS filesystem to store the PNOR RW partitions) - In review

**17.03.08:**
- https://github.com/openbmc/openbmc/issues/1277 (ItemUpdater: Initial creation and monitoring of new Software.Versions)
    - 03/09: Start to create app
- https://github.com/openbmc/openbmc/issues/1278 (ItemUpdater: Create UBI volumes when RequestedActivation)

-------------------------------------------------------------------------------------------------
**Gunnar:gtmills**
**17.02.08:**
- https://github.com/openbmc/openbmc/issues/1047 (Parse PNOR TOC and create metadata file) - in review

**17.03.08:**
-  https://github.com/openbmc/openbmc/issues/1276 (ImageManager: Initial creation and population of Software.Version properties)
    - 03/09: Made initial commit

-------------------------------------------------------------------------------------------------
**Andrew:geissonator -** 
**17.02.08:**
- https://github.com/openbmc/openbmc/issues/1094 (Handle a BMC reset while Host is running) - in review
    - 03/09: Testing

**17.02.22:**
- https://github.com/openbmc/openbmc/issues/954 (chassis / host state questions) - Can be closed

**17.03.08:**
- https://github.com/openbmc/openbmc/issues/1274 (host state heartbeat adn target work on r/r)
    - 03/09: Starting

-------------------------------------------------------------------------------------------------
**Josh:JoshDKing -** 
Attends Tuesday and Thursday

**17.02.22:**
- https://github.com/openbmc/openbmc/issues/1194 (Chassis and Host state services show incorrect states when targets exit)
- https://github.com/openbmc/openbmc/issues/1205 (Refactor action target names)

**17.03.08:**
- < incorporate new soft power off function into host state >

-------------------------------------------------------------------------------------------------
**Michael:mtritz -** 
**17.03.08:**
- https://github.com/openbmc/openbmc/issues/773 (discover_system state refactor
    - 03/09: Meeting with Andrew to get details on the story.

-------------------------------------------------------------------------------------------------
**Saqib: -** 

**17.02.22:**
- https://github.com/openbmc/openbmc/issues/1143
- https://github.com/openbmc/openbmc/issues/931 (Update yocto)
    - 03/09: Will ask Patrick if he's taking over the review.
- https://github.com/openbmc/openbmc/issues/1181 (Auto reboot never stops) 
    - 03/09: Need to implement changes per comments from Patrick.

**17.03.08:**
- https://github.com/openbmc/openbmc/issues/932 (Upgrade systemd to 231 or above.)
    - 03/09: Starting.

-------------------------------------------------------------------------------------------------
**Leonel: -** 
**17.02.08:**
- https://github.com/openbmc/openbmc/issues/1137 (dbus msg support in openbmc CI)
    - 03/09: Got feedback from Patrick.

**17.02.22:**
- https://github.com/openbmc/openbmc/issues/1112 (standardize dbus names)
    - 03/09: Checking if there are any repos left that need updating.

**17.03.08:**
- https://github.com/openbmc/openbmc-test-automation/issues/406 (real validation of boot flag)
    - 03/09: Starting

closed 2017-03-30 by rfrandse