openbmc/openbmc #1272: Mnfg: DHCP issues on Witherspoon

cdeadmin, 2017-03-08:


geissonator, 2017-03-08:
Our manufacturing team requires the openbmc to be in DHCP mode.  They are trying to enable it and hitting the below issue:

"It appears that the MAC address is correctly set, but that the system is broadcasting for an IP address and not supplying the MAC correctly.
The DHCP server sees a request and provides an IP, but the MAC associated with that IP address is wrong."

williamspatrick, 2017-03-08:
The issue is not that the BMC is sending the wrong MAC address.  The issue is that newer DHCP clients, including the one on openbmc, default to sending DUIDs per RFC 4361 instead of MAC.  We can likely change the configuration of our DHCP client to revert back to pre-RFC4361 config, but I'm not sure if that is the correct thing for us to do publicly.  We might need another option to satisfy IBM MFG long term..

closed 2017-03-09 by williamspatrick