openbmc/openbmc #1273: Failed to Fetch version date  with latest build [VERSION_ID="v1.99.3-45-g8e6b232"]

SrideviRamesh, 2017-03-08:
Issue:
Failed to fetch version date info after updating barreleye with latest BMC build

```
root@barreleye:~#  cat /etc/version
cat: can't open '/etc/version': No such file or directory
```

System Info:
```
cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-45"
VERSION_ID="v1.99.3-45-g8e6b232"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-45"
BUILD_ID="v1.99.3"
```
please find the attached FFDC, journal, dmesg, serial port logs
[serialport_logs.txt](https://github.com/openbmc/openbmc/files/826855/serialport_logs.txt)
[journal_barreleye_logs.txt](https://github.com/openbmc/openbmc/files/826856/journal_barreleye_logs.txt)
[dmesg.txt](https://github.com/openbmc/openbmc/files/826859/dmesg.txt)
[20170308001838178251_MyFFDCLogs.zip](https://github.com/openbmc/openbmc/files/826861/20170308001838178251_MyFFDCLogs.zip)

gkeishin, 2017-03-08:
This is not a bug.. Dev decided to remove this from driver build https://gerrit.openbmc-project.xyz/#/c/2818/

closed 2017-03-08 by gkeishin

sivassrr, 2017-03-08:
Is there a way to get the build date?

sivassrr, 2017-03-27:
To complete the story, dev decided to have /etc/timestamp for the same.