openbmc/openbmc #1274: Handle a BMC reset while Host is powered running

geissonator, 2017-03-08:
Similar to #1094 , need to ensure the host stays up and running when the BMC is rebooted. Need to update host state object correctly and get the systemd target, obmc-chassis-start.target started without having it disrupt the host.

geissonator, 2017-03-08:
- [ ] Write host IPMI heart beat interface
- [ ] Create new service to call IPMI heart beat and create r/r file indication if needed
- [ ] Update service files to look for r/r indication
- [ ] Update target files and enable function

geissonator, 2017-03-21:
#1322 is going to be a blocker to this function once enabled (on witherspoon at least).

GHF, 2017-04-20:
@geissonator this is the journal I pulled from a Zaius BMC that I rebooted while its host is booted to Linux. I built the BMC firmware at dd8a5477.
[reboot_bmc-host_up.txt](https://github.com/openbmc/openbmc/files/945659/reboot_bmc-host_up.txt)

This part suggests to me that the BMC successfully rebooted without powering down the host:
```
749 Apr 20 22:39:20 zaius systemd[1]: Started Check Chassis0 pgood on reset.         
750 Apr 20 22:39:20 zaius systemd[1]: Reached target Chassis0 power on after reset.  
751 Apr 20 22:39:20 zaius systemd[1]: Starting Create file to indicate chassis0 pgood is on...
752 Apr 20 22:39:20 zaius systemd[1]: Started Create file to indicate chassis0 pgood is on.
753 Apr 20 22:39:21 zaius system_manager.py[890]: Running System State: HOST_BOOTED  
754 Apr 20 22:39:21 zaius discover_system_state.py[961]: Traceback (most recent call last):
755 Apr 20 22:39:21 zaius systemd[1]: Started Start chassis0 on after BMC reset.     
756 Apr 20 22:39:21 zaius systemd[1]: Reached target Chassis0 (Reset Check).
```

In fact, the host power is still on even when we hit `obmc-quiesce-host@0.target`, according to the power monitor:
```
root@zaius:~# grep "" /sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/*_input
/sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/in10_input:1107
/sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/in11_input:1103
/sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/in12_input:942
/sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/in13_input:910
/sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/in1_input:12165
/sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/in2_input:5051
/sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/in3_input:3314
/sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/in4_input:1806
/sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/in5_input:1054
/sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/in6_input:1006
/sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/in7_input:830
/sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/in8_input:830
/sys/bus/i2c/drivers/ucd9000/0-0064/hwmon/hwmon5/in9_input:1505
```
The host is definitely hanged at that point, however.

On the host serial console, the watchdog reset job complained before the power went down:
```
[  650.603068] IPMI Watchdog: response: Error ce on cmd 22
[  661.817536] IPMI Watchdog: response: Error ce on cmd 22
[  673.189008] IPMI Watchdog: response: Error ce on cmd 22
[  684.229373] IPMI Watchdog: response: Error ce on cmd 22
[  695.273336] IPMI Watchdog: response: Error ce on cmd 22
[  706.331616] IPMI Watchdog: response: Error ce on cmd 22
```

geissonator, 2017-04-21:
Oops, @GHF , I must have missed this one, or it's new, meta-openbmc-machines/meta-openpower/meta-ingrasys/meta-zaius/recipes-phosphor/chassis/refclock-fixup/op-refclock-fixup@.service  doesn't have the line to say "do nothing if the power-already-on file created".  You'll see this in the other files like avsbus-disable@.service

Need to add this line (and to any other services associated with power on that may be Zaius specific).
ConditionPathExists=!/run/openbmc/chassis@%i-on

GHF, 2017-04-21:
@geissonator Oh yeah I didn't even see that it was running upon reboot. Running that will drop the clock for some interval, so I could see how that would cause a problem.

Thanks!

closed 2017-05-10 by williamspatrick

rfrandse, 2017-06-08:
<https://gerrit.openbmc-project.xyz/4647> meta-zaius: Don't run refclock fixup if host is up
Fixes  #1274 Handle a BMC reset while Host is running

kyle-ibm, 2017-08-21:
@williamspatrick 
updated witherspoon wsp001 to latest FW based on 
http://pfd.austin.ibm.com/releasenotes/openpower9/OP910/OP910_1734A/OP910_1734B_RelNote_Main.html

root@witherspoon:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x0280e000..0x0280f000 !
[                                                  ] 0%IBM-witherspoon-redbud-ibm-OP9_v1.18_1.37
	op-build-v1.18-71-g4b4f764-dirty
	buildroot-2017.05-163-g8035745
	skiboot-v5.7-92-g0818df625e35-pac242dd
	hostboot-42ecd45
	linux-4.12.5-openpower1-pdf91abd
	petitboot-v1.5.1-p6a468eb
	machine-xml-9a28007-p02f99d9
	occ-71b5f68-p136ce26
	hostboot-binaries-cf51511
	capp-ucode-9c73e9f
	sbe-bcb079b
[==================================================] 100%

[kloh@lcb328 ~/scripts]$ ./restcmd -H wsp001 -u /xyz/openbmc_project/software/enumerate -o GET
  "data": {
    "/xyz/openbmc_project/software/7b61bb25": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
      "Path": "", 
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC", 
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None", 
      "Version": "ibm-v1.99.8-98-g06734f31-r4-0-gd8ea5a3"
    }
  }, 

Booted system into Host OS (tried with both Ubuntu and RHEL)
issued reboot command in wsp001 bmc.
System goes from Running, into powered off state
[kloh@lcb328 ~/scripts]$ ./restcmd -H wsp001 -o state
BMC Power State       : Ready
Chassis Power State   : On
Host Power State      : Running
[kloh@lcb328 ~/scripts]$ ssh root@wsp001
Warning: Permanently added 'wsp001,9.3.20.191' (RSA) to the list of known hosts.
root@wsp001's password: 
root@witherspoon:~# 
root@witherspoon:~# reboot
Connection to wsp001 closed by remote host.
Connection to wsp001 closed.
[kloh@lcb328 ~/scripts]$ ./restcmd -H wsp001 -o state
BMC Power State       : Ready
Chassis Power State   : Off
Host Power State      : Off

kyle-ibm, 2017-08-21:
RHEL:
OS Version: Red Hat Enterprise Linux Server release 7.4 Beta (Pegas)
Kernel Version: 4.11.0-16.el7a.ppc64le
  
Ubuntu:
OS Version: Ubuntu 16.04.3 LTS  
Kernel Version: 4.11.0-041100rc4-generic

kyle-ibm, 2017-08-22:
chatted with @geissonator . seems like the fixes have not made it into ibm-v1.99.8-98-g06734f31-r4-0-gd8ea5a3 . Will wait for the next release.