openbmc/openbmc #1279: witherspoon - no hostboot output to serial port

cdeadmin, 2017-03-08:


williamspatrick, 2017-03-08:
@jk-ozlabs - I don't believe we currently support mirroring to the physical UART.  Do you agree?  I think this is a minor new feature for obmc-console, or maybe it is already supported and we just don't have it configured right?

shenki, 2017-03-10:
Zaius has this feature enabled. We should be able to copy the config from that machine to Witherspoon.

@mine260309 Does Romulus have a rear UART? We should make the change to all P9 machines at once

amboar, 2017-03-10:
Which tty should the data be routed to? ttyS0?

From [`obmc-console.conf.sample`](https://github.com/openbmc/obmc-console/blob/master/obmc-console.conf.sample#L5):

```
# To mirror to a local tty device (typically a hardware UART), set local-tty
# local-tty = ttyS0
```

As @shenki mentioned this has already been done for Zaius: https://github.com/openbmc/openbmc/commit/cd021293f1264ace3b0833e63a3f9b4a20fd1ef4

mine260309, 2017-03-10:
@shenki As my understanding, OpenBMC uses VUART by default, and hostboot console is reached by `obmc-console-client` or `ssh -p 2200`.
So I think Romulus does not need this change, unless @williamspatrick agrees to redirect the host console to the physical UART.

williamspatrick, 2017-03-10:
@mine260309 - If Romulus has a physical UART it probably should do this mirroring as well.

williamspatrick, 2017-03-10:
@amboar - Your change doesn't seem to work on the system we attempted.  It also appears that ttyS0 is defaulting to 9600 baud, but even setting the serial server to that we do not get any characters.

williamspatrick, 2017-03-10:
Just so we are clear, we edited the conf file to match what you provided in your patch and then restarted the obmc-console@ttyVUART0 service.  I confirmed that the obmc-console server was using ttyS0 by looking at /proc/<pid>/fd.

mine260309, 2017-03-13:
@williamspatrick @shenki Romulus does have physical UART.
So I will try to add similar change to Romulus and test it.

amboar, 2017-03-14:
> @amboar - Your change doesn't seem to work on the system we attempted. It also appears that ttyS0 is defaulting to 9600 baud, but even setting the serial server to that we do not get any characters.

@williamspatrick I take it you verified the host was booting via output on the ssh console? We're not doing anything funny with the UART routing, so something else must be at fault. Maybe hook up a scope and verify we're seeing data on the wire? I guess we could also strace to check we're writing to ttyS0 from the software side.

mine260309, 2017-03-23:
I tried to mirror VUART to ttyS0/physical UART on Romulus, with the serials of patches:
https://gerrit.openbmc-project.xyz/#/c/2316
...
https://gerrit.openbmc-project.xyz/#/c/2322

The host console works fine on both `obmc-console-client` and physical UART.
The limitation for now is that the UART shall be configured to `9600 8N1`, which is super slow, and `Backspace` or `Ctrl-H` key does not work, it just prints `^H` in the physical UART console.

mine260309, 2017-03-24:
Romulus kernel patches are merged:
https://patchwork.ozlabs.org/patch/742516/
https://patchwork.ozlabs.org/patch/742518/

I will wait for the above serial of patches to be merged, and then I'll make the change on `meta-romulus` to mirror host console to UART1. Without the above patches, mirroring host console to UART1 causes a lot of error logs in journal.

paulgrobertson, 2017-04-13:
We also see NO serial port at runtime.   I connected minicom on laptop to minicom running on ttyS0 on witherspoon / ubuntu.  I should be able to echo keystokes back an forth.   ....nothing.     Is this related to this issue,  or a seperate problem?

williamspatrick, 2017-04-14:
@paulgrobertson - Yes, that is this issue.  There is also a CQ defect internal at IBM to track it.

closed 2017-04-26 by williamspatrick