openbmc/openbmc #1280: Net-DHCP: Assess DUID vs MAC.

williamspatrick, 2017-03-09:
See openbmc/openbmc#1272 for background.

Traditional DHCP clients will send the MAC address in the DHCP request and DHCP servers are often populated with a database of MAC<->DesiredIP.  With IPv6, DHCP introduced a non-MAC based identifier called the DUID.  The MAC vs DUID go into the 'client ID' portion of the DHCP request.

The intention of the DUID is so that a "system" can have multiple network devices, but the DHCP server can identify the various DHCP requests as belonging to that single system.  This allows device bonding, failover, load balancing, etc. to be done somewhat automatically.  (Consider a server with two network adapters or a laptop with an ethernet and wifi device.)  The DUID support was also backported to IPv4 via RFC 4361.

The current version of systemd we are running on openbmc (v229) does not allow configuration of the DUID itself.  It only allows switching between DUID or MAC-based client IDs.  Later versions of systemd allow switching IPv4 between DUID and MAC and they also allow switching the DUID between a few different types: link-layer, link-layer-timestamp, and uuid.  uuid is the default and is created by a hash (of some sort) of the /etc/machine-id.  The link-layer should continue to be based on the MAC in some way.

What we need to do is determine what is the appropriate behavior for a BMC with respect to client-ids and if we use DUID which type we should use.  Since a server often has a barcode, of some sort, with the MAC address it seems like there might be strong desire to use `ClientIdentifier=mac` for IPv4 and `DUIDType=link-layer` for IPv6.  This would circumvent any automatic behavior by the DHCP server if the BMC did have two (redundant) ethernet devices.  An alternative might be to have system manufacturers print, or otherwise distribute, the generated machine-id (and/or uuid-based client id) instead of or in addition to the MAC.

Another aspect to consider is the behavior of u-boot.  In doing background reading, I have seen reported issues where u-boot uses a MAC and then Linux uses a DUID and this causes the BMC to receive two different IP addresses between u-boot (for netboot) and Linux.  Since we do not netboot in production this may not be a serious consideration.

Once we determine the appropriate behavior we will need to update our systemd-networkd configuration files to match.

williamspatrick, 2017-03-09:
A short-term fix for openbmc/openbmc#1272 was added to Gerrit here:
    https://gerrit.openbmc-project.xyz/#/c/2933/

This ignores IPv6 and forces IPv4 to use 'mac' for client-id.

hramasub, 2017-05-11:
RFC 6355 defines the usage of DUID-UUID over and above DUID-LLT, DUID-EN, DUID-LL defined in RFC 3315. Accordingly there are 4 different options for constructing DUID.

_What is the appropriate behavior for a BMC with respect to client-ids and if we use DUID which type we should use ?_

**The following are likely the desirable properties of IP address assigned to servers which impact the IP address management (IPAM)** :

1.  Durability requirements on IP address:
     1. persistent across boots.
     2. persistent across hardware repairs.
     3. persistent across firmware flash/updates.
     4. restore default upon factory reset.
2. Spatial association of IP address:
     1. Location (rack/shelf/chassis) based IP address allocation
3. Redundancy group association:
     1. Same-subnet addresses for servers belonging to a cluster.
4. User association:
     1. Same-subnet addresses for servers in a farm (baremetal servers allocated to a client)

Requirements 2,3,4 above are controlled by the DHCPv6 service deployment architecture (DHCPv6 servers and DHCPv6 relay agents) in the data center.

Requirements 1.1, 1.2, 1.3, 1.4 pertain to the usage of DUID for IP address allocation.

**Comparsion of DUID types for durability constraints**

----------------------------------------------------------------------------------------------------------------
DUID Type     | Boot Persistence | Repair Persistence | Update Persistence | Recreate on Factory Reset
----------------------------------------------------------------------------------------------------------------
  DUID-LLT     |             Y              |               N             |                Y               |                       Y
  DUID-EN      |             Y              |               Y             |                 Y              |                        Y   <<<===
  DUID-LL       |             Y              |               N             |                 Y             |                        Y
 DUID-UUID  |              Y              |               Y             |                  Y             |                        N

DUID-LL and DUID_LLT are liable to change due to FRU replacement resulting in a change in the MAC address.
DUID-UUID can not recreate the same ID upon factory reset due to the inherent randomness in its creation.
DUID-EN is thus the DUID type which reliably satisfies all the durability constraints.

**DUID provisioning use cases**
All OpenBMC systems are automatically provisioned with a random DUID-UUID at the first boot and this ID is persisted in the file /etc/machine-id. Subsequently, Manufacturing shall provision a Genesis / Factory setting DUID-EN. The DUID-EN is generated using the IANA ID of the Vendor and the MTM and Serial number of the system. Once deployed in the field, system administrator can modify
1. Default DUID -- systemd generated 
2. Genesis / Factory setting -- DUID-EN based.
3. User configuration -- Any of the DUID scheme

**Manufacturing network requirements**
The DHCP servers in the Manufacturing network must not derive the mac address of the client from the DUID.

**OpenBMC REST Interfaces**
1. GET the DUID of the system.
2. PATCH the DUID type and DUID string of the system.
3. POST request to FactoryReset the system.

**References**
[https://tools.ietf.org/html/rfc6355](url)
[https://tools.ietf.org/html/rfc4122](url)
[https://tools.ietf.org/html/rfc6866](url)
[https://github.com/systemd/systemd/issues/394](url)
[https://blogs.technet.microsoft.com/privatecloud/2012/03/22/dhcp-in-a-cloud-enabled-world-by-erik-lofstrand/](url)
[https://sourceforge.net/p/xcat/mailman/xcat-commit/?page=1](url)
[http://docs.oracle.com/cd/E19253-01/816-4554/eyatl/index.html](url)
[http://h20564.www2.hpe.com/hpsc/doc/public/display?docId=c04398276](url)
[http://www.cisco.com/c/en/us/products/collateral/ios-nx-os-software/enterprise-ipv6-solution/whitepaper_c11-689821.pdf](url)

geissonator, 2017-10-13:
@dkodihal - please asses phase for this one

ratagupt, 2019-02-26:
https://github.com/ibm-openbmc/dev/issues/389

truehomie, 2019-06-04:
wassup folks, how i can change/renew/refresh/reset/modify my DUID client DHCPv6 ??? after i changed my MAC address is consists of two parts :  my new MAC address + my old MAC address...

closed 2020-01-03 by stale[bot]