openbmc/openbmc #1286: Incorporate new soft power off function into host state

anoo1, 2017-03-09:


JoshDKing, 2017-03-29:
*   Move op-stop-host@.service under the obmc-chassis-stop@0.target
*   Make sure this service will hold off all other power off services until it’s done

sivassrr, 2017-04-19:
@anoo1 @JoshDKing  Do FVT need to change any TCs related to Power off. Or it is internal change only?

JoshDKing, 2017-04-19:
This change will just be updating the targets so the change is internal.

geissonator, 2017-05-11:
Need to get this in first - https://gerrit.openbmc-project.xyz/#/c/4032/

Then just add the application to the shutdown target and update phosphor host state code to use the shutdown target but to monitor for the host-off target.

rfrandse, 2017-05-18:
https://gerrit.openbmc-project.xyz/4224 Use shutdown target to stop the host
Resolves #1286 Incorporate new soft power off function into host state

williamspatrick, 2017-05-24:
Code was reverted, reopening.

closed 2017-06-06 by williamspatrick