openbmc/openbmc #1287: [BarrelEye v1.99.3-48] Looks like completely unstable build: 1. Impossible booting host. 2. Problems with Flash-reading. 3. CPU up to 100% busy all the time. 4. ipmid (ipmi-host) fails? or "obmcutil poweron" fails? 5. The service system_manager.py restarts in cycle.

yadron, 2017-03-09:
[BarrelEye v1.99.3-48] Looks like completely unstable build:

1. Impossible booting host.

2. Problems with Flash-reading and with ./update to the other build.

3. CPU up to 100% busy all the time.

4. ipmid (ipmi-host) fails? or "obmcutil poweron" fails?

5. The service system_manager.py restarts in cycle.

More details here:
https://github.com/openbmc/openbmc/issues/1262
(because I think that it is the same for Palmetto).

Note that I've built it from scratch see https://openpowerblog.wordpress.com/2016/04/23/how-to-build-openbmc-binary-for-barreleye/
(e.g.
```
rm -Rf ./
git clone https://github.com/openbmc/openbmc
cd openbmc
TEMPLATECONF=meta-openbmc-machines/meta-openpower/meta-rackspace/meta-barreleye/conf . oe-init-build-env
bitbake obmc-phosphor-image
```
),
and I've rebuild it once for ./update again before any debugging.
So checked twice, and sure to be buggy.

I understand that this is only for this build and prev and next should be better (it is normal when some builds during development completely unstable).

williamspatrick, 2017-03-09:
@shenki @amboar - It feels like we have some GPIO issues introduced recently.  Have there been any code changes in the kernel there?  @geissonator is also seeing lots of trouble with the pgood GPIO never coming on, even though power is in-fact on.  He's seen it both on Barreleye and Witherspoon.

geissonator, 2017-03-09:
#832 is the issue I've been digging into

amboar, 2017-03-09:
@williamspatrick the only major change to the GPIO subsystem is adding support for the additional banks. That affects userspace with the change in base number, but that should now be properly derived by userspace rather than hard-coded.

Possibly it's a pinmux problem, but we'd need more logging for evidence that it *never* works, rather than just suggestions that it might not work.

shenki, 2017-03-10:
I haven't seen any GPIO issues. I've been using master to boot a plametto on a daily basis for the past week or two and it has worked.

adamliyi, 2017-03-10:
@williamspatrick , @shenki : not sure this issue has the same cause of issue#1262 - corrupted PNOR make the booting fail. Flashing PNOR can solve the issue.
I've been told that the PNOR may be corrupted if suddenly power off while booting, since the flash may be written during booting.

yadron, 2017-03-10:
Yes, I think that PNOR/flashcp works Ok for my BarrelEye. So my issue different from https://github.com/openbmc/openbmc/issues/1262 , but similar in mean that cannot boot host and both some problems with flash memory (with SQUASHFS, see below).

What is PNOR? Is it the same as the BMC flash updated by ./update that is flashcp or other?
I can see https://github.com/open-power/pnor , but not sure yet.

The main question for me. Why did I get these errors at Barreleye for just reading some files from /usr/sbin/ of BMC:
```
SQUASHFS error: Unable to read page, block c872c, size bcb4
SQUASHFS error: Unable to read data cache entry [c872c]
```
(_probably means cannot read from Flash FS and both flashcp failed with verification during ./update then_)
with "v1.99.3-48" and no similar "SQUASHFS" errors when reverted back to "v1.99.3-dirty".

For me it looks like if something broken with BMC services (for example service "system_manager.py" restarts in cycle all the time), then ipmid-host becomes broken too (because it cannot fetch some info via DBUS from system_manager.py), then host cannot be poweron (because ipmid-host broken), and the worse that overall BMC system becomes unstable (CPU 100% all the time) and "SQUASHFS" errors (some system or kernel unknown error with Flash FS when system 100% busy?) and for user it could be "send the server for refurbish" problem (because ./update doesn't work anymore too).

For me it looks like an Architecture concerns that OpenBMC should be more reliable, than it now:

1. host need to boot even if DBUS services failed (for example, the critical for poweron functions of IPMID-host should be inside IPMID-host service, so no dependence of IPMID on DBUS anymore).

2. The ./update for BMC image should work even if system 100% busy (because with v1.99.3-dirty it was already about 50-70% CPU busy most of the time, so if CPU 100% busy as I seen in v1.99.3-48, then it should not make SQUASHFS and flashcp unstable that I see in v1.99.3-48, so need to debug this case why verification during update by flashcp on v1.99.3-48 100% CPU busy system failed or some workaround could be: when ./update executed, stop services before running flashcp).

3. Need to check what happens if different services of DBUS failed. Need to be sure that no CPU-busy cycles when some service failed. Need to check most CPU-hungry services adding some sleep to it or even doing some refactoring for making it not so CPU-hungry.

4. Need to check which service depends on which one. Are there any architectural schema of OpenBMC/DBUS-services that shows which one should start first and which next? Now it looks like sometimes system works (if services started in this order) and sometimes not (if in other order) and DBUS doesn't give any warranty that services will start in this order (and if one service failed, then it is being restarted automatically, but depended services not, so BMC can become unstable after one service failed).

yadron, 2017-03-10:
For "v1.99.3-48". I've changed slp back to "obmc-host-state-mgmt, obmc-chassis-state-mgmt, obmc-bmc-state-mgmt", but it didn't help. Many things fixed by this (like ipmid started Ok, and no system_manager.py restarting), but cannot boot host. Logs:

```
root@barreleye:~# obmcutil bootprogress
units = 
value = Off
error = 0
```

And the BCM still being stalled for some seconds for no reason (like v.1.99

```
root@barreleye:~# obmcutil power           
pgood = 1
state = 1
pgood_timeout = 10
```

And this suppose to be Ok state (looking to https://github.com/openbmc/openbmc/issues/832 ), but still cannot boot host:

```
root@barreleye:~# obmcutil power           
pgood = 1
state = 1
pgood_timeout = 10
root@barreleye:~# ps|grep mapper
  724 root       1:55 python /usr/sbin/phosphor-mapper
 1292 root       0:00 grep mapper
root@barreleye:~# obmcutil poweron
root@barreleye:~# ps|grep state-ma
  809 root       0:00 /usr/sbin/phosphor-host-state-manager
  810 root       0:00 /usr/sbin/phosphor-bmc-state-manager
  818 root       0:00 /usr/sbin/phosphor-chassis-state-manager
 1295 root       0:00 grep state-ma
root@barreleye:~# obmcutil bootprogress
units = 
value = Off
error = 0
root@barreleye:~# ps|grep power_co
  824 root       0:01 /usr/sbin/power_control.exe 3000 10
 1362 root       0:00 grep power_co
root@barreleye:~# obmcutil power
pgood = 1
state = 1
pgood_timeout = 10
root@barreleye:~# cat /sys/class/gpio/gpio315/active_low
0
root@barreleye:~# cat /sys/class/gpio/gpio315/direction
in
root@barreleye:~# cat /sys/class/gpio/gpio315/edge
none
root@barreleye:~# cat /sys/class/gpio/gpio315/value

1
root@barreleye:~#

```

When poweron (unsuccessfull), services saying:

```
  op-power-start@0.service                   loaded active     exited          Start Power0
  op-start-host@0.service                    loaded inactive   dead      start Start Host0
  op-wait-power-on@0.service                 loaded activating start     start Wait for Power0 to turn on
```

And some logs from "journalctl" :

...
```
[... 86 lines cut ...]
...
Mar 10 16:17:49 barreleye power_control.exe[824]: PowerControl: pgood: 1, setting reset USB_RESET to 1
Mar 10 16:17:49 barreleye system_manager.py[752]: Running System State: HOST_POWERED_ON
Mar 10 16:17:49 barreleye timemanager[872]:  PGOOD has changed..
Mar 10 16:17:49 barreleye timemanager[872]: Changing time settings is *deferred* now
Mar 10 16:17:49 barreleye timemanager[872]:  User Settings have changed..
Mar 10 16:17:49 barreleye systemd[1]: Started Wait for Power0 to turn on.
...
Mar 10 16:17:50 barreleye systemd[1]: Reached target Chassis0 (Power On).
Mar 10 16:17:50 barreleye systemd[1]: Reached target Power0 On.
Mar 10 16:17:50 barreleye systemd[1]: Reached target Power0 (On).
Mar 10 16:17:50 barreleye systemd[1]: Starting Start Host0...
Mar 10 16:17:50 barreleye systemd[1]: Starting PCIE Slot Presence Detection For Chassis0...
Mar 10 16:17:50 barreleye phosphor-chassis-state-manager[818]: Recieved signal that power ON is complete
...
Mar 10 16:17:50 barreleye phosphor-chassis-state-manager[818]: Change to Chassis Power State
Mar 10 16:17:50 barreleye control_host.exe[827]: Booting host
Mar 10 16:17:50 barreleye systemd[1]: Started Start Host0.
Mar 10 16:17:50 barreleye systemd[1]: Reached target Start Host0.
Mar 10 16:17:50 barreleye systemd[1]: Reached target Host0 (Started).
Mar 10 16:17:50 barreleye systemd[1]: Starting Start Watchdog0...

```",

but host not booted! Why?

geissonator, 2017-03-10:
That latest fail above just looks like your PNOR needs a reflash (i.e. you got pgood but host didn't start).  9 times out of 10 that's a bad PNOR.  Try one from https://openpower.xyz/job/openpower-op-build/

The pgood issue seems to cause a PNOR corruption when it hits so we have to reflash PNOR a lot.

yadron, 2017-03-13:
Yes, updating .pnor helped booting the host (note that ver v1.99.3-dirty worked well with my older .pnor, like I tried revering from v1.99.3-48 to v1.99.3-dirty and then host booted without any .pnor upgrade, so the .pnor flash itself probably was Ok). It looks like something changed both in OpenBMC and in .pnor, so upgrade from v1.99.3-dirty to v1.99.3-48 required both upgrading .pnor to https://openpower.xyz/job/openpower-op-build/lastStableBuild/distro=ubuntu,target=barreleye/ (file barreleye.pnor) .  I've run these commands (as after debugging/writing https://github.com/openbmc/openbmc/issues/1262 ):

1. Upgraded OpenBMC to v1.99.3-48 (with reverting back "slp" to "obmc-host-state-mgmt, obmc-chassis-state-mgmt, obmc-bmc-state-mgmt", because slp is too CPU-hungry now and both service "/usr/sbin/phosphor-host-state-manager" missed for some other and it was not executed well in original OpenBMC v1.99.3-48 configuration that was with "slp" instead).

2. root@barreleye:~# pflash -E -f -i -p /tmp/barreleye.pnor
(the file barreleye.pnor from https://openpower.xyz/job/openpower-op-build/lastStableBuild/distro=ubuntu,target=barreleye/).

3. reboot

4. BMC rebooted Ok, but host not booted yet, and the output after BMC booted is (so still the bug that host needs manual boot after BMC rebooted):

```
root@barreleye:~# obmcutil power
pgood = 1
state = 1
pgood_timeout = 10
root@barreleye:~# obmcutil bootprogress
units = 
value = FW Progress, Starting OS
error = 0
```

5. Executed then (for manual host booting):

```
root@barreleye:~# obmcutil poweroff
root@barreleye:~# obmcutil poweron
```

and then host booted well (and "obmc-console-client" works), the output of obmc-console-client:

```
root@barreleye:~# obmc-console-client 
  3.32547|ISTEP  6. 3
  3.69572|ISTEP  6. 4
  3.69628|ISTEP  6. 5
  8.90456|HWAS|PRESENT> DIMM[03]=A0A0A0A0A0A0A0A0
  8.90457|HWAS|PRESENT> Membuf[04]=CCCC000000000000
  8.90457|HWAS|PRESENT> Proc[05]=C000000000000000
  9.08600|ISTEP  6. 6
...
[   72.389742998,5] SkiBoot skiboot-5.4.3 starting...
...
 Petitboot (v1.4.1-pb89a0fa)                         1A42BKT00-600-G 65D0001X02
...
```
and IPMID commands executed during booting (that is the output of "/usr/sbin/ipmid"):

```
Handling Netfn:[0x6], Cmd:[0x36]
IPMI SET ACPI STATE Ignoring for now
Failed to call the method: Connection timed out
Dest: :1.15, Path: /org/openbmc/HostIpmi/1
WATCHDOG SET Timer:[0x4B0] 100ms intervals
Failed to call the method: Input/output error
Dest: :1.15, Path: /org/openbmc/HostIpmi/1
WATCHDOG RESET
Failed to call the method: Input/output error
Dest: :1.15, Path: /org/openbmc/HostIpmi/1
IPMI SET_SENSOR [0x05]
Failed to call the method: Input/output error
Dest: :1.15, Path: /org/openbmc/HostIpmi/1
WATCHDOG RESET
Failed to call the method: Input/output error
Dest: :1.15, Path: /org/openbmc/HostIpmi/1
WATCHDOG RESET
Failed to call the method: Input/output error
Dest: :1.15, Path: /org/openbmc/HostIpmi/1
fru data size:[168], area offset:[8], area_size:[248]
Incomplete Fru file.. Size:[168]
[... 1139 lines cut ...]
IPMID boot option property get: {default}.
IPMID boot option property get: {ONETIME}.
Failed to call the method: Input/output error
Dest: :1.15, Path: /org/openbmc/HostIpmi/1
BootPolicy is[ONETIME]IPMI GET_SYS_BOOT_OPTIONS
IPMID boot option property get: {ipaddress=,prefix=,gateway=,mac=,addr_type=}.
Configuration String[ipaddress=,prefix=,gateway=,mac=,addr_type=]
 host_config.macaddress.c_str()=[]
sscanf Failed in extracting mac address.
getHostNetworkData failed for get_sys_boot_options.
ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Failed to call the method: Input/output error
Dest: :1.15, Path: /org/openbmc/HostIpmi/1
IPMI Handling GET-SEL-TIME
Host Time read:[Mon Mar 13 09:30:22 2017 GMT] :: [1489397422]
Failed to call the method: Input/output error
Dest: :1.15, Path: /org/openbmc/HostIpmi/1
IPMI Handling GET-SEL-TIME
Host Time read:[Mon Mar 13 09:30:22 2017 GMT] :: [1489397422]
Failed to call the method: Input/output error
Dest: :1.15, Path: /org/openbmc/HostIpmi/1
IPMI GET DEVICE GUID
Failed to call the method: Input/output error
Dest: :1.15, Path: /org/openbmc/HostIpmi/1
```

yadron, 2017-03-16:
@geissonator , will comparing logs (below for non-working and then working host boot) help you with debugging https://github.com/openbmc/openbmc/issues/832 issue?

After updating bmc-image to the same one (v1.99.3-48) that worked after .pnor update for my Barreleye (I did updating bmc-image again for removing any changes made by me, see msg above), cannot boot host still and power_control does something, but not working, so see output below (so power_control changing PIN value every few seconds and updating timeout=50 or timeout=500 didn't help too; and see that PIN_VALUE=0 means start booting and PIN_VALUE=1 means shutdown):

```
root@barreleye:~# /usr/sbin/power_control.exe_ 3000 50  
Power GPIO latch output: 
Power GPIO power good input: PGOOD
Power GPIO 1 power_up outputs
Power GPIO power_up[0] = POWER_PIN active LOW
Power GPIO 1 reset outputs
Power GPIO reset[0] = USB_RESET active LOW
Power GPIO 0 pci reset outputs
FSI DATA GPIO: FSI_DATA
FSI CLK GPIO: FSI_CLK
FSI ENABLE GPIO: FSI_ENABLE
CRONUS SEL GPIO: CRONUS_SEL
Hostctl GPIO optionals: 2
Hostctl optional GPIO[0] = BMC_THROTTLE active HIGH
Hostctl optional GPIO[1] = IDBTN active LOW
GPIO Lookup:  PGOOD = 315,in
GPIO Lookup:  POWER_PIN = 325,out
GPIO Lookup:  USB_RESET = 306,out
Pgood state: 1

Power already at requested state: 1
PowerControl: setting power up POWER_PIN to 1
PowerControl: pgood: 0, setting reset USB_RESET to 0

PowerControl: setting power up POWER_PIN to 0
PowerControl: pgood: 1, setting reset USB_RESET to 1
PowerControl: setting power up POWER_PIN to 1
PowerControl: pgood: 0, setting reset USB_RESET to 0
PowerControl: setting power up POWER_PIN to 0
PowerControl: pgood: 1, setting reset USB_RESET to 1
PowerControl: setting power up POWER_PIN to 1
PowerControl: pgood: 0, setting reset USB_RESET to 0
PowerControl: setting power up POWER_PIN to 0
PowerControl: pgood: 1, setting reset USB_RESET to 1
PowerControl: setting power up POWER_PIN to 1
PowerControl: pgood: 0, setting reset USB_RESET to 0
PowerControl: setting power up POWER_PIN to 0
PowerControl: pgood: 1, setting reset USB_RESET to 1
PowerControl: setting power up POWER_PIN to 1
PowerControl: pgood: 0, setting reset USB_RESET to 0
PowerControl: setting power up POWER_PIN to 0
PowerControl: pgood: 1, setting reset USB_RESET to 1
PowerControl: setting power up POWER_PIN to 1
PowerControl: pgood: 0, setting reset USB_RESET to 0
PowerControl: setting power up POWER_PIN to 0
PowerControl: pgood: 1, setting reset USB_RESET to 1
PowerControl: setting power up POWER_PIN to 1
PowerControl: pgood: 0, setting reset USB_RESET to 0
PowerControl: setting power up POWER_PIN to 0
PowerControl: pgood: 1, setting reset USB_RESET to 1

```

Logs:

```
root@barreleye:~# journalctl |grep GPIO
Mar 16 08:27:19 barreleye host_checkstop.exe[813]: GPIO Lookup:  CHECKSTOP = 417,falling
Mar 16 08:27:20 barreleye system_manager.py[762]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('USB_RESET', False)], [], 'FSI_DATA', 'FSI_CLK', 'FSI_ENABLE', 'CRONUS_SEL', [('BMC_THROTTLE', True), ('IDBTN', False)]]
Mar 16 08:27:21 barreleye button_power.exe[810]: GPIO Lookup:  POWER_BUTTON = 324,both
Mar 16 08:27:22 barreleye button_reset.exe[811]: GPIO Lookup:  RESET_BUTTON = 326,both
Mar 16 08:27:22 barreleye power_control.exe[820]: Power GPIO latch output:x crc err
Mar 16 08:27:22 barreleye power_control.exe[820]: Power GPIO power good input: PGOODo long
Mar 16 08:27:22 barreleye power_control.exe[820]: Power GPIO 1 power_up outputs
Mar 16 08:27:22 barreleye power_control.exe[820]: Power GPIO power_up[0] = POWER_PIN active LOW
[... 837 lines cut ...]
Mar 16 09:37:06 barreleye systemd[1]: org.openbmc.control.Power@0.service: Service hold-off time over, scheduling restart.
Mar 16 09:37:06 barreleye systemd[1]: Stopped Phosphor Power0 Control.
Mar 16 09:37:06 barreleye systemd[1]: org.openbmc.control.Power@0.service: Start request repeated too quickly.
Mar 16 09:37:06 barreleye systemd[1]: Failed to start Phosphor Power0 Control.
Mar 16 09:37:25 barreleye system_manager.py[756]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('USB_RESET', False)], [], 'FSI_DATA', 'FSI_CLK', 'FSI_ENABLE', 'CRONUS_SEL', [('BMC_THROTTLE', True), ('IDBTN', False)]]
Mar 16 09:37:43 barreleye systemd[1]: Started Stop Power0.
Mar 16 09:37:43 barreleye systemd[1]: Starting Wait for Power0 to turn off...
Mar 16 09:37:47 barreleye systemd[1]: Started Wait for Power0 to turn off.
Mar 16 09:37:47 barreleye systemd[1]: Reached target Chassis0 (Power Off).
Mar 16 09:37:47 barreleye phosphor-chassis-state-manager[812]: Change to Chassis Power State
Mar 16 09:37:47 barreleye systemd[1]: Reached target Power0 Off.
Mar 16 09:38:05 barreleye systemd[1]: Starting Start Power0...
Mar 16 09:38:05 barreleye systemd[1]: Starting Wait for Power0 to turn on...
Mar 16 09:38:05 barreleye systemd[1]: Stopped target Power0 Off.
Mar 16 09:38:06 barreleye systemd[1]: Stopped target Chassis0 (Power Off).
Mar 16 09:38:06 barreleye phosphor-chassis-state-manager[812]: Change to Chassis Power State
Mar 16 09:38:06 barreleye systemd[1]: Stopped Stop Power0.
Mar 16 09:38:06 barreleye systemd[1]: Stopped Wait for Power0 to turn off.
Mar 16 09:38:06 barreleye systemd[1]: Started Start Power0.
Mar 16 09:38:08 barreleye systemd[1]: Started Wait for Power0 to turn on.
Mar 16 09:38:10 barreleye systemd[1]: Reached target Chassis0 (Power On).
Mar 16 09:38:10 barreleye phosphor-chassis-state-manager[812]: Change to Chassis Power State
Mar 16 09:38:15 barreleye systemd[1]: Reached target Power0 On.
Mar 16 09:38:15 barreleye systemd[1]: Reached target Power0 (On).
```

gtmills, 2018-07-05:
@yadron Still a problem? Or can this be closed? Thanks!

gtmills, 2018-07-18:
No updates in over a year, Barreleye is no longer a supported machine, and this is a very old build. Closing. Please reopen if still an issue. Thanks!

closed 2018-07-18 by gtmills