openbmc/openbmc #1288: led: Lamp Test

rfrandse, 2017-03-09:
For lamp test we just need to define a single lamp test group that turns on all the LEDs in the system.

sivassrr, 2017-03-10:
May know what are all interfaces it will be enabled, REST/IPMI, right. Any other?

vishwabmc, 2017-03-24:
@rfrandse Since this is not scheduled now, can this be removed from my name and assign when we schedule it ?

vishwabmc, 2017-03-28:
@rfrandse @williamspatrick what is the requirement here ?. We do not have the generated device tree being used to create physical LED so we can not test this until we get that in.

@spinler I am assuming the dev tree has the `hyphen` converted over to `underscore` ?

vishwabmc, 2017-04-12:
I am thinking there are 2 ways of doing it.
```
- [ 1 ] ( My vote for this )
* Create a group in MRW and name it LampTest
* Add all the BMC controlled FRUs that have LEDs on 'em

 Doing this way will give flexibility to the system designer on what LEDs they want tested as part of lamp test although I understand lamp test generally means light up all LEDs.

- [ 2 ]
 * Let the MRW parser look at all the FRUs that have LED on 'em and create a group.
 - But this takes away the flexibility.
```
--------------------------------
Open question : What about the LEDs that are not controlled by BMC ?. 

We do not have a application / infrastructure yet that would  let Host light these LEDs.
Also, the LEDs that are controlled directly by HW ( like HDD, PS) are not to be part of lamp test.

@spinler @williamspatrick let me know what you think.

alfordja, 2017-04-13:
Is Lamp Test a requirement for the web GUI?  i.e., do we need to include a user interface for turning all LEDs on/off?

vishwabmc, 2017-04-13:
@alfordja lamp test looks mostly a manufacturing requirement as opposed to field.  But I see it can also be executed at field. Let me get the manufacturing requirement from Venkatesh and update.

williamspatrick, 2017-04-14:
@alfordja - We should not give this feature in the webUI.

alfordja, 2017-04-17:
OK, thanks -- no lamp test GUI.....

vishwabmc, 2017-05-05:
- [ ] Update MRW parser to look for all LEDs that are driven off of BMC
- [ ] Generate LED group for that
- [ ] Integrate with LED manager
- [ ] Test it and make sure that all the LEDs BLINK as part of asserting this new group

vishwabmc, 2017-05-10:
Assert the group : /xyz/openbmc_project/led/groups/lamp_test

```
root@witherspoon:/sys/class/leds# cat rear_fault/brightness  rear_id/brightness rear_power/brightness 
0
0
0
root@witherspoon:/sys/class/leds# cat rear_fault/brightness  rear_id/brightness rear_power/brightness 
255
255
255
...

root@witherspoon:/sys/class/leds# cat rear_fault/brightness  rear_id/brightness rear_power/brightness 
0
0
0
root@witherspoon:/sys/class/leds# cat rear_fault/brightness  rear_id/brightness rear_power/brightness 
255
255
255
root@witherspoon:/sys/class/leds# cat rear_fault/brightness  rear_id/brightness rear_power/brightness 
255
255
255

root@witherspoon:/sys/class/leds# cat rear_fault/brightness  rear_id/brightness rear_power/brightness 
0
0
0
root@witherspoon:/sys/class/leds# cat rear_fault/brightness  rear_id/brightness rear_power/brightness 
0
0
0
...
```

closed 2017-05-17 by williamspatrick