openbmc/openbmc #1289: Kernel version

shenki, 2017-03-09:
When the kernel is built the version string is generated from a few variables, including the latest tag and the git SHA. eg. `4.7.10-00322-g3fd9c552f124`

However, when we build it in yocto the version string is forced to be <tag>-SHA. This SHA is not not meaningful without going to look up the git repository, so I would prefer we keep the normal version string.

This issue is to figure out how the string is being overridden and undo that.

amboar, 2017-03-10:
I would be interested in adding the `HEAD` commit `YYYYMMDD` to the version string (as well as the tag, patch count and short hash, though maybe it's getting ridiculous?): e.g. `4.7.10-00322-g3fd9c552f124-20170310`. Thoughts?

amboar, 2017-09-25:
Also `-dirty` if yocto has applied patches to the kernel tree.

shenki, 2019-04-29:
I would like this resolved. It is important for understanding how the kernel sources relate to the binary that is shipped, which is important for complying to the licensing requirements, and debugging of failures in the field. 

It requires some yocto hacking. Brad, are you able to help find someone to work on this?

mine260309, 2019-04-30:
I could take a look at this.

mine260309, 2019-04-30:
Yocto/OpenBMC does below to set the kernel version:
1. `linux-aspeed` sets `LINUX_VERSION_EXTENSION ?= "-${SRCREV}"`
2. `kernel-yocto.bbclass` has `do_kernel_configme()` to set:
```
echo "CONFIG_LOCALVERSION="\"${LINUX_VERSION_EXTENSION}\" >> ${B}/.config
```
   So in `.config` we have set `CONFIG_LOCALVERSION`
3. Then it is appended to kernel version string.

mine260309, 2019-04-30:
How about:
* If there is no yocto patch, keep current version string and make it short, like `5.0.7-b1b37a2`
* If there are yocto patches, use a string like `5.0.7-b1b37a2-dirty-84fa553`, where `dirty` tells it has local patch, and `84fa553` tells the local patch's revision.

@amboar It does not make senses to add `YYYYMMDD` in the version string, because kernel build already has the date info in `UTS_VERSION`, so when you run `uname -a` the date is shown in the output.

rfrandse, 2019-04-30:
<https://gerrit.openbmc-project.xyz/c/openbmc/meta-aspeed/+/21087> Improve kernel version string
Resolves  #1289 Kernel version

closed 2019-05-21 by geissonator