openbmc/openbmc #1296: ssh port 22 is intermittently going down

prkatti1, 2017-03-13:
We ran long duration ping to ssh port 22 for 12 hours, we sent 1 packet/second.
Noticed 1.6% packets loss, ssh port is intermittently going up & down.

Version info
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-59"
VERSION_ID="v1.99.3-59-ga782550"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-59"
BUILD_ID="v1.99.3"

prkatti1, 2017-03-13:
@sivassrr #1161 could be because of this

sivassrr, 2017-03-13:
@prkatti1 Please add your data here. That helps.

sivassrr, 2017-03-13:
REST also dropping the same way.

prkatti1, 2017-03-15:
Here is the sample output of long duration test
SENT (93654.8982s) TCP src_ip:1446 > dst_ip:22 S ttl=64 id=56808 iplen=40  seq=967766239 win=1480 
RCVD (93655.2833s) TCP dst_ip:22 > src_ip:1446 SA ttl=49 id=0 iplen=44  seq=2788051748 win=29200 <mss 1380>
SENT (93655.8992s) TCP src_ip:1446 > dst_ip:22 S ttl=64 id=56808 iplen=40  seq=967766239 win=1480 
RCVD (93656.2833s) TCP dst_ip:22 > src_ip:1446 SA ttl=49 id=0 iplen=44  seq=1390922926 win=29200 <mss 1380>
SENT (93656.9001s) TCP src_ip:1446 > dst_ip:22 S ttl=64 id=56808 iplen=40  seq=967766239 win=1480 
RCVD (93657.2833s) TCP dst_ip:22 > src_ip:1446 SA ttl=49 id=0 iplen=44  seq=2796409115 win=29200 <mss 1380>
SENT (93657.9012s) TCP src_ip:1446 > dst_ip:22 S ttl=64 id=56808 iplen=40  seq=967766239 win=1480 
RCVD (93658.2834s) TCP dst_ip:22 > src_ip:1446 SA ttl=49 id=0 iplen=44  seq=2193144971 win=29200 <mss 1380>
SENT (93658.9021s) TCP src_ip:1446 > dst_ip:22 S ttl=64 id=56808 iplen=40  seq=967766239 win=1480 
RCVD (93659.2834s) TCP dst_ip:22 > src_ip:1446 SA ttl=49 id=0 iplen=44  seq=2320478554 win=29200 <mss 1380>
SENT (93659.9025s) TCP src_ip:1446 > dst_ip:22 S ttl=64 id=56808 iplen=40  seq=967766239 win=1480 
RCVD (93660.2833s) TCP dst_ip:22 > src_ip:1446 SA ttl=49 id=0 iplen=44  seq=2970851060 win=29200 <mss 1380>
^C 
Max rtt: 3475.010ms | Min rtt: 0.050ms | Avg rtt: 336.555ms
Raw packets sent: 93007 (3.720MB) | Rcvd: 91263 (4.198MB) | Lost: 1744 (1.88%)
Nping done: 1 IP address pinged in 93660.68 seconds

williamspatrick, 2017-03-15:
Is there any journald content from this time interval?

sivassrr, 2017-03-20:
Team is recreating and explained to them not to reboot to avoid journalctl preserve logs issue. Will be collecting the journalctl periodically.

prkatti1, 2017-03-21:
Pls find attached journalctl logs.
[journallog.txt](https://github.com/openbmc/openbmc/files/857276/journallog.txt)

williamspatrick, 2017-03-21:
This is a palmetto system and the journal show power on / off activity while the test is done. Are you expecting this?

The NC-SI chip in Palmetto has a known hardware bug where the NC-SI is reset when the host issues a PCI reset. Therefore, we lose the network connection momentarily on every boot. 

Do you see this problem on Witherspoon or do you see the problem when the system is simply left at standby or at power on?

prkatti1, 2017-03-22:
Ran this same on BMC machine bareleye & on nonbmc machine which is on Austin, we sent 40000 packets to both machines and noticed 4 packets loss on nonbmc machine where as it is 790 packets on bmc.

Packet loss summary on xx.xxx.xxx.xxx
*********************
*********************
Raw packets sent: 40000 (1.600MB) | Rcvd: 39996 (1.840MB) | Lost: 4 (0.01%)

Network Stack Stress Test :: Pump network packets to target.          
Packet loss summary on bareleye bmc
*********************
*********************
Raw packets sent: 40000 (1.600MB) | Rcvd: 39214 (1.804MB) | Lost: 786 (1.97%)

Network Stack Stress Test :: Pump network packets to target.          | FAIL |
Failure, Packet loss.: 'Raw packets sent: 40000 (1.600MB) | Rcvd: 39214 (1.804MB) | Lost: 786 (1.97%)' does not contain 'Lost: 0 (0.00%)'
------------------------------------------------------------------------------
Confirm The Issue :: Determine whether issue is with Network or BMC.  | FAIL |
Failed, there is an issue with BMC.

Pls find dmesg logs also.
[dmsglogs.txt](https://github.com/openbmc/openbmc/files/860636/dmsglogs.txt)

sivassrr, 2017-04-03:
Clearly seeing BMC network services down even before we call BMC reset.

williamspatrick, 2017-04-03:
Palmetto and Barreleye both use the known-buggy NCSI chip. I did not get answers to my questions from 12 days ago so I'm going to assume this is the NCSI problem. There is nothing we can do about it unless:

1) you are seeing this issue while the system is stationary at standby or runtime. 

2) you are seeing this on a Witherspoon.

closed 2017-04-03 by williamspatrick