openbmc/openbmc #1297: eSEL Missing AdditionalData [ Witherspoon ]

gkeishin, 2017-03-13:
```
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/620/ 
- [x] VERSION_ID : v1.99.3-68-g35f91b9
- [x] Kernel     : Linux witherspoon 4.7.10-ec62e5e4b9bb4f4e9df718fa7aa4d452fa906e1e #1 Fri Mar 10 16:20:13 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.1
        op-build-v1.15-7-g385e16c
        buildroot-2017.02-4-gc28573e
        skiboot-28e96ff
        hostboot-6f2a36c
        linux-4.10.1-p58a0795
        petitboot-v1.4.1-pf956511
        witherspoon-xml-065aaa2
        occ-4141b5f
        hostboot-binaries-4b44090
        capp-ucode-1bb7503
```
`During Poweron, the error esel AddtionalData is EMPTY in the REST but the journal log does have the data`

```
Executing: GET  /xyz/openbmc_project/logging/enumerate

/xyz/openbmc_project/logging/enumerate
{
    "/xyz/openbmc_project/logging/internal/manager": {},
    "/xyz/openbmc_project/logging/entry/2": {
        "Timestamp": 1487744170849,
        "AdditionalData": [
            "ESEL=00 00 df 00 00 00 00 20 00 04 07 a4 07 aa 00 00 50 48 00 30 01 00 07 00 00 00 00 04 a2 a4 4b a0 00 00 00 04 a3 63 d4 7e 42 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 90 00 00 42 90 00 00 42 55 48 00 18 01 00 07 00 10 03 40 00 00 00 00 00 ff ff 00 00 00 00 00 00 50 53 00 50 01 01 00 00 02 00 00 09 07 05 00 48 00 00 00 e0 00 00 05 00 00 00 00 00 00 20 00 00 00 01 00 00 00 05 00 01 02 00 ff ff 01 90 00 48 42 43 31 30 30 37 30 35 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 55 44 00 24 01 06 01 00 01 54 41 4b 00 00 00 06 00 00 00 00 00 00 00 00 00 00 00 11 23 01 00 02 00 05 01 00 55 44 00 1c 01 06 01 00 02 54 41 4b 00 00 00 01 00 00 00 55 00 00 00 2b 00 00 00 00 55 44 00 50 01 03 01 00 00 00 00 00 00 03 f3 24 00 00 00 00 00 03 b6 94 00 00 00 00 40 25 93 94 00 00 00 00 40 25 9c 68 00 00 00 00 40 25 9d c4 00 00 00 00 40 25 bd 08 00 00 00 00 40 25 be f0 00 00 00 00 40 0c 30 9c 00 00 00 00 00 00 25 80 55 44 00 38 01 01 01 00 48 6f 73 74 62 6f 6f 74 20 42 75 69 6c 64 20 49 44 3a 20 68 6f 73 74 62 6f 6f 74 2d 36 66 32 61 33 36 63 2f 68 62 69 63 6f 72 65 2e 62 69 6e 00 55 44 00 fc 01 15 31 00 01 28 00 42 49 32 43 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f4 00 00 00 00 00 00 00 f4 00 00 00 05 00 00 00 00 00 00 00 26 33 83 b1 43 00 00 00 c9 00 08 43 4f 83 b9 0f 9a 00 00 0d 0f 00 00 00 00 00 00 00 01 00 00 00 24 00 00 00 26 33 86 b9 d9 00 00 00 c9 00 10 43 4f 10 87 d4 ee 00 00 0c cc 00 00 00 00 00 05 00 01 00 00 00 00 00 00 00 02 00 00 00 2c 00 00 00 26 33 86 c9 97 00 00 00 c9 00 08 43 4f 83 b9 0f 9a 00 00 0d 0f 00 00 00 00 00 00 00 02 00 00 00 24 00 00 00 26 33 86 fa 69 00 00 00 c9 00 18 43 4f 1b ce fc ec 00 00 0a 1e 00 00 00 00 00 05 00 01 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 34 00 00 00 26 34 ae d5 f3 00 00 00 c9 00 08 43 4f 65 c9 23 ee 00 00 07 6b 00 01 00 00 00 00 00 00 00 00 00 24 "
        ],
        "Message": "org.open_power.Error.Host.Event.Event",
        "Id": 2,
        "Severity": "xyz.openbmc_project.Logging.Entry.Level.Emergency"
    },
    "/xyz/openbmc_project/logging/entry/1": {
        "Timestamp": 1487744167566,
        "AdditionalData": [],
        "Message": "org.open_power.Error.Host.Event.Event",
        "Id": 1,
        "Severity": "xyz.openbmc_project.Logging.Entry.Level.Emergency"
    }
}
```

`busctl also shows the same EMPTY field`

```
root@witherspoon:~# busctl introspect xyz.openbmc_project.Logging /xyz/openbmc_project/logging/entry/1 | cat
NAME                                TYPE      SIGNATURE RESULT/VALUE                             FLAGS
org.freedesktop.DBus.Introspectable interface -         -                                        -
.Introspect                         method    -         s                                        -
org.freedesktop.DBus.Peer           interface -         -                                        -
.GetMachineId                       method    -         s                                        -
.Ping                               method    -         -                                        -
org.freedesktop.DBus.Properties     interface -         -                                        -
.Get                                method    ss        v                                        -
.GetAll                             method    s         a{sv}                                    -
.Set                                method    ssv       -                                        -
.PropertiesChanged                  signal    sa{sv}as  -                                        -
xyz.openbmc_project.Logging.Entry   interface -         -                                        -
.AdditionalData                     property  as        0                                        emits-change writable
.Id                                 property  u         1                                        emits-change writable
.Message                            property  s         "org.open_power.Error.Host.Event.Event"  emits-change writable
.Severity                           property  s         "xyz.openbmc_project.Logging.Entry.Le... emits-change writable
.Timestamp                          property  t         1487744167566                            emits-change writable

root@witherspoon:~# busctl introspect xyz.openbmc_project.Logging /xyz/openbmc_project/logging/entry/2 | cat
NAME                                TYPE      SIGNATURE RESULT/VALUE                             FLAGS
org.freedesktop.DBus.Introspectable interface -         -                                        -
.Introspect                         method    -         s                                        -
[... 53 lines cut ...]
        "_BOOT_ID" : "9a2599f6f5d347419feaec4374486116",
        "PRIORITY" : "6",
        "_UID" : "0",
        "_GID" : "0",
        "_SYSTEMD_SLICE" : "system.slice",
        "_MACHINE_ID" : "3044ce9543df45ed9e5f5d69892993cc",
        "_HOSTNAME" : "witherspoon",
        "_TRANSPORT" : "journal",
        "_CAP_EFFECTIVE" : "3fffffffff",
        "CODE_FILE" : "/usr/include/phosphor-logging/log.hpp",
        "CODE_LINE" : "100",
        "CODE_FUNC" : "helper_log",
        "SYSLOG_IDENTIFIER" : "ipmid",
        "_PID" : "860",
        "_COMM" : "ipmid",
        "_EXE" : "/usr/sbin/ipmid",
        "_CMDLINE" : "/usr/sbin/ipmid",
        "_SYSTEMD_CGROUP" : "/system.slice/org.openbmc.HostServices.service",
        "_SYSTEMD_UNIT" : "org.openbmc.HostServices.service",
        "MESSAGE" : "Received Host Event",
        "TRANSACTION_ID" : "3036231542",
        "ESEL" : "00 00 df 00 00 00 00 20 00 04 07 a4 07 aa 00 00 50 48 00 30 01 00 07 00 00 00 00 04 a2 a4 4b a0 00 00 00 04 a3 63 d4 7e 42 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 90 00 00 42 90 00 00 42 55 48 00 18 01 00 07 00 10 03 40 00 00 00 00 00 ff ff 00 00 00 00 00 00 50 53 00 50 01 01 00 00 02 00 00 09 07 05 00 48 00 00 00 e0 00 00 05 00 00 00 00 00 00 20 00 00 00 01 00 00 00 05 00 01 02 00 ff ff 01 90 00 48 42 43 31 30 30 37 30 35 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 55 44 00 24 01 06 01 00 01 54 41 4b 00 00 00 06 00 00 00 00 00 00 00 00 00 00 00 11 23 01 00 02 00 05 01 00 55 44 00 1c 01 06 01 00 02 54 41 4b 00 00 00 01 00 00 00 55 00 00 00 2b 00 00 00 00 55 44 00 50 01 03 01 00 00 00 00 00 00 03 f3 24 00 00 00 00 00 03 b6 94 00 00 00 00 40 25 93 94 00 00 00 00 40 25 9c 68 00 00 00 00 40 25 9d c4 00 00 00 00 40 25 bd 08 00 00 00 00 40 25 be f0 00 00 00 00 40 0c 30 9c 00 00 00 00 00 00 25 80 55 44 00 38 01 01 01 00 48 6f 73 74 62 6f 6f 74 20 42 75 69 6c 64 20 49 44 3a 20 68 6f 73 74 62 6f 6f 74 2d 36 66 32 61 33 36 63 2f 68 62 69 63 6f 72 65 2e 62 69 6e 00 55 44 00 fc 01 15 31 00 01 28 00 42 49 32 43 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f4 00 00 00 00 00 00 00 f4 00 00 00 05 00 00 00 00 00 00 00 26 33 83 b1 43 00 00 00 c9 00 08 43 4f 83 b9 0f 9a 00 00 0d 0f 00 00 00 00 00 00 00 01 00 00 00 24 00 00 00 26 33 86 b9 d9 00 00 00 c9 00 10 43 4f 10 87 d4 ee 00 00 0c cc 00 00 00 00 00 05 00 01 00 00 00 00 00 00 00 02 00 00 00 2c 00 00 00 26 33 86 c9 97 00 00 00 c9 00 08 43 4f 83 b9 0f 9a 00 00 0d 0f 00 00 00 00 00 00 00 02 00 00 00 24 00 00 00 26 33 86 fa 69 00 00 00 c9 00 18 43 4f 1b ce fc ec 00 00 0a 1e 00 00 00 00 00 05 00 01 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 34 00 00 00 26 34 ae d5 f3 00 00 00 c9 00 08 43 4f 65 c9 23 ee 00 00 07 6b 00 01 00 00 00 00 00 00 00 00 00 24 ",
        "_SOURCE_REALTIME_TIMESTAMP" : "1487744170779756"
}
```

gkeishin, 2017-03-13:
- [x]  Journal log 
[journald_v1.99.3-68-g35f91b9_eSEL.txt](https://github.com/openbmc/openbmc/files/837493/journald_v1.99.3-68-g35f91b9_eSEL.txt)

gkeishin, 2017-03-13:
@anoo1 Looks like this one is coming back ..

gkeishin, 2017-03-14:
Not sure if this will help.. but we hit this one again on poweron

```
- [x] SYSTEM     : W42 ( Witherspoon - 2 Sockets - 16 DIMMs )
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/622/ 
- [x] VERSION_ID : v1.99.3-71-ge8fdc66
- [x] Kernel     : Linux witherspoon 4.7.10-ec62e5e4b9bb4f4e9df718fa7aa4d452fa906e1e #1 Mon Mar 13 22:20:38 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.1
        op-build-v1.15-7-g385e16c
        buildroot-2017.02-4-gc28573e
        skiboot-28e96ff
        hostboot-6f2a36c
        linux-4.10.1-p58a0795
        petitboot-v1.4.1-pf956511
        witherspoon-xml-065aaa2
        occ-4141b5f
        hostboot-binaries-4b44090
        capp-ucode-1bb7503

root@witherspoon:~# journalctl -f /usr/sbin/ipmid
-- Logs begin at Wed 2017-02-22 06:04:58 UTC. --
Feb 22 06:10:48 witherspoon ipmid[885]: Failed to validate common header
Feb 22 06:10:48 witherspoon ipmid[885]: Invalid entry:[0] in byte-0
Feb 22 06:10:48 witherspoon ipmid[885]: Failed to validate common header
Feb 22 06:10:48 witherspoon ipmid[885]: Invalid entry:[0] in byte-0
Feb 22 06:10:48 witherspoon ipmid[885]: Failed to validate common header
Feb 22 06:10:55 witherspoon ipmid[885]: Received Host Event
Feb 22 06:10:55 witherspoon ipmid[885]: A host system event was received
Feb 22 06:10:58 witherspoon ipmid[885]: Received Host Event
Feb 22 06:10:58 witherspoon ipmid[885]: A host system event was received
Feb 22 06:12:59 witherspoon ipmid[885]: ERROR:[0xFF] handling NetFn:[0xA], Cmd:[0x49]

Executing: GET  /xyz/openbmc_project/logging/enumerate

/xyz/openbmc_project/logging/enumerate
{
    "/xyz/openbmc_project/logging/internal/manager": {},
    "/xyz/openbmc_project/logging/entry/2": {
        "Timestamp": 1487743858536,
        "AdditionalData": [
            "ESEL=00 00 df 00 00 00 00 20 00 04 12 01 6f aa 00 00 50 48 00 30 01 00 17 00 00 00 00 04 3a 18 84 e4 00 00 00 04 3a 1b 25 68 42 00 00 07 00 00 00 00 00 00 00 00 00 00 00 00 90 00 00 0b 90 00 00 0b 55 48 00 18 01 00 17 00 8a 03 40 00 00 00 00 00 ff ff 00 00 00 00 00 00 50 53 00 50 01 01 00 00 02 00 00 09 17 0e 00 48 00 00 00 e0 00 00 0a 00 00 00 00 00 00 20 00 00 00 05 00 01 00 2d 00 04 00 00 00 00 00 00 00 00 42 43 38 41 31 37 30 45 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 55 44 00 1c 01 06 01 00 02 54 41 4b 00 00 00 06 00 00 00 55 00 03 b6 94 00 00 00 00 55 44 00 4c 01 02 01 00 ee ee ee ee 54 61 72 67 65 74 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 05 00 01 00 00 00 03 00 00 00 05 00 00 00 14 0f 7a bb 7c 23 01 00 02 00 05 01 0b 5a fc d7 13 01 00 02 00 05 01 00 00 55 44 00 30 01 03 01 00 00 00 00 00 00 03 f3 24 00 00 00 00 00 03 b6 94 00 00 00 00 40 03 dd 68 00 00 00 00 40 03 ba 48 00 00 00 00 00 00 25 80 55 44 00 38 01 01 01 00 48 6f 73 74 62 6f 6f 74 20 42 75 69 6c 64 20 49 44 3a 20 68 6f 73 74 62 6f 6f 74 2d 36 66 32 61 33 36 63 2f 68 62 69 63 6f 72 65 2e 62 69 6e 00 "
        ],
        "Message": "org.open_power.Error.Host.Event.Event",
        "Id": 2,
        "Severity": "xyz.openbmc_project.Logging.Entry.Level.Emergency"
    },
    "/xyz/openbmc_project/logging/entry/1": {
        "Timestamp": 1487743855465,
        "AdditionalData": [],
        "Message": "org.open_power.Error.Host.Event.Event",
        "Id": 1,
        "Severity": "xyz.openbmc_project.Logging.Entry.Level.Emergency"
    }
}

root@witherspoon:~# busctl introspect xyz.openbmc_project.Logging /xyz/openbmc_project/logging/entry/1 | cat
NAME                                TYPE      SIGNATURE RESULT/VALUE                             FLAGS
org.freedesktop.DBus.Introspectable interface -         -                                        -
.Introspect                         method    -         s                                        -
org.freedesktop.DBus.Peer           interface -         -                                        -
.GetMachineId                       method    -         s                                        -
.Ping                               method    -         -                                        -
org.freedesktop.DBus.Properties     interface -         -                                        -
.Get                                method    ss        v                                        -
.GetAll                             method    s         a{sv}                                    -
.Set                                method    ssv       -                                        -
.PropertiesChanged                  signal    sa{sv}as  -                                        -
xyz.openbmc_project.Logging.Entry   interface -         -                                        -
.AdditionalData                     property  as        0                                        emits-change writable
[... 61 lines cut ...]
        "_SYSTEMD_SLICE" : "system.slice",
        "_MACHINE_ID" : "3044ce9543df45ed9e5f5d69892993cc",
        "_HOSTNAME" : "witherspoon",
        "_TRANSPORT" : "journal",
        "_CAP_EFFECTIVE" : "3fffffffff",
        "CODE_FILE" : "/usr/include/phosphor-logging/log.hpp",
        "CODE_LINE" : "100",
        "CODE_FUNC" : "helper_log",
        "SYSLOG_IDENTIFIER" : "ipmid",
        "_PID" : "885",
        "_COMM" : "ipmid",
        "_EXE" : "/usr/sbin/ipmid",
        "_CMDLINE" : "/usr/sbin/ipmid",
        "_SYSTEMD_CGROUP" : "/system.slice/org.openbmc.HostServices.service",
        "_SYSTEMD_UNIT" : "org.openbmc.HostServices.service",
[... cut ...]  37 9e b4 e6 00 00 00 c9 00 18 43 4f 1b ce fc ec 00 00 0a 1e 00 00 00 00 00 05 00 01 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 34 00 00 00 26 38 c8 33 1c 00 00 00 c9 00 08 43 4f 65 c9 23 ee 00 00 07 6b 00 01 00 00 00 00 00 00 00 00 00 24 ",
        "_SOURCE_REALTIME_TIMESTAMP" : "1487743858500513"
}
root@witherspoon:~#
````

- [x]  Journald log
[journald_v1.99.3-71-ge8fdc66_eSEL_EMPTY.txt](https://github.com/openbmc/openbmc/files/840339/journald_v1.99.3-71-ge8fdc66_eSEL_EMPTY.txt)

gkeishin, 2017-03-31:
This one is pretty simple to reproduce now.. few back to back poweron would show this missing bug. Probably run on witherspoon for easy recreate if needed. Thanks

gkeishin, 2017-05-09:
Still seeing this .. anything on this one...

rfrandse, 2017-06-07:
<https://gerrit.openbmc-project.xyz/4646> commit: Add call to Synchronize journal dbus method
Fixes:  #1297 eSEL Missing AdditionalData [ Witherspoon ]

gkeishin, 2017-06-07:
Great thanks @anoo1 .. Will verify once in build.. Thanks

closed 2017-08-07 by williamspatrick