openbmc/openbmc #1298: log to journal when we retry UCD operations

geissonator, 2017-03-13:
We're chasing a pgood issue ( #832).  One theory is that there's an underlying I2C or UCD issue and that when we hit the error, and do the retry operation, we end up programming invalid data to the UCD and the corresponding voltage rails.

https://gerrit.openbmc-project.xyz/#/c/2327/ was where we added the retries.  For this issue, we just want to add a echo to the journal when we hit the retry loop.

geissonator, 2017-03-13:
Another question with ucd_disable_vcs.sh is what happens when the i2cset commands fail?  Does the whole script fail or do we just keep going?  Maybe we need some retries around those  as well?

saqibkh, 2017-03-13:
when the i2cset fails we get the following error:

Error: Could not set address to 0x64: Device or resource busy

geissonator, 2017-03-13:
ok cool, does it cause the whole script to fail?

saqibkh, 2017-03-13:
yes

closed 2017-03-20 by williamspatrick