openbmc/openbmc #1301: IPL: DBus.Error.ServiceUnknown: The name :1.50 was not provided by any .service files during bmc update on w15 with 910.1712.20170313a

cdeadmin, 2017-03-13:


williamspatrick, 2017-03-13:
This is a defect in openbmc-test-automation.

gkeishin, 2017-03-14:
Looking into it..

gkeishin, 2017-03-14:
```
22:35:05.096 	TRACE 	Arguments: [ ${uri}='/org/openbmc/control/flash/bmc/action/PrepareForUpdate' | ${timeout}='10' | ${quiet}='0' | &{kwargs}={'data': {'data': []}} ] 	
22:35:07.188 	TRACE 	Return: <Response [200]> 	
22:35:07.188 	INFO 	${resp} = <Response [200]
```
```
Start / End / Elapsed: 	20170313 22:35:08.034 / 20170313 22:35:13.136 / 00:00:05.102
22:35:08.034 	TRACE 	Arguments: [ 'openbmc' | '/org/openbmc/control/flash/bmc' | timeout='10' ] 	
22:35:13.135 	INFO 	Get Request using : alias=openbmc, uri=/org/openbmc/control/flash/bmc, headers=None 	
22:35:13.135 	TRACE 	Return: <Response [500]> 	
22:35:13.136 	INFO 	${ret} = <Response [500]
```

The PrepareForUpdate we read /org/openbmc/control/flash/bmc to check the status "status": "Switch to update mode in progress".

In this case the REST call to get the property and the time when the BMC is taking reboot in progress are seems to be happening progress resulting to this.. 

Need to look at the Dbus interface reboot action timing vs this REST call we make.

gkeishin, 2017-03-14:
Code in review..

gkeishin, 2017-03-15:
This is an automation bug.. so fix is review.

gkeishin, 2017-03-15:
```
Prepare For Update
    [Documentation]   Switch to update mode in progress. This method calls
    ...               the Abort method to remove the pending update if there
    ...               is any before code activation.
    ${data}=   create dictionary   data=@{EMPTY}
    ${resp}=    openbmc post request    ${BMC_PREP_METHOD}   data=${data}    <-- This triggers reboots in 3 seconds
    should be equal as strings   ${resp.status_code}   ${HTTP_OK}

    ${data}=      Read Properties     ${BMC_UPD_ATTR}    <-- we come and read post 3 seconds in this failure
    should contain    ${data['status']}   Switch to update mode in progress
```

closed 2017-03-15 by williamspatrick