openbmc/openbmc #1302: Can't unbind fsi-master-gpio driver prior to connecting FSP2

GHF, 2017-03-13:
I was trying to unbinding the BMC FSI master driver on Zaius BMC to be able to use FSP2.

```
echo 1 > /sys/bus/platform/drivers/fsi-master-gpio/unbind
-sh: echo: write error: No such device
```

I got this result on both the latest kernel revision a6f406cd33d43167948bcc947734c36d6f7c322c and just prior to the version bump.

I also tried `/sys/bus/fsi/drivers/iic_fsi_dd/unbind` but I don't think that's right.

eddiejames, 2017-03-13:
you need

`echo fsi-master > /sys/bus/platform/drivers/fsi-master-gpio/unbind`

GHF, 2017-03-13:
Oh look at that. Thanks!

Is there a way to rebind using the GPIO pins from device tree?

mdmillerii, 2017-03-13:
In general, echo the device name (the same string to `unbind`) to the `bind` file in the drivers directory.  The names of devices for a given bus can be found as symlinks in the `devices` directory, or bound devices can be found in the `drivers`/xxx directory (for driver xxx).

It should be:
`echo  fsi-master > /sys/bus/platform/drivers/fsi-master-gpio/bind`

amboar, 2017-03-13:
> Is there a way to rebind using the GPIO pins from device tree?

It doesn't do this automatically?

GHF, 2017-03-14:
>> Is there a way to rebind using the GPIO pins from device tree?
>
> It doesn't do this automatically?

`echo fsi-master > /sys/bus/platform/drivers/fsi-master-gpio/bind` is what I had tried and I guess it does use the device tree node. I saw that it failed and assumed I had provided the wrong arguments when it was actually a bug: #1261

closed 2017-03-14 by GHF