openbmc/openbmc #1310: Power Operations on Chassis

iffy-i, 2017-03-14:
Right now warm reboot or Orderly Shutdown are not in the API yet as these requires Chassis power operations. 
I am planning to link these to cold reboot and immediate shutdown for now. 
I will revisit this once the API supports these operations. 

Here is a screenshot for our interface for reference -

[image]

geissonator, 2017-03-14:
Once Vishwa's code is in for soft power off, we need to move to that as our default for power off.  And ensure the "hard" power off would work via the chassis target.
We then need to figure out what a warm reboot looks like and implement that (keep chassis power on)

geissonator, 2017-03-23:
- [ ] Ensure a chassis power off correctly "hard" shuts down system
- [ ] Move to new soft power off code in normal power off and reboot paths
- [ ] Work out warm reboot requirements
- [ ] Implement warm reboot requirements

geissonator, 2017-05-11:
Some things to consider on this one.
- We really have 3 targets to power off a system
  - obmc-host-shutdown.target, obmc-host-stop.target, obmc-chassis-poweroff.target
  - shutdown notifies the host, then goes into stop and then poweroff
  - stop run and then goes into poweroff
  - poweroff would just cut the power to the chassis
- A soft power off is to call shutdown
- A hard power off is to either call stop or poweroff (both will **not** notify the host and just cut power)
- The default when a user powers of a system will be to call shutdown (it has a quick 2-3s timer for the host to respond, and if they don't it simply goes into the stop target)

We want to allow a user to call any of those power off targets and result in a stable system that can be booted back up.

There are two things to think of when potentially "skipping" a target
- What services will not get run
- What conflicting services will not get stopped

shutdown.target
- This just calls the phosphor-softpoweroff application which is not needed if we don't want to notify the host
- It conflicts with 3 targets and 1 service.  In most cases it was added redundantly with the stop target

stop.target
- This runs the poweroff target and it runs the stop-watchdog.service.
  - stop-watchdog is important so we need to be sure it runs, even if someone just runs poweroff
- It conflicts with a LOT of other targets and services

I'm thinking with this issue I'll fix up the conflicts using shutdown to just use stop, and then ensure the stop target gets run, even if someone just does a poweroff.  As long as we design the services associated with stop to not break anything if the corresponding start hasn't been run, we should e fine.  The alternative is to add some logic in the poweroff to check if start was run and if so, run stop, but I'd prefer and avoid that complexity if possible.

rfrandse, 2017-05-16:
https://gerrit.openbmc-project.xyz/4228 Ensure host-stop target called on chassis-poweroff
Resolves #1310 Power Operations on Chassis

closed 2017-05-19 by williamspatrick