openbmc/openbmc #1311: system does not indicate startup finished

shenki, 2017-03-15:
We used to get this string in the journal when the system had finished booting:

 `systemd[1]: Startup finished in 12.033s (kernel) + 1min 41.650s (userspace) = 1min 53.683s.`

I no longer see this on Witherspoon with [v1.99.3-79-g6b945f4](6b945f4). Does that mean we're not reaching a state that we should be?

shenki, 2017-03-15:
```
root@w23:~# systemctl is-system-running
starting
```

shenki, 2017-03-15:
[w23-journal.txt](https://github.com/openbmc/openbmc/files/843357/w23-journal.txt)

shenki, 2017-03-15:
```
root@w23:~# systemctl status --no-pager
● w23
    State: starting
     Jobs: 2 queued
   Failed: 0 units
    Since: Thu 1970-01-01 00:00:06 UTC; 47 years 1 months ago
```

williamspatrick, 2017-03-15:
`systemctl list-jobs` showed:

```
systemctl list-jobs
JOB UNIT                                                       TYPE  STATE  
158 op-vpd-parser.service                                      start waiting
206 mapper-wait@-xyz-openbmc_project-Inventory-Manager.service start running
```

I added a fix to Gerrit for this:

remote:   https://gerrit.openbmc-project.xyz/3029 openpower-fru-vpd: fix mapper Wants for service
remote:   https://gerrit.openbmc-project.xyz/3030 inventory-manager: add Before stmt for service

@geissonator - It seems like we need some sort of timeout condition for the obmc-standby.target at least.  If we take more than N minutes to get to standby we need that reported, right?

shenki, 2017-03-15:
I verified that these fixes resolve the issue on a witherspoon.

geissonator, 2017-03-15:
@williamspatrick yeah we should probably add a timeout to all our major action targets.

closed 2017-03-15 by williamspatrick