openbmc/openbmc #1315: Ensure uboot fw_env update script uses new inventory object.

williamspatrick, 2017-03-15:
The inventory objects were changed from /org to /xyz namespace.  There is a script that updates the u-boot fw_env for the MAC address that needs to be updated to use the new inventory interfaces.

chinaridinesh, 2017-03-30:
Changes pushed into repository via: https://gerrit.openbmc-project.xyz/#/c/3387/

chinaridinesh, 2017-06-20:
Following changes pushed as part of this story:

https://gerrit.openbmc-project.xyz/#/c/3387/
https://gerrit.openbmc-project.xyz/#/c/4006/
https://gerrit.openbmc-project.xyz/#/c/4879/
https://gerrit.openbmc-project.xyz/#/c/4880/

chinaridinesh, 2017-07-13:
Accessing the rick of updating chassis_control.py from old uuid object to new one, since chassis_control.py implements lot more than just uuid. This is part of implementing uuid with new inventory objects.

chinaridinesh, 2017-07-14:
Working on testing the changes done for uuid and xyz name space.

chinaridinesh, 2017-07-17:
Need some design inputs on UUID formats. Thats blocking now.

mdmillerii, 2017-07-17:
systemd supports just hex digits or digits with hypens in "right" space.  Must be be (a uuid, not a half-le guid).

chinaridinesh, 2017-07-18:
Pushed code for review.

chinaridinesh, 2017-07-19:
Waiting on review comment from Deepak for https://gerrit.openbmc-project.xyz/#/c/5358/

chinaridinesh, 2017-07-20:
Received review comments from @dkodihal . Implemented already as suggested. Not getting expected test result. Debugging

chinaridinesh, 2017-07-21:
Test still not showing expected result. Need more time to debug.

chinaridinesh, 2017-07-24:
Test of uuid via busctl tool works, but it shows empty due to no keyword defined for vpd parsing.

chinaridinesh, 2017-07-25:
STATUS

7/25
Done with the changes. In final test with changed namespace from org.openbmc to xyz.openbmc_project .

rfrandse, 2017-08-30:
Critical changes are in for phase 5 moving to phase 5.1

closed 2017-09-13 by rfrandse