openbmc/openbmc #1318: Fan Presence - Add gpio based presence detection

msbarth, 2017-03-15:
Determine how pca9552 is accessed in user space and add an application within openbmc/phosphor-fan-presence that creates/updated inventory for fans defined to be gpio based presence detection.

msbarth, 2017-03-24:
An additional bitbake argument is necessary to pass to the mrw parser to setup whether a fan should be tach or gpio based presence detection when creating the yaml file.
https://gerrit.openbmc-project.xyz/#/c/3122/1/gen_presence_yaml.pl@43

bradbishop, 2017-05-18:
[ ] - build skeleton - create stubbed application with gpio and inventory parameters
[ ] - add template systemd unit
[ ] - add application package to existing phosphor-fans recipe
[ ] - add package to inventory packagegroup
[ ] - write error yaml and run elog-gen.py ( general logic failures, libevdev failures )

closed 2017-08-09 by williamspatrick