openbmc/openbmc #1322: pgood is lost on BMC reboot on witherspoon

geissonator, 2017-03-15:
While testing my code to detect when pgood is on after a BMC reset, and setup the BMC userspace applications correctly to handle that, I noticed that pgood always gets reset to 0 on a BMC reset.

My simple test is the following:

Turn just chassis power on
```
busctl set-property xyz.openbmc_project.State.Chassis /xyz/openbmc_project/state/chassis0 xyz.openbmc_project.State.Chassis RequestedPowerTransition s xyz.openbmc_project.State.Chassis.Transition.On
```

Do a BMC "reboot"

Then, as soon as you have console access, start running:
```
gpioutil -p D2 -d  in # check pgood status pin
0
```

It's 0 as soon as I can look at it.  This is before we run most of the userspace applications that would touch it so kind of makes me thing kernel init area.  Not for sure though.

Same test works fine on barreleye.

@amboar any thoughts?

geissonator, 2017-03-15:
```
[witherspoon]# dmesg
Booting Linux on physical CPU 0x0
Linux version 4.7.10-ec62e5e4b9bb4f4e9df718fa7aa4d452fa906e1e (jenkins@openbmc-builder1) (gcc version 5.3.0 (GCC) ) #1 Wed Mar 15 15:45:58 UTC 2017
CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Machine model: Witherspoon BMC
Memory policy: Data cache writeback
SOC Rev: 04030303
On node 0 totalpages: 110592
free_area_init_node: node 0, pgdat 8071d26c, node_mem_map 9ec19000
  Normal zone: 992 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 110592 pages, LIFO batch:31
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0 
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 109600
Kernel command line: console=ttyS4,115200n8 root=/dev/ram rw
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Memory: 430372K/442368K available (4203K kernel code, 149K rwdata, 896K rodata, 1024K init, 110K bss, 11996K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0x9f800000 - 0xff800000   (1536 MB)
    lowmem  : 0x80000000 - 0x9f000000   ( 496 MB)
      .text : 0x80008000 - 0x805faff4   (6092 kB)
      .init : 0x80600000 - 0x80700000   (1024 kB)
      .data : 0x80700000 - 0x80725528   ( 150 kB)
       .bss : 0x80725528 - 0x807410a8   ( 111 kB)
NR_IRQS:16 nr_irqs:16 16
clocksource: moxart_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 77222644334 ns
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
Calibrating delay loop... 789.70 BogoMIPS (lpj=3948544)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x80100000 - 0x80100038
devtmpfs: initialized
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
hw-breakpoint: maximum watchpoint size is 4 bytes.
clocksource: Switched to clocksource moxart_timer
NET: Registered protocol family 2
TCP established hash table entries: 4096 (order: 2, 16384 bytes)
TCP bind hash table entries: 4096 (order: 2, 16384 bytes)
TCP: Hash tables configured (established 4096 bind 4096)
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
RPC: Registered named UNIX socket transport module.
RPC: Registered udp transport module.
RPC: Registered tcp transport module.
RPC: Registered tcp NFSv4.1 backchannel transport module.
Trying to unpack rootfs image as initramfs...
Freeing initrd memory: 1536K (9ea08000 - 9eb88000)
futex hash table entries: 256 (order: -1, 3072 bytes)
workingset: timestamp_bits=29 max_order=17 bucket_order=0
squashfs: version 4.0 (2009/01/31) Phillip Lougher
jffs2: version 2.2. (NAND) (SUMMARY)  \ufffd\ufffd 2001-2006 Red Hat, Inc.
io scheduler noop registered (default)
ipmi-bt-host 1e789140.ibt: Found bt bmc device
ipmi-bt-host 1e789140.ibt: Using IRQ 21
Serial: 8250/16550 driver, 6 ports, IRQ sharing enabled
1e783000.serial: ttyS0 at MMIO 0x1e783000 (irq = 22, base_baud = 1500000) is a 16550A
[... 94 lines cut ...]
systemd[1]: Created slice system-org.openbmc.control.Checkstop.slice.
systemd[1]: Created slice system-obmc\x2dconsole.slice.
systemd[1]: Created slice system-serial\x2dgetty.slice.
systemd[1]: Created slice system-org.openbmc.buttons.Power.slice.
systemd[1]: Created slice system-org.openbmc.control.Chassis.slice.
systemd[1]: Starting udev Coldplug all Devices...
systemd[1]: Starting Apply Kernel Variables...
systemd[1]: Mounting Debug File System...
systemd[1]: Starting Remount Root and Kernel File Systems...
systemd[1]: Mounted Debug File System.
systemd[1]: Mounted Temporary Directory.
systemd[1]: Started Journal Service.
systemd-journald[441]: Received request to flush runtime journal from PID 1
ftgmac100 1e660000.ethernet eth0: NCSI interface up
random: nonblocking pool is initialized
[witherspoon]# 
[witherspoon]# 
[witherspoon]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-80"
VERSION_ID="v1.99.3-80-gf4cade7"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-80"
BUILD_ID="v1.99.3"
```

amboar, 2017-03-16:
I thought we weren't resetting the GPIO controller on BMC reset, but maybe we are?

geissonator, 2017-03-16:
Did a quick test where I removed the obmc-standby.target from the reboot (to prevent all the user space applications).  Still saw the gpio pin go to 0.  It still did start the Avahi and a few default services but pointing more and more to something in the kernel init area.

Is the GPIO controller path different on barreleye vs. witherspoon @amboar ?  Everything works fine on barreleye.

geissonator, 2017-03-24:
A potential fix from Xo in this area, although I'm pretty sure in my last test I disabled the power_control code from running at all - https://gerrit.openbmc-project.xyz/#/c/2837/

chinaridinesh, 2017-06-01:
Jun 01 19:21:12 witherspoon systemd[1]: op-reset-pgood-check@0.service: Main process exited, code=exited, status=1/FAILURE
Jun 01 19:21:12 witherspoon systemd[1]: Failed to start Check Chassis0 pgood on reset.
Jun 01 19:21:12 witherspoon systemd[1]: Dependency failed for Create file to indicate chassis0 pgood is on.
Jun 01 19:21:12 witherspoon systemd[1]: op-reset-pgood-check@0.service: Unit entered failed state.
Jun 01 19:21:12 witherspoon systemd[1]: op-reset-pgood-check@0.service: Failed with result 'exit-code'.

sivassrr, 2017-06-16:
When can we expect this changes? any idea. That help us to stage our test accordingly. Thank you.

sivassrr, 2017-06-18:
Oops, by mistake got unassigned. sorry dont know how it happened.

spinler, 2017-07-05:
After I reboot, I see:
```
root@w5:~# gpioutil -p R1   #SOFTWARE_PGOOD
1
root@w5:~# gpioutil -p D1  #BMC_POWER_UP
0
```
User space code handles these together, and sets to 1 on power on and 0 on a power off.  So it looks like only R1 survived the reboot, and also I would guess the kernel is doing it.

@amboar Is it a hint at all that a D block GPIO gets reset but R is OK?

spinler, 2017-07-05:
The Zaius fix referred to doesn't apply here, as that system has a latch for the power_up signal in between the BMC and UCD to hold its state while the BMC reboots, and Witherspoon doesn't.  That would have solved our problem though.

(For some reason, the schematic shows that Witherspoon does have a jumper on that signal which would give someone the ability to always keep it high.)

amboar, 2017-07-06:
@spinler from a quick look at the datasheet, the GPIO reset is an all-or-nothing operation. I'm not aware of the ability to select banks. Further, unless the GPIO block is reset on a watchdog event, the kernel should not be touching GPIO values unless we instruct it to (e.g. via the devicetree).

amboar, 2017-07-07:
Correction: @mdmillerii points out that there are reset tolerance registers in the GPIO block. I had overlooked them. I'll cook up a hack in the short term, but longer term we'll need some bindings and some additions to the GPIO driver.

sivassrr, 2017-07-19:
Any update here?

mdmillerii, 2017-07-19:
The interim patch openbmc/linux@b2cd8101eec3c200940405f12582473617a520f7 was applied and merged openbmc/openbmc@ee3530e085ab39016936700ef8426f34 last week.  Are additional fixes required?

geissonator, 2017-07-19:
This is fixed and in master.  There have been some other issues uncovered now that we preserve the pgood on the reboot, but not caused by this change.

closed 2017-07-19 by geissonator