openbmc/openbmc #1323: openpower-proc-control:  Callout processors on FSI failures

spinler, 2017-03-15:
The start_host and vcs_workaround procedures write P9 CFAM registers with FSI.  If we get failures on reads or writes, I need to create an elog with a callout.  The logging code already has support for converting an FSI device path to the processor callout.

At the time of this writing, processing the error/metadata yaml files in openpower-dbus-interfaces hasn't been turned on yet, which I'll need.

dhruvibm, 2017-04-06:
- [ ] Understanding the code change

- [ ] Create error log with callout for FSI write or read

spinler, 2017-04-06:
Note that I created https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Common/File.errors.yaml and File.metadata.yaml for the errors for the open() and seek() calls.  These don't need callouts, just regular error logs.

rfrandse, 2017-05-04:
[gerrit 3907](https://gerrit.openbmc-project.xyz/#/c/3907/)

closed 2017-05-18 by williamspatrick