openbmc/openbmc #1324: [BarrelEye] /usr/sbin/phosphor-host-state-manager depends on /usr/sbin/system_manager.py

yadron, 2017-03-15:
If /usr/sbin/system_manager.py fails (for reproducing try something like at BMC

```
cp /usr/sbin/system_manager.py /usr/sbin/system_manager.py_
rm  /usr/sbin/system_manager.py
ps|grep system_manager.py
kill [num]
```

), then /usr/sbin/phosphor-host-state-manager doesn't work too (e.g.

```
root@barreleye:~# /usr/sbin/phosphor-host-state-manager &
[7] 5194
root@barreleye:~# 
[7]+  Segmentation fault      (core dumped) /usr/sbin/phosphor-host-state-manager
```

), then everything fails (like 

```
root@barreleye:~# obmcutil state
Traceback (most recent call last):
  File "/usr/sbin/obmcutil", line 196, in <module>
    obj = bus.get_object(bus_name, obj_path)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.managers.System was not provided by any .service files
```

or

```
root@barreleye:~# /usr/sbin/power_control.exe__ 3000 10 &
[6] 4339
root@barreleye:~# Power GPIO: call to getGpioConfiguration failed: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.managers.System was not provided by any .service files
ERROR PowerControl: could not read power GPIO configuration

(process:4339): GLib-CRITICAL **: g_variant_new_string: assertion 'string != NULL' failed
ERROR PowerControl: GPIO setup (rc=2)

root@barreleye:~# 
root@barreleye:~# 
root@barreleye:~# ERROR PowerControl: GPIO open error (gpio=(null),rc=2)
```

).

If /usr/sbin/system_manager.py fixed, then everything works again:

```
root@barreleye:~# ps|grep system_manager.py 
 5751 root       0:01 python /usr/sbin/system_manager.py

root@barreleye:~# /usr/sbin/phosphor-host-state-manager &
[7] 5810

root@barreleye:~# obmcutil state
 = BMC_STARTING2
```

I think that this is very bad (bug in /usr/sbin/phosphor-host-state-manager), because one python program (/usr/sbin/system_manager.py) that is not controlling anything (but just management) is the most important one right now (like all other control services doesn't work without this one).

I already written about this here
https://github.com/openbmc/openbmc/issues/1262
, but that was a very long message about PNOR issues (so this note lost there), and I've just checked this again, so created this separate bug.

BTW, some of the services itself were described here some time ago (like which one does what):
https://github.com/openbmc/openbmc/issues/262

geissonator, 2017-03-15:
Yes, I have a TODO in openbmc/phosphor-state-manager/host_state_manager.cpp, line 62 to fix this dependency once some of our other pieces are in place.  First I need to get #1274 done.

Why are you going around killing system_manager.py to begin with here?  :)

yadron, 2017-03-15:
The original reason that I'm having again problems with powering on host (the update of .pnor doesn't help anymore).

It could be timeout problems of /usr/sbin/power_control.exe or maybe code power_control_obj.c duplicated in many services (obmc-control-chassis, obmc-mgr-system, obmc-op-control-power, obmc-mgr-state, ...) or something else.

If power_control_obj.c works in many services simultaneous, could it be a random gpio write/read for checking PGood state from different services?

If yes, then it can answer why sometimes boot host works and sometimes now _(then just depends on any of these services and which one was executed first and both on how system busy, so different delays)_.

For me it looks like these services and code power_control_obj.c needs refactoring, but I'm not sure yet what is the workaround for fixing it right now. See my last comment here for more details: https://github.com/openbmc/openbmc/issues/1287

One more similar issue. Try saving a big file to BMC _(for example, to /usr/sbin/bigfile)_ until no space left and then nothing will work anymore:

```
root@barreleye:/run/initramfs# reboot
-sh: /sbin/reboot: Input/output error

root@barreleye:/run/initramfs# dmesg|tail
SQUASHFS error: Unable to read data cache entry [c879c]
SQUASHFS error: Unable to read page, block c879c, size bcb4
```

and even worse:

```
root@barreleye:/run/initramfs# rm /usr/sbin/bigfile 
rm: can't remove '/usr/sbin/systemctl': No space left on device
```

yadron, 2017-03-16:
Regarding flash space overflow bug (in msg above), normally I have (and everything works):

```
root@barreleye:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
dev                     247708         0    247708   0% /dev
tmpfs                   248988      9156    239832   4% /run
/dev/mtdblock4           14976     14976         0 100% /run/initramfs/ro
/dev/mtdblock5            4096       868      3228  21% /run/initramfs/rw
cow                       4096       868      3228  21% /
tmpfs                   248988         0    248988   0% /dev/shm
tmpfs                   248988         0    248988   0% /sys/fs/cgroup
tmpfs                   248988         0    248988   0% /tmp
tmpfs                   248988         4    248984   0% /var/volatile
root@barreleye:~# df -h
Filesystem                Size      Used Available Use% Mounted on
dev                     241.9M         0    241.9M   0% /dev
tmpfs                   243.2M      8.9M    234.2M   4% /run
/dev/mtdblock4           14.6M     14.6M         0 100% /run/initramfs/ro
/dev/mtdblock5            4.0M    868.0K      3.2M  21% /run/initramfs/rw
cow                       4.0M    868.0K      3.2M  21% /
tmpfs                   243.2M         0    243.2M   0% /dev/shm
tmpfs                   243.2M         0    243.2M   0% /sys/fs/cgroup
tmpfs                   243.2M         0    243.2M   0% /tmp
tmpfs                   243.2M      4.0K    243.1M   0% /var/volatile
```

And when "No space left on device", it was:

```
root@barreleye:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
dev                     247708         0    247708   0% /dev
tmpfs                   248988      9160    239828   4% /run
/dev/mtdblock4           14976     14976         0 100% /run/initramfs/ro
/dev/mtdblock5            4096       856      3240  21% /run/initramfs/rw
cow                       4096       856      3240  21% /
tmpfs                   248988         0    248988   0% /dev/shm
tmpfs                   248988         0    248988   0% /sys/fs/cgroup
tmpfs                   248988         0    248988   0% /tmp
tmpfs                   248988         4    248984   0% /var/volatile
```

williamspatrick, 2017-03-16:
@yadron: if you are running out of space look at the contents of /run/initramfs/rw/cow. You probably have some big debug files left over in there that you can erase (and then reboot).

Since the current file system is an overlay, all changes build up in the JFFS2 volume until you manually delete them or do a code update again.

yadron, 2017-03-17:
@williamspatrick , thank you for the comment regarding disk space.

After BMC running for one or two days (ver v1.99.3 without modifications), I'm trying updating it to the last version (and no out of space disk problems). Why this error happens when I try "reboot':

```
root@barreleye:/run/initramfs# reboot
-sh: /sbin/reboot: Input/output error

root@barreleye:/run/initramfs# dmesg|tail
"SQUASHFS error: Unable to read data cache entry [daa94]
SQUASHFS error: Unable to read page, block daa94, size bcb4
SQUASHFS error: Unable to read data cache entry [daa94]
SQUASHFS error: Unable to read page, block daa94, size bcb4"
```

The same for:

```
root@barreleye:/run/initramfs# tail /bin/systemctl 
tail: read error: Input/output error
```

If I reboot BMC immediately after booting, then no problems (it works). If I try it after some hours, then this error happens.

What I did before seeing this error, as usually:

1. `scp ./image-bmc root@172.17.27.92:/run/initramfs/`

2 "ERROR: Device mtd4 is mounted, image-bmc is busy.", so
```
root@barreleye:/run/initramfs# umount /dev/mtdblock4
root@barreleye:/run/initramfs# umount /dev/mtdblock5
```

3. 
```
root@barreleye:/run/initramfs# ./update 
update:
Updating bmc...
Erasing block: 128/128 (100%) 
Writing kb: 32768/32768 (100%) 
Verifying kb: 32768/32768 (100%) 
```

4. `reboot`
- cannot reboot, saying
```
root@barreleye:/run/initramfs# reboot
-sh: /sbin/reboot: Input/output error
```

Info regarding disk (flash) space:
```
root@barreleye:/run/initramfs# df
Filesystem           1K-blocks      Used Available Use% Mounted on
dev                     247708         0    247708   0% /dev
tmpfs                   248988     31288    217700  13% /run
cow                       4096       900      3196  22% /
tmpfs                   248988         0    248988   0% /dev/shm
tmpfs                   248988         0    248988   0% /sys/fs/cgroup
tmpfs                   248988        32    248956   0% /tmp
tmpfs                   248988        20    248968   0% /var/volatile
```

Workaround for rebooting:

1. Upload systemctl again to BMC:

`[build@obmc build]$ scp ./tmp/work/armv5e-openbmc-linux-gnueabi/systemd/1_229+gitAUTOINC+714c62b463-r0/image/bin/systemctl root@172.17.27.92:/usr/sbin/
`

2. Remove original files (that cannot read):

```
root@barreleye:/run/initramfs# rm /bin/systemctl
root@barreleye:/run/initramfs# rm /sbin/reboot
```

3. Use new file instead:
`root@barreleye:/run/initramfs# ln /usr/sbin/systemctl /sbin/reboot
`

4. Reboot
`root@barreleye:/run/initramfs# reboot`

williamspatrick, 2017-03-17:
That sounds like an issue either in the flash driver or the flash chip. Once the corruption happens, if you AC cycle does "reboot" come back or is it still broken? If it is still there, it sounds like a weak cell in your PNOR. In that case v1.99.3 was just "unlucky" in which program hit the bad cell, I suspect. Since we are using a static layout right now, the same adddress will be used for the same contents in the same version no matter how many times you reflash. Do you have a spare PNOR chip you can swap in?

williamspatrick, 2017-03-17:
NOR ... not PNOR. PNOR is the host firmware.

yadron, 2017-03-17:
After rebooting it is not broken. This is being interesting, because I don't think that this is flash error as result (is it often for flash that the same block becomes broken after some hours of working, but not immediately?).

Sorry for using this issue just for talking, but anyway. I have similar question like this issue topic that something broken regarding state/management other scripts.

I've just updated to last version "v1.99.3-93" as is (without any modifications). It is still broken like 1.99.3-48 was, but differently.

Remember, in 1.99.3-38 I did this change:

```
[build@obmc openbmc2]$ diff ~/prev/conf/openbmc-phosphor/openbmc-phosphor.conf ./new/openbmc/meta-phosphor/conf/distro/openbmc-phosphor.conf
35,37c35
<         obmc-host-state-mgmt \
<       obmc-chassis-state-mgmt \
<       obmc-bmc-state-mgmt \
---
>         slp \
```
for fixing _"The name xyz.openbmc_project.State.Host was not provided by any .service files"_ error.

This error happens again exactly after booted in "v1.99.3-93".

Currently with fresh unmodified OpenBMC build the same error:

```
root@barreleye:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-93"
VERSION_ID="v1.99.3-93-gf0a463a"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-93"
BUILD_ID="v1.99.3"

root@barreleye:~# 
root@barreleye:~# 
root@barreleye:~# obmcutil poweroff
Traceback (most recent call last):
  File "/usr/sbin/obmcutil", line 221, in <module>
    obj = bus.get_object(bus_name, obj_path)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name xyz.openbmc_project.State.Host was not provided by any .service files

root@barreleye:~# ps|grep state
  821 root       0:00 /usr/sbin/netman_watch_dns /run/systemd/netif/state
  879 root       0:00 /usr/sbin/phosphor-chassis-state-manager
  883 root       0:00 /usr/sbin/phosphor-bmc-state-manager
  898 root       0:00 /usr/sbin/phosphor-host-state-manager
```

I've already written regarding this error in https://github.com/openbmc/openbmc/issues/1287 and in https://github.com/openbmc/openbmc/issues/1262 _(find by word "xyz.openbmc_project.State.Host" in long text of 1262)_ some days ago.

Now I see that the workaround for fixing it is restarting service "phosphor-host-state-manager" (or maybe just I need to wait for longer time after boot and this restarting not required then):

```
root@barreleye:~# ps|grep state-ma
  879 root       0:00 /usr/sbin/phosphor-chassis-state-manager
  883 root       0:00 /usr/sbin/phosphor-bmc-state-manager
  898 root       0:00 /usr/sbin/phosphor-host-state-manager
  981 root       0:00 grep state-ma
root@barreleye:~# 
root@barreleye:~# kill 898
root@barreleye:~# ps|grep state-ma
[... 80 lines cut ...]
Mar 17 11:34:26 barreleye power_control.exe[877]: Power GPIO 1 power_up outputs
Mar 17 11:34:27 barreleye power_control.exe[877]: Power GPIO power_up[0] = POWER_PIN active LOW
Mar 17 11:34:27 barreleye power_control.exe[877]: Power GPIO 1 reset outputs
Mar 17 11:34:27 barreleye power_control.exe[877]: Power GPIO reset[0] = USB_RESET active LOW
Mar 17 11:34:27 barreleye power_control.exe[877]: Power GPIO 0 pci reset outputs
Mar 17 11:34:27 barreleye power_control.exe[877]: FSI DATA GPIO: FSI_DATA
Mar 17 11:34:27 barreleye power_control.exe[877]: FSI CLK GPIO: FSI_CLK
Mar 17 11:34:27 barreleye power_control.exe[877]: FSI ENABLE GPIO: FSI_ENABLE
Mar 17 11:34:27 barreleye power_control.exe[877]: CRONUS SEL GPIO: CRONUS_SEL
Mar 17 11:34:27 barreleye power_control.exe[877]: Hostctl GPIO optionals: 2
Mar 17 11:34:27 barreleye power_control.exe[877]: Hostctl optional GPIO[0] = BMC_THROTTLE active HIGH
Mar 17 11:34:27 barreleye power_control.exe[877]: Hostctl optional GPIO[1] = IDBTN active LOW
Mar 17 11:34:27 barreleye system_manager.py[797]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('USB_RESET', False)], [], 'FSI_DATA', 'FSI_CLK', 'FSI_ENABLE', 'CRONUS_SEL', [('BMC_THROTTLE', True), ('IDBTN', False)]]
Mar 17 11:34:27 barreleye power_control.exe[877]: GPIO Lookup:  PGOOD = 315,in
Mar 17 11:34:27 barreleye control_host.exe[880]: GPIO Lookup:  FSI_DATA = 297,out
Mar 17 11:34:27 barreleye host_checkstop.exe[901]: GPIO Lookup:  CHECKSTOP = 417,falling
Mar 17 11:34:27 barreleye control_host.exe[880]: GPIO Lookup:  FSI_CLK = 296,out
Mar 17 11:34:28 barreleye power_control.exe[877]: GPIO Lookup:  POWER_PIN = 325,out
Mar 17 11:34:28 barreleye control_host.exe[880]: GPIO Lookup:  FSI_ENABLE = 316,out
Mar 17 11:34:28 barreleye control_host.exe[880]: GPIO Lookup:  CRONUS_SEL = 298,out
Mar 17 11:34:28 barreleye power_control.exe[877]: GPIO Lookup:  USB_RESET = 306,out
Mar 17 11:34:28 barreleye control_host.exe[880]: GPIO Lookup:  BMC_THROTTLE = 367,out
Mar 17 11:34:29 barreleye control_host.exe[880]: GPIO Lookup:  IDBTN = 427,out

```

yadron, 2017-03-20:
I've changed flash chip of BMC in Barreleye. It didn't help. Still:

```
root@barreleye:/run/initramfs#./update
root@barreleye:/run/initramfs#reboot
-sh: /sbin/reboot: Input/output error
root@barreleye:/run/initramfs# dmesg|tail
SQUASHFS error: xz decompression failed, data probably corrupt
SQUASHFS error: squashfs_read_data failed to read block 0xc8770
SQUASHFS error: Unable to read data cache entry [c8770]
SQUASHFS error: Unable to read page, block c8770, size bcb4
SQUASHFS error: Unable to read data cache entry [c8770]
SQUASHFS error: Unable to read page, block c8770, size bcb4
SQUASHFS error: Unable to read data cache entry [c8770]
SQUASHFS error: Unable to read page, block c8770, size bcb4
SQUASHFS error: Unable to read data cache entry [c8770]
SQUASHFS error: Unable to read page, block c8770, size bcb4
SQUASHFS error: Unable to read data cache entry [c8770]
SQUASHFS error: Unable to read page, block c8770, size bcb4
SQUASHFS error: Unable to read data cache entry [c8770]
SQUASHFS error: Unable to read page, block c8770, size bcb4

```
- so this is software problem (probably flash driver), but not flash chip.

Still problems with booting host.

yadron, 2017-03-21:
Continuing my previous message, using the same "v1.99.3-110" I've just updated .pnor (using the same of 13 march like I did before: _13march2017$ scp ./barreleye.pnor root@172.17.27.98:/tmp/; root@barreleye:~# pflash -E -f -i -p /tmp/barreleye.pnor_), and after that poweron/obmc-console at least output something (before updating .pnor I could not see anything at all, but now it does something with output like "STEP 6... again STEP 6... again ... then "98.89788|htmgt|OCCs are not active (rc=0x250A). Attempting OCC Reset" ... then will continue in safe mode ... then ipmi error "Request to get power limit information failed" and other ... then host booted! ), like this that is output of obmc-console-client:

```
 9.37244|HWAS|PRESENT> DIMM[03]=A0A0A0A0A0A0A0A0
  9.37245|HWAS|PRESENT> Membuf[04]=CCCC000000000000
  9.37245|HWAS|PRESENT> Proc[05]=C000000000000000
  9.55206|ISTEP  6. 6
  9.67354|ISTEP  6. 7
 11.48856|ISTEP  6. 8
 11.53531|ISTEP  6. 9
 12.58917|ISTEP  6.10
 12.60856|ISTEP  6.11
 14.32542|ISTEP  6.12
 14.32697|ISTEP  6.13
 14.32747|ISTEP  7. 1
 14.39362|ISTEP  7. 2
 14.50468|ISTEP  7. 3
 14.53661|ISTEP  7. 4
 14.57566|ISTEP  7. 5
 14.79508|ISTEP  7. 6
 14.83284|ISTEP  7. 7
 15.05038|ISTEP  7. 8
 15.21872|ISTEP  7. 9
 15.21932|ISTEP  8. 1
 15.35956|ISTEP  8. 2
 16.18563|ISTEP  8. 3
 16.19542|ISTEP  8. 4
 16.46154|ISTEP  8. 5
 16.46234|ISTEP  8. 6
 17.06995|ISTEP  8. 7
 17.07066|ISTEP  8. 8
 17.11412|ISTEP  9. 1
 17.53677|ISTEP  9. 2
 18.49097|ISTEP 10. 1
 18.73123|ISTEP 10. 2
 19.99643|ISTEP 10. 3
 19.99718|ISTEP 10. 4
 19.99804|ISTEP 10. 5
 19.99862|ISTEP 10. 6
 19.99919|ISTEP 10. 7
 19.99979|ISTEP 10. 8
 20.00042|ISTEP 10. 9
 20.00104|ISTEP 10.10
 20.00172|ISTEP 10.11
 20.00233|ISTEP 10.12
 20.00299|ISTEP 10.13
 20.00361|ISTEP 10.14
 20.00426|ISTEP 11. 1
 20.16735|ISTEP 11. 2
 20.16848|ISTEP 11. 3
 20.32718|ISTEP 11. 4
 20.51911|ISTEP 11. 5
 20.71530|ISTEP 11. 6
 22.83525|ISTEP 11. 7
 22.83649|ISTEP 11. 8
 23.42025|ISTEP 11. 9
 23.42154|ISTEP 11.10
 23.48794|ISTEP 11.11
 23.48860|ISTEP 11.12
 23.48945|ISTEP 11.13
 23.49057|ISTEP 12. 1
 23.60257|ISTEP 12. 2
 23.70088|ISTEP 12. 3
 23.73151|ISTEP 12. 4
 24.11740|ISTEP 12. 5
 24.11839|ISTEP 13. 1
 24.20795|ISTEP 13. 2
 24.28801|ISTEP 13. 3
 24.28900|ISTEP 13. 4
 24.32983|ISTEP 13. 5
[... 447 lines cut ...]
[    5.136212111,7] FLASH: CAPP partition isn't signed
[    5.136213455,7] FLASH: flash subpartition eyecatcher CAPP
[    5.136214707,7] FLASH: Computed CAPP partition size: 40960 (subpart 131306 size 36432 offset 4096)
[    5.150311130,5] STB: sb_verify skipped resource 2, secure_mode=0
[    5.150312683,5] STB: tb_measure skipped resource 2, trusted_mode=0
[    5.150499821,7] FFS: Partition map size: 0x1000
[    5.151916426,7] FLASH: BOOTKERNEL partition doesn't have ECC
[    5.153220910,7] FLASH: BOOTKERNEL partition isn't signed
[    5.153222281,7] FLASH: computed BOOTKERNEL size 15427328
[    5.223845823,5] Chip 0 Found PBCQ0 at /xscom@3fc0000000000/pbcq@2012000
[    5.243543011,5] Chip 0 Found PBCQ1 at /xscom@3fc0000000000/pbcq@2012400
[    5.302894562,5] Chip 0 Found PBCQ2 at /xscom@3fc0000000000/pbcq@2012800
[    5.362246863,5] Chip 8 Found PBCQ0 at /xscom@3fc4000000000/pbcq@2012000
[    5.401923147,5] Chip 8 Found PBCQ1 at /xscom@3fc4000000000/pbcq@2012400
[    5.418859871,5] Chip 8 Found PBCQ2 at /xscom@3fc4000000000/pbcq@2012800
[    6.068591926,3] PHB#0000: Base location code not found !
[    6.437646751,3] PHB#0001: Base location code not found !
[    8.266841409,3] PHB#0002: Base location code not found !
[    9.171921579,3] PHB#0020: Base location code not found !
[    9.436951652,3] PHB#0021: Base location code not found !
[   10.217442722,3] PHB#0022: Base location code not found !

[and host booted at last]

```

gtmills, 2018-05-24:
@yadron Still a problem or can this be closed?

gtmills, 2018-08-28:
Barreleye was removed here: https://gerrit.openbmc-project.xyz/#/c/openbmc/openbmc/+/12072/. @geissonator Can this be closed or is this still tracking something?

geissonator, 2018-09-04:
This bug morphed around a bit.  I think the initial issue of phosphor-state-manager depending on system_manager.py has now been fixed so closing this out.

closed 2018-09-04 by geissonator