openbmc/openbmc #1326: IPMID code dumped on Poweron [Witherspoon]

gkeishin, 2017-03-16:
```
root@witherspoon:~# ls /tmp/core_ipmid.852
/tmp/core_ipmid.852
root@witherspoon:~#

Feb 22 07:12:45 witherspoon ipmid[852]: Error updating inventory
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Serial Number] : Type Code=[ASCII+Latin] : Len = [16]terminate called after throwing an instance of 'std::logic_error'
```
`Driver Info`

```
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/630/
- [x] VERSION_ID : v1.99.3-93-gf0a463a
- [x] Kernel     : Linux witherspoon 4.7.10-c72ced5a19ceae19c7d792ae3e982bee5607dab4 #1 Wed Mar 15 22:22:06 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.3
        op-build-v1.15-15-gce29449
        buildroot-2017.02-4-gc28573e
        skiboot-28e96ff
        hostboot-86bae1c
        linux-4.10.1-p976d063
        petitboot-v1.4.1-p3ec3d6e
        witherspoon-xml-065aaa2
        occ-4141b5f
        hostboot-binaries-4b44090
        capp-ucode-1bb7503
```

- [x]  Core dump file  
[core_ipmid.852.gz](https://github.com/openbmc/openbmc/files/846814/core_ipmid.852.gz)
- [x] Journal data 
[journald_v1.99.3-93-gf0a463a_core_ipmid.852.txt](https://github.com/openbmc/openbmc/files/846815/journald_v1.99.3-93-gf0a463a_core_ipmid.852.txt)

gkeishin, 2017-03-16:
@tomjoseph83 is looking into this.. He is aware of the failure.

ratagupt, 2017-03-16:
Seems host is not sending the data as per the protocol.
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Serial Number] : Type Code=[ASCII+Latin] : Len = [16]terminate called after throwing an instance of 'std::logic_error'
Feb 22 07:16:21 witherspoon ipmid[852]:   what():  basic_string::_M_construct null not valid
Feb 22 07:16:21 witherspoon ipmid[852]:  : Val = [000000000000]

Here I can see that len is 16 bytes but host sendsthe 12 bytes of data which is causing the memory access problem.

            printf ("_append_to_dict: VPD Key = [%s] : Type Code=[ASCII+Latin]"
                    " : Len = [%d] : Val = [%s]\n",
                    vpd_key_names [vpd_key_id], vpd_val_len, &vpd_key_val[1]);
            info[vpd_key_id] = std::make_pair(
                                   vpd_key_names[vpd_key_id],
                                   std::string(vpd_key_val + 1,
                                               vpd_key_val + 1 + type_length));

ratagupt, 2017-03-16:
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Serial Number] : Type Code=[ASCII+Latin] : Len = [16]terminate called after throwing an instance of 'std::logic_error'
Feb 22 07:16:21 witherspoon ipmid[852]:   what():  basic_string::_M_construct null not valid
Feb 22 07:16:21 witherspoon ipmid[852]:  : Val = [000000000000]
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Part Number] : Type Code=[ASCII+Latin] : Len = [16] : Val = [0000000]
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [FRU File ID] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 1] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 2] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 3] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 4] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 5] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 6] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 7] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 8] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: fru data size:[56], area offset:[8], area_size:[48]
Feb 22 07:16:21 witherspoon ipmid[852]: Successfully verified area checksum. offset:[8]
Feb 22 07:16:21 witherspoon ipmid[852]: SUCCESS: Populated FRU areas for:[/tmp/ipmifru04]
>>>>>>Core happened during parsing of fruid=04
>>>>>>Field Name = Serial Number, Expected Length = 16,Actual Data = 12 bytes long.
>>>>>>Field Name = PartNumber , Exptceted Length = 16,Actual Data = 7 bytes long.
>>>>>>This difference is causing the core.

ratagupt, 2017-03-16:
Could be a bad pnor problem.

williamspatrick, 2017-03-16:
We need to protect ourselves against a malicious host.

geissonator, 2017-03-16:
Yeah, so lets use this bug to update our code to handle this error path and open a separate bug on hostboot to fix the data (https://github.com/open-power/hostboot/issues)

gkeishin, 2017-03-16:
Opened one for hostboot https://github.com/open-power/hostboot/issues/94

closed 2017-08-10 by gkeishin