openbmc/openbmc #1328: Power On failed - PGOOD service failed to start to Standby [ Witherspoon ]

gkeishin, 2017-03-17:
`Driver Info`

```
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/633/
- [x] VERSION_ID : v1.99.3-106-gc6c2998
- [x] Kernel     : Linux witherspoon 4.7.10-c72ced5a19ceae19c7d792ae3e982bee5607dab4 #1 Thu Mar 16 22:19:27 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.3
IBM-witherspoon-ibm-OP9_v1.15_1.3
        op-build-v1.15-15-gce29449
        buildroot-2017.02-4-gc28573e
        skiboot-28e96ff
        hostboot-86bae1c
        linux-4.10.1-p976d063
        petitboot-v1.4.1-p3ec3d6e
        witherspoon-xml-065aaa2
        occ-4141b5f
        hostboot-binaries-4b44090
        capp-ucode-1bb7503
```

```
Feb 22 06:06:34 witherspoon systemd[1]: op-reset-pgood-check@0.service: Main process exited, code=exited, status=1/FAILURE
Feb 22 06:06:34 witherspoon systemd[1]: Failed to start Check Chassis0 pgood on reset.
Feb 22 06:06:34 witherspoon systemd[1]: Dependency failed for Chassis0 (Reset Check).
Feb 22 06:06:34 witherspoon systemd[1]: obmc-chassis-reset@0.target: Job obmc-chassis-reset@0.target/start failed with result 'dependency'.
Feb 22 06:06:34 witherspoon systemd[1]: Dependency failed for Create file to indicate chassis0 pgood is on.
Feb 22 06:06:34 witherspoon systemd[1]: Dependency failed for Start chassis0 on after BMC reset.
Feb 22 06:06:34 witherspoon systemd[1]: op-reset-chassis-on@0.service: Job op-reset-chassis-on@0.service/start failed with result 'dependency'.
Feb 22 06:06:34 witherspoon systemd[1]: op-reset-set-power-on@0.service: Job op-reset-set-power-on@0.service/start failed with result 'dependency'.
Feb 22 06:06:34 witherspoon systemd[1]: op-reset-pgood-check@0.service: Unit entered failed state.
Feb 22 06:06:34 witherspoon systemd[1]: op-reset-pgood-check@0.service: Failed with result 'exit-code'
```

```
root@witherspoon:~# systemctl status op-reset-chassis-on@0.service | cat
● op-reset-chassis-on@0.service - Start chassis0 on after BMC reset
   Loaded: loaded (/lib/systemd/system/obmc-chassis-reset@0.target.requires/../op-reset-chassis-on@.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

Feb 22 06:06:34 witherspoon systemd[1]: Dependency failed for Start chassis0 on after BMC reset.
Feb 22 06:06:34 witherspoon systemd[1]: op-reset-chassis-on@0.service: Job op-reset-chassis-on@0.service/start failed with result 'dependency'.
root@witherspoon:~# systemctl status op-reset-pgood-check@0.service | cat
● op-reset-pgood-check@0.service - Check Chassis0 pgood on reset
   Loaded: loaded (/lib/systemd/system/obmc-chassis-reset@0.target.requires/../op-reset-pgood-check@.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2017-02-22 06:06:34 UTC; 52s ago
  Process: 893 ExecStart=/bin/sh -c busctl get-property `mapper get-service /org/openbmc/control/power%i` /org/openbmc/control/power%i org.openbmc.control.Power pgood | sed 's/i *[1]/on/' | grep on (code=exited, status=1/FAILURE)
 Main PID: 893 (code=exited, status=1/FAILURE)
```

- [x] Journal log  
[journald_v1.99.3-106-gc6c2998_service_EXIT.txt](https://github.com/openbmc/openbmc/files/849959/journald_v1.99.3-106-gc6c2998_service_EXIT.txt)

gkeishin, 2017-03-17:
we see PGOOD timeout and even with script to fix it, the poweron will not progress

```
root@witherspoon:~# obmcutil poweron
root@witherspoon:~# jouucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
rnalctl -f
-- Logs begin at Wed 2017-02-22 06:04:59 UTC. --
Feb 22 06:10:39 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Main process exited, code=exited, status=1/FAILURE
Feb 22 06:10:39 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit entered failed state.
Feb 22 06:10:39 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Failed with result 'exit-code'.
Feb 22 06:10:39 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit not needed anymore. Stopping.
Feb 22 06:10:39 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Feb 22 06:10:39 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Feb 22 06:10:39 witherspoon kernel[759]: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Feb 22 06:10:40 witherspoon systemd[1]: Started Disable UCD VCS rails.
Feb 22 06:10:40 witherspoon systemd[1]: Starting Perform AVS bus workaround on VRMs...
Feb 22 06:10:40 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Feb 22 06:10:40 witherspoon systemd[1]: Started Perform AVS bus workaround on VRMs.
Feb 22 06:10:41 witherspoon systemd[1]: Starting Disable the AVS bus on the VRMs...
Feb 22 06:10:41 witherspoon systemd[1]: Started Disable the AVS bus on the VRMs.
Feb 22 06:10:41 witherspoon systemd[1]: Starting Apply voltage overrides to VRMs...
Feb 22 06:10:41 witherspoon vrm-control.sh[957]: rail        set        read        current
Feb 22 06:10:41 witherspoon vrm-control.sh[957]: -------        ------- ------- -------
Feb 22 06:10:42 witherspoon vrm-control.sh[957]: vdna        0.898V        0.004V        0.000A
Feb 22 06:10:43 witherspoon vrm-control.sh[957]: vdnb        0.898V        0.004V        0.000A
Feb 22 06:10:43 witherspoon systemd[1]: Started Apply voltage overrides to VRMs.
Feb 22 06:10:43 witherspoon systemd[1]: Starting Enable the AVS bus on VRMs...
Feb 22 06:10:43 witherspoon systemd[1]: Started Enable the AVS bus on VRMs.
Feb 22 06:10:43 witherspoon systemd[1]: Starting Disable VCS before Witherspoon power on...
Feb 22 06:10:44 witherspoon systemd[1]: Started Disable VCS before Witherspoon power on.
Feb 22 06:10:44 witherspoon systemd[1]: Reached target Power0 On (Pre).
Feb 22 06:10:44 witherspoon systemd[1]: Starting Start Power0...
Feb 22 06:10:44 witherspoon power_control.exe[866]: PowerControl: setting power up SOFTWARE_PGOOD to 1
Feb 22 06:10:44 witherspoon power_control.exe[866]: PowerControl: setting power up BMC_POWER_UP to 1
Feb 22 06:10:44 witherspoon systemd[1]: Started Start Power0.
Feb 22 06:10:44 witherspoon timemanager[899]:  PGOOD has changed..
Feb 22 06:10:44 witherspoon system_manager.py[826]: Running System State: HOST_POWERING_ON
Feb 22 06:10:58 witherspoon power_control.exe[866]: ERROR PowerControl: Pgood poll timeout
Feb 22 06:10:58 witherspoon timemanager[899]:  PGOOD has changed..

root@witherspoon:~# obmcutil poweroff
root@witherspoon:~# sh /tmp/pgood_fix.sh
Finished with 1
Finished with 2
Finished with 3
Finished with 4
Finished with 5
Finished with 6
Finished with 7
Finished with 8
Finished with 9
Finished with 10
Finished with 11
Finished with 12
Finished with 13
Finished with 14
Finished with 15
ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
root@witherspoon:~# obmcutil poweron
root@witherspoon:~# jucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
ournalctl -f
-- Logs begin at Wed 2017-02-22 06:04:59 UTC. --
Feb 22 06:13:01 witherspoon kernel[759]: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Feb 22 06:13:02 witherspoon systemd[1]: Started Disable UCD VCS rails.
Feb 22 06:13:02 witherspoon systemd[1]: Starting Perform AVS bus workaround on VRMs...
Feb 22 06:13:02 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Feb 22 06:13:02 witherspoon systemd[1]: Started Perform AVS bus workaround on VRMs.
Feb 22 06:13:02 witherspoon systemd[1]: Starting Disable the AVS bus on the VRMs...
Feb 22 06:13:03 witherspoon systemd[1]: Started Disable the AVS bus on the VRMs.
Feb 22 06:13:03 witherspoon systemd[1]: Starting Apply voltage overrides to VRMs...
[... 6 lines cut ...]
Feb 22 06:13:05 witherspoon systemd[1]: Started Enable the AVS bus on VRMs.
Feb 22 06:13:05 witherspoon systemd[1]: Starting Disable VCS before Witherspoon power on...
Feb 22 06:13:05 witherspoon systemd[1]: Started Disable VCS before Witherspoon power on.
Feb 22 06:13:06 witherspoon systemd[1]: Reached target Power0 On (Pre).
Feb 22 06:13:06 witherspoon systemd[1]: Starting Start Power0...
Feb 22 06:13:06 witherspoon phosphor-hwmon-readd[1154]: Input/output error
Feb 22 06:13:06 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Main process exited, code=exited, status=1/FAILURE
Feb 22 06:13:06 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit entered failed state.
Feb 22 06:13:06 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Failed with result 'exit-code'.
[... cut ...]
Feb 22 06:13:07 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit entered failed state.
Feb 22 06:13:07 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Failed with result 'exit-code'.
Feb 22 06:13:07 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Service hold-off time over, scheduling restart.
Feb 22 06:13:07 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Feb 22 06:13:07 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
```

gkeishin, 2017-03-17:
`I think the above crash have implications to the entire powering on`

```
Feb 22 06:13:51 witherspoon systemd[1]: Starting Run VCS workaround on host0...
Feb 22 06:13:51 witherspoon openpower-proc-control[3131]: Usage: /usr/sbin/openpower-proc-control [action]
Feb 22 06:13:51 witherspoon openpower-proc-control[3131]:    actions:
Feb 22 06:13:51 witherspoon systemd[1]: vcs_workaround@0.service: Main process exited, code=exited, status=255/n/a
Feb 22 06:13:51 witherspoon systemd[1]: Failed to start Run VCS workaround on host0.
Feb 22 06:13:51 witherspoon systemd[1]: Dependency failed for Chassis0 (Power On).
Feb 22 06:13:51 witherspoon systemd[1]: Dependency failed for Chassis0 (Start).
Feb 22 06:13:51 witherspoon systemd[1]: obmc-chassis-start@0.target: Job obmc-chassis-start@0.target/start failed with result 'dependency'.
Feb 22 06:13:51 witherspoon systemd[1]: obmc-chassis-start@0.target: Triggering OnFailure= dependencies.
Feb 22 06:13:51 witherspoon systemd[1]: obmc-power-chassis-on@0.target: Job obmc-power-chassis-on@0.target/start failed with result 'dependency'.
Feb 22 06:13:51 witherspoon systemd[1]: obmc-power-chassis-on@0.target: Triggering OnFailure= dependencies.
Feb 22 06:13:52 witherspoon systemd[1]: vcs_workaround@0.service: Unit entered failed state.
Feb 22 06:13:52 witherspoon systemd[1]: vcs_workaround@0.service: Failed with result 'exit-code'
```

```
root@witherspoon:~# systemctl status vcs_workaround@0.service | cat
● vcs_workaround@0.service - Run VCS workaround on host0
   Loaded: loaded (/lib/systemd/system/obmc-power-chassis-on@0.target.requires/../vcs_workaround@.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2017-02-22 06:14:15 UTC; 2min 17s ago
  Process: 3752 ExecStart=/usr/sbin/openpower-proc-control vcsWorkaround (code=exited, status=255)
 Main PID: 3752 (code=exited, status=255)

Feb 22 06:14:14 witherspoon systemd[1]: Starting Run VCS workaround on host0...
Feb 22 06:14:15 witherspoon openpower-proc-control[3752]: Usage: /usr/sbin/openpower-proc-control [action]
Feb 22 06:14:15 witherspoon openpower-proc-control[3752]:    actions:
Feb 22 06:14:15 witherspoon systemd[1]: vcs_workaround@0.service: Main process exited, code=exited, status=255/n/a
Feb 22 06:14:15 witherspoon systemd[1]: Failed to start Run VCS workaround on host0.
Feb 22 06:14:15 witherspoon systemd[1]: vcs_workaround@0.service: Unit entered failed state.
Feb 22 06:14:15 witherspoon systemd[1]: vcs_workaround@0.service: Failed with result 'exit-code'.
Feb 22 06:14:22 witherspoon systemd[1]: Stopped Run VCS workaround on host0.

```

- [x]  Journal log 
[journald_v1.99.3-106-gc6c2998_VCS.txt](https://github.com/openbmc/openbmc/files/850199/journald_v1.99.3-106-gc6c2998_VCS.txt)

williamspatrick, 2017-03-17:
Looks like there is a problem both with the pgood reset/reload check and the cvs workaround. Likely two different issues.

williamspatrick, 2017-03-17:
op-proc-control seems like a compile problem of not including the right chip procedures.

geissonator, 2017-03-17:
The failing of obmc-chassis-reset@0.target (and it’s services) is expected after a bmc reboot when the chassis power was not left on prior to the reboot.  The target failure just indicates that chassis power was not on prior to the reboot.

#1321 tracks us not recovering from a pgood timeout when we hit it

This issue should track the fact we don't boot, which I assume is related to the vcs service you noted above:
```
Feb 22 06:14:14 witherspoon systemd[1]: Starting Run VCS workaround on host0...
Feb 22 06:14:15 witherspoon openpower-proc-control[3752]: Usage: /usr/sbin/openpower-proc-control [action]
Feb 22 06:14:15 witherspoon openpower-proc-control[3752]:    actions:
Feb 22 06:14:15 witherspoon systemd[1]: vcs_workaround@0.service: Main process exited, code=exited, status=255/n/a
Feb 22 06:14:15 witherspoon systemd[1]: Failed to start Run VCS workaround on host0.
Feb 22 06:14:15 witherspoon systemd[1]: vcs_workaround@0.service: Unit entered failed state.
Feb 22 06:14:15 witherspoon systemd[1]: vcs_workaround@0.service: Failed with result 'exit-code'.
Feb 22 06:14:22 witherspoon systemd[1]: Stopped Run VCS workaround on host0.
```

spinler, 2017-03-17:
I have 2 fixes:
https://gerrit.openbmc-project.xyz/#/c/3119/    #fix for building in bitbake
https://gerrit.openbmc-project.xyz/#/c/3120/    #fix for the FSI slave path changing out from under me

williamspatrick, 2017-03-17:
Commits are merged and master is updated via @7f0646c

closed 2017-03-17 by williamspatrick