openbmc/openbmc #1329: power supply presence detection

spinler, 2017-03-17:
The Witherspoon power supplies have GPIOs for presence detects wired to the BMC.   We need to monitor those and add the PSs to the inventory when they are present.  If they are hot removed, we'd need to mark them as not present.

In addition the power supplies have serial numbers that should listed in the inventory, but we'd need device driver support to get them.

williamspatrick, 2017-06-02:
I'm estimating as 2 for both the presence and serial numbers, but that would include driver enhancements.

bradbishop, 2017-06-13:
Moved a point to serial number story #1782.

gtmills, 2017-06-13:
- [ ] Create phosphor-gpio-presence app in phosphor-gpio-monitor
- [ ] Have phosphor-gpio-presence determine presence and update inventory based on gpio presence line (passed in)
- [ ] Watches for gpio state changes and update inventory 
- [ ] Start 2 instances of phosphor-gpio-presence service, one for each power supply for Witherspoon

gtmills, 2017-06-22:
Need https://github.com/openbmc/openbmc/issues/1783

gtmills, 2017-07-13:
07/13 - Working on Recipe

rfrandse, 2017-07-16:
<https://gerrit.openbmc-project.xyz/5190> Add GPIO Presence
Resolves  #1329 power supply presence detection

closed 2017-07-31 by williamspatrick