openbmc/openbmc #1330: DT bindings for occ-hwmon

bradbishop, 2017-03-17:
Under the current driver impl tempN_input for a given N, changes from one boot to the next.  For example, temp2_input may correlate to DIMM2 in one case and CORE4 in another.

A proper driver would instantiate the same number of channels every time, and the meaning of each channel is always the same.

I believe some DT bindings are required here so that the driver can know which ipmi sensors correlate to which hwmon channels.

@jk-ozlabs @shenki @amboar @eddiejames Please poke holes.

eddiejames, 2017-04-07:
- [ ] read entries from device tree under the OCC bus device (i2c or sbe) in the driver
- [ ] map hwmon entry number to sensor id when reading a sensor
- [ ] test it.

bradbishop, 2017-05-16:
Had a meeting with @jk and @eddiejames this evening and the outcome of that was that occ-hwmon will just put the IPMI sensor number in the label property.  Renaming accordingly.

bradbishop, 2017-05-16:
Archiving the old description here...

```Under the current driver impl tempN_input for a given N, changes from one boot to the next.  For example, temp2_input may correlate to DIMM2 in one case and CORE4 in another.

A proper driver would instantiate the same number of channels every time, and the meaning of each channel is always the same.

I believe some DT bindings are required here so that the driver can know which ipmi sensors correlate to which hwmon channels.

@jk-ozlabs @shenki @amboar @eddiejames Please poke holes.```

rfrandse, 2017-05-31:
Code Done 
https://github.com/openbmc/openbmc/issues/1693
tracks the upstreaming process

closed 2017-05-31 by rfrandse