openbmc/openbmc #1334: Sprint 17.03.22 - Committed

rfrandse, 2017-03-17:


saqibkh, 2017-03-21:
OpenBMC

rfrandse, 2017-03-22:
-------------------------------------------------------------------------------------------------
***Squad 2 -***
-------------------------------------------------------------------------------------------------
-------------------------------------------------------------------------------------------------
**Matt:spinler -** 
**17.01.11:**
- https://github.com/openbmc/openbmc/issues/629 (integrate device tree into kernel) - waiting

**17.03.08:**
- https://github.com/openbmc/openbmc/issues/1209 (MRW fan work) - up in gerrit

**17.03.22:**
- \<power / thermal>
- watchdog?  or #1342 / #1343 with brandon

-------------------------------------------------------------------------------------------------
**Matt B.:msbarth -** 

**17.02.22:**
- https://github.com/openbmc/openbmc/issues/962 (Add support for fanx_target hwmons)

**17.03.22:**
- #1347 (Handle application restarts where fan speeds should be set to max.) 

-------------------------------------------------------------------------------------------------
**Brad:bradbishop -**

**17.02.22:**
-  https://github.com/openbmc/openbmc/issues/1176  (power/thermal - iio to hwmon bridge) - what's next?
- https://github.com/openbmc/openbmc/issues/1130 (Phosphor Hwmon Poller Crashed )  - up for review 

**17.03.08:**
- https://github.com/openbmc/openbmc/issues/1257 (sbefifo driver base)

**17.03.22:**
- #1303 (inventory Monitoring - Determine design and yaml input layout) 

-------------------------------------------------------------------------------------------------
**Chris -**
**17.02.22:**
- https://github.com/openbmc/openbmc/issues/1207 (openFSI: segfault on reboot when Cronus is enabled)
- https://github.com/openbmc/openbmc/issues/1226 (OpenFSI is segfaulting) - patch submitted 
- https://github.com/openbmc/openbmc/issues/1240 (OpenFSI seg fault on P9 # 2 slave read via raw sysfs file) - patches submitted
- https://github.com/openbmc/openbmc/issues/1259 (OpenFSI on re-scan hub devices are not removed from /sys/devices) - patches submitted

**17.03.08:**
- https://github.com/openbmc/openbmc/issues/1219 (OpenFSI targeted error handling for mulitple hub links)

**17.03.22:**
- #1222 ( OpenFSI: interrupt support for hub links.)

------------------------------------------------------------------------------------------------
**Eddie:eddiejames -**
**16.12.28:**
- https://github.com/openbmc/openbmc/issues/624 (initial occ intf refactor) - waiting on community
- https://github.com/openbmc/openbmc/issues/625 (p9 occ data) - waiting on community

**17.01.11:**
- https://github.com/openbmc/openbmc/issues/1083 - patches in mailing list

**17.02.08:**
- https://github.com/openbmc/openbmc/issues/1004 (I2C migration to official interfaces)

**17.03.22:**
- #1258 (sbefifo in kernel api)
- #1348 (OCC: force hwmon attribute numbering to conform to dts sensor definitions)

-------------------------------------------------------------------------------------------------
**Brandon Wyman -**
**17.03.22:**
- \<power / thermal>
- #1342 / #1343

-------------------------------------------------------------------------------------------------
**Dinesh Chinari  -**
- https://github.com/openbmc/openbmc/issues/753 (Add rsync to the image) - in review

- https://github.com/openbmc/openbmc/issues/1315 (Ensure uboot fw_env update script uses new inventory object)

rfrandse, 2017-03-22:
-------------------------------------------------------------------------------------------------
***Squad 1 -***
-------------------------------------------------------------------------------------------------
-------------------------------------------------------------------------------------------------
**Deepak:** 

**17.03.08:**
- #1040 (Add support for existing IPMI OCC pass through command) - in review
- #1167 ( (Build openpower-dbus-interfaces) - in review

**17.03.22:**
-  \<mark error as resolved>
- #1349 (rest server handling multiple objects)
-------------------------------------------------------------------------------------------------
**Tom:tomjoseph83 -**
**17.01.25:**
- #856 (Attach RMCP session to SOL, part 1)
 
**17.03.08:**
- #857 (Attach RMCP session to SOL, part 2)
- <help new people with https://github.com/openbmc/openbmc/issues/921 >

**17.03.22:**

-------------------------------------------------------------------------------------------------
**Ratan:ratagupt -**
**17.01.25:**
- #985 (Avahi fails to start on Zaius) - low priority
 
**17.03.08:**
- https://github.com/openbmc/openbmc/issues/1284 (Define the dbus interfaces for network manager)
- https://github.com/openbmc/openbmc/issues/1293 (n/w d-bus interface implementation)

**17.03.22:**
- #1283 (recipe files for non mrw systems)
- #665 (FRU vpd)

-------------------------------------------------------------------------------------------------
**Vishwa:vishwabmc -**
**17.01.11:**
- https://github.com/openbmc/openbmc/issues/802 (move all existing code to new LED interfaces)
    - 03/13: Merged. Story can be closed.
    - 03/17: Some gaps that Vishwa needs to close on, once he's back.

- https://github.com/openbmc/openbmc/issues/930 (soft power off in host ipmi with watchdog timers)
    - https://gerrit.openbmc-project.xyz/#/c/1972/
    - 03/14: Patrick reviewed. Vishwa has comments to address.

**17.03.08:**
- https://github.com/openbmc/openbmc/issues/766 (host checkstop refactor)

**17.03.22:**
- <carry over>
-------------------------------------------------------------------------------------------------
**Jayanth-**
**17.03.08:**
- #1213 (occ active sensors)

**17.03.22:**
- #868 (BMC dump story break down and phosphor-dbus-interface definition)

-------------------------------------------------------------------------------------------------
**Dhruv-**
**17.03.08:**
- #1213 (occ active sensors)

**17.03.22:**
- #1321 (pgood polling is broken after error)
- #941 (assert fault LED's when association is set and cleared)

-------------------------------------------------------------------------------------------------
**Devender-**
**17.03.22:**
- #1155 (friendlier create and commit interface for error logging)

dkodihal, 2017-03-23:
Squad 1

Deepak
17.03.08
1040 - Deepak needs to rework based on Matt S' and Ratan's comments.
1161 - Deepak needs to rework based on Matt S' and Ratan's comments.
17.03.22
1349 (rest server handling multiple objects) - Deepak needs to create task list, before that need discussion with Brad and Adriana.
1379 (Mark error log as resolved) - Please assign to Deepak. Estimate 2 story points. Task list added.

Tom
17.01.25
856 - Deepak reviewed. Tom re-working and pushing more commits for this one.
17.03.08
857 - coding in progress.

Ratan
17.03.08
1284 (Define the dbus interfaces for network manager) - All relevant code up for review. Deepak and Brad to review.
1293 (n/w d-bus interface implementation) - Please move to 03.22 sprint. This needs to be resized. Should be 2 points.
17.03.22
1283 (recipe files for non mrw systems) - This was high prio, so Ratan did it as part of 03.08. Can this be moved to be part of the 03.08 sprint? Code up for review. Deepak and Tom to review.
665 (FRU vpd)

Vishwa
802 (move all existing code to new LED interfaces) - Vishwa reworking.
930 (soft power off in host ipmi with watchdog timers) - Vishwa reworked. Patrick to get back.
766 (host checkstop refactor) - Vishwa needs new repo. Currently code uploaded to skeleton. Deepak reviewing.

Jayanth, Dhruv:
1213 - Code up for review. Deepak and Tom to review.

Jayanth
03.22
Please assign
https://github.com/openbmc/openbmc/issues/1377
https://github.com/openbmc/openbmc/issues/1378
1 sp each
Task lists created.

Dhruv
03.22
https://github.com/openbmc/openbmc/issues/1321
Needs sizing - 2 points. Task list created.

closed 2017-04-18 by rfrandse