openbmc/openbmc #1336: BMC host watchdog cuts power to host despite host reaching hostboot

amboar, 2017-03-20:
Current master reboots the host despite the host successfully beginning to work its way through IPL.

On the host console:
```
--== Welcome to Hostboot hostboot-86bae1c/hbicore.bin ==--

  4.38459|secure|SecureROM invalid - skipping functionality
 11.83259|ISTEP  6. 5
 12.39218|ISTEP  6. 6
 12.39344|ISTEP  6. 7
 13.46392|HWAS|PRESENT> DIMM[03]=F900000000000000
 13.46393|HWAS|PRESENT> Proc[05]=8000000000000000
 13.74313|ISTEP  6. 8
 13.79454|ISTEP  6. 9
 14.43132|ISTEP  6.10
 14.48283|ISTEP  6.11
 14.48340|ISTEP  6.12
 14.71297|ISTEP  7. 2
 16.64959|ISTEP  7. 3
 17.20172|ISTEP  7. 4
 19.91795|ISTEP  7. 5
 20.02034|ISTEP  8. 1
 20.24963|ISTEP  8. 2
 20.27605|ISTEP  8. 3
 20.27762|ISTEP  8. 4
 20.40470|ISTEP  8. 5
 20.43081|ISTEP  8. 6
 20.43386|ISTEP  8. 7
 20.58604|ISTEP  8. 8
 20.63660|ISTEP  8. 9
 20.76469|ISTEP  8.10
 20.76632|ISTEP  8.11
 20.76697|ISTEP  8.12
 21.07234|ISTEP  8.13
 21.07424|ISTEP  8.14
 21.10048|ISTEP  8.15
 21.10248|ISTEP  8.16
 21.25532|ISTEP  8.17
 21.37550|ISTEP  8.18
 21.37700|ISTEP  8.19
 21.40419|ISTEP  8.20
 21.40570|ISTEP  9. 1
 21.53281|ISTEP  9. 2
 21.55916|ISTEP  9. 3
 21.56065|ISTEP  9. 4
 21.58829|ISTEP  9. 5
 21.58968|ISTEP  9. 6
 21.69096|ISTEP  9. 7
 21.74305|ISTEP  9. 8
 21.77011|ISTEP 10. 1
 22.27427|ISTEP 10. 2
```

And then nothing.

Whilst [in the journal](https://github.com/openbmc/openbmc/files/853711/journal.txt):

```
...
Feb 22 06:12:37 witherspoon phosphor-host-state-manager[864]: Recieved signal that host is running
Feb 22 06:12:37 witherspoon phosphor-host-state-manager[864]: Change to Host State
Feb 22 06:12:57 witherspoon ipmid[854]: No registered handlers for NetFn:[0x6], Cmd:[0x36] trying Wilcard implementation
Feb 22 06:12:57 witherspoon ipmid[854]: No Registered handlers for NetFn:[0x6],Cmd:[0xFF]
Feb 22 06:12:57 witherspoon ipmid[854]: ERROR:[0xC1] handling NetFn:[0x6], Cmd:[0x36]
Feb 22 06:12:59 witherspoon ipmid[854]: No registered handlers for NetFn:[0x6], Cmd:[0x6] trying Wilcard implementation
Feb 22 06:12:59 witherspoon ipmid[854]: No Registered handlers for NetFn:[0x6],Cmd:[0xFF]
Feb 22 06:12:59 witherspoon ipmid[854]: ERROR:[0xC1] handling NetFn:[0x6], Cmd:[0x6]
Feb 22 06:12:59 witherspoon ipmid[854]: No registered handlers for NetFn:[0x6], Cmd:[0x24] trying Wilcard implementation
Feb 22 06:12:59 witherspoon ipmid[854]: No Registered handlers for NetFn:[0x6],Cmd:[0xFF]
Feb 22 06:12:59 witherspoon ipmid[854]: ERROR:[0xC1] handling NetFn:[0x6], Cmd:[0x24]
Feb 22 06:13:00 witherspoon ipmid[854]: No registered handlers for NetFn:[0x6], Cmd:[0x22] trying Wilcard implementation
Feb 22 06:13:00 witherspoon ipmid[854]: No Registered handlers for NetFn:[0x6],Cmd:[0xFF]
Feb 22 06:13:00 witherspoon ipmid[854]: ERROR:[0xC1] handling NetFn:[0x6], Cmd:[0x22]
Feb 22 06:13:00 witherspoon ipmid[854]: No registered handlers for NetFn:[0x4], Cmd:[0x30] trying Wilcard implementation
Feb 22 06:13:00 witherspoon ipmid[854]: No Registered handlers for NetFn:[0x4],Cmd:[0xFF]
Feb 22 06:13:00 witherspoon ipmid[854]: ERROR:[0xC1] handling NetFn:[0x4], Cmd:[0x30]
Feb 22 06:13:00 witherspoon ipmid[854]: No registered handlers for NetFn:[0x6], Cmd:[0x22] trying Wilcard implementation
Feb 22 06:13:00 witherspoon ipmid[854]: No Registered handlers for NetFn:[0x6],Cmd:[0xFF]
Feb 22 06:13:00 witherspoon ipmid[854]: ERROR:[0xC1] handling NetFn:[0x6], Cmd:[0x22]
Feb 22 06:13:00 witherspoon ipmid[854]: No registered handlers for NetFn:[0x6], Cmd:[0x22] trying Wilcard implementation
Feb 22 06:13:00 witherspoon ipmid[854]: No Registered handlers for NetFn:[0x6],Cmd:[0xFF]
Feb 22 06:13:00 witherspoon ipmid[854]: ERROR:[0xC1] handling NetFn:[0x6], Cmd:[0x22]
Feb 22 06:13:01 witherspoon ipmid[854]: Incomplete Fru file.. Size:[57]
Feb 22 06:13:01 witherspoon ipmid[854]: Populating FRU areas failed for:[1]
Feb 22 06:13:01 witherspoon ipmid[854]: No registered handlers for NetFn:[0x4], Cmd:[0x30] trying Wilcard implementation
Feb 22 06:13:01 witherspoon ipmid[854]: No Registered handlers for NetFn:[0x4],Cmd:[0xFF]
...
Feb 22 06:13:07 witherspoon ipmid[854]: ERROR:[0xC1] handling NetFn:[0x4], Cmd:[0x30]
Feb 22 06:13:07 witherspoon ipmid[854]: No registered handlers for NetFn:[0x4], Cmd:[0x30] trying Wilcard implementation
Feb 22 06:13:07 witherspoon ipmid[854]: No Registered handlers for NetFn:[0x4],Cmd:[0xFF]
Feb 22 06:13:07 witherspoon ipmid[854]: ERROR:[0xC1] handling NetFn:[0x4], Cmd:[0x30]
Feb 22 06:13:07 witherspoon host_watchdog.exe[764]: Polling watchdog: 0
Feb 22 06:13:07 witherspoon chassis_control.py[789]: Watchdog Error, Going to quiesce
Feb 22 06:13:07 witherspoon ipmid[854]: No registered handlers for NetFn:[0x4], Cmd:[0x30] trying Wilcard implementation
Feb 22 06:13:07 witherspoon ipmid[854]: No Registered handlers for NetFn:[0x4],Cmd:[0xFF]
Feb 22 06:13:07 witherspoon ipmid[854]: ERROR:[0xC1] handling NetFn:[0x4], Cmd:[0x30]
Feb 22 06:13:07 witherspoon systemd[1]: Reached target Quiesce Target.
Feb 22 06:13:07 witherspoon ipmid[854]: No registered handlers for NetFn:[0x4], Cmd:[0x30] trying Wilcard implementation
Feb 22 06:13:07 witherspoon ipmid[854]: No Registered handlers for NetFn:[0x4],Cmd:[0xFF]
Feb 22 06:13:07 witherspoon ipmid[854]: ERROR:[0xC1] handling NetFn:[0x4], Cmd:[0x30]
Feb 22 06:13:07 witherspoon phosphor-host-state-manager[864]: Auto reboot enabled. Beginning reboot...
Feb 22 06:13:07 witherspoon phosphor-host-state-manager[864]: Host State transaction request
Feb 22 06:13:07 witherspoon ipmid[854]: No registered handlers for NetFn:[0x6], Cmd:[0x22] trying Wilcard implementation
Feb 22 06:13:07 witherspoon ipmid[854]: No Registered handlers for NetFn:[0x6],Cmd:[0xFF]
Feb 22 06:13:07 witherspoon ipmid[854]: ERROR:[0xC1] handling NetFn:[0x6], Cmd:[0x22]
Feb 22 06:13:07 witherspoon ipmid[854]: No registered handlers for NetFn:[0x6], Cmd:[0x22] trying Wilcard implementation
Feb 22 06:13:07 witherspoon ipmid[854]: No Registered handlers for NetFn:[0x6],Cmd:[0xFF]
Feb 22 06:13:07 witherspoon ipmid[854]: ERROR:[0xC1] handling NetFn:[0x6], Cmd:[0x22]
Feb 22 06:13:07 witherspoon systemd[1]: Stopped target Quiesce Target.
Feb 22 06:13:07 witherspoon systemd[1]: Stopped target Chassis0 (Start).
Feb 22 06:13:07 witherspoon systemd[1]: Stopped Start Watchdog0.
Feb 22 06:13:07 witherspoon systemd[1]: Created slice system-op\x2dwait\x2dpower\x2doff.slice.
Feb 22 06:13:07 witherspoon systemd[1]: Starting Wait for Power0 to turn off...
...
```

Ignoring the typos it seems `ipmid` has some serious issues.

This is from the tip of openbmc/openbmc as of 20/03/2017:

```
# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-110"
VERSION_ID="v1.99.3-110-g4a03f88"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-110"
BUILD_ID="v1.99.3"
```

amboar, 2017-03-20:
I just noticed this in the journal as I was scrolling through for another issue:

```
Feb 22 06:05:24 witherspoon netipmid[779]: Registering handler: /usr/lib/net-ipmid/libapphandler.so.0
Feb 22 06:05:24 witherspoon netipmid[779]: Error opening /usr/lib/net-ipmid/libapphandler.so.0/usr/lib/net-ipmid/libapphandler.so.0: undefined symbol: _ZN9sdbusplus3xyz15openbmc_project5State6server17convertForMessageB5cxx11ENS3_4Host10TransitionE
Feb 22 06:05:25 witherspoon led_controller.exe[791]: Error starting LED Services. Exiting
```

geissonator, 2017-03-20:
Kind of looks like we're not correctly processing the watchdog reset commands coming down from the host @ratagupt?  Or did hostboot hang and the watchdog hit was legit?

amboar, 2017-03-20:
@geissonator I don't believe hostboot hung - from comparing the journal and hostboot output it was progressing quite normally until the BMC cut it down.

The reply I posted above suggests we're mangling the path of the shared object that we should load to process the IPMI commands.

ratagupt, 2017-03-20:
seems ipmi unable to load the library libapphandler.so due to undefined symbol.
Feb 22 06:06:31 witherspoon ipmid[854]: ERROR opening [/usr/lib/host-ipmid/libapphandler.so.0]: /usr/lib/host-ipmid/libapphandler.so.0: undefined symbol: _ZN9sdbusplus3xyz15openbmc_project5State6server17convertForMessageB5cxx11ENS3_4Host10TransitionE

geissonator, 2017-03-21:
Resolved here - https://gerrit.openbmc-project.xyz/#/c/3139/

closed 2017-03-21 by geissonator