openbmc/openbmc #1337: op-reset-pgood-check@0.service: Main process exited, code=exited, status=1/FAILURE

amboar, 2017-03-20:
On the tip of master as of 20/03/2017:

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-110"
VERSION_ID="v1.99.3-110-g4a03f88"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-110"
BUILD_ID="v1.99.3"
```

I found the following [in the journal](https://github.com/openbmc/openbmc/files/853720/journal.txt)

```
Feb 22 06:06:41 witherspoon systemd[1]: Started Restore Last State.
Feb 22 06:06:42 witherspoon systemd[1]: op-reset-pgood-check@0.service: Main process exited, code=exited, status=1/FAILURE
Feb 22 06:06:42 witherspoon systemd[1]: Failed to start Check Chassis0 pgood on reset.
Feb 22 06:06:42 witherspoon systemd[1]: Dependency failed for Chassis0 (Reset Check).
Feb 22 06:06:42 witherspoon systemd[1]: obmc-chassis-reset@0.target: Job obmc-chassis-reset@0.target/start failed with result 'dependency'.
Feb 22 06:06:42 witherspoon systemd[1]: Dependency failed for Create file to indicate chassis0 pgood is on.
Feb 22 06:06:42 witherspoon systemd[1]: Dependency failed for Start chassis0 on after BMC reset.
Feb 22 06:06:42 witherspoon systemd[1]: op-reset-chassis-on@0.service: Job op-reset-chassis-on@0.service/start failed with result 'dependency'.
Feb 22 06:06:42 witherspoon systemd[1]: op-reset-set-power-on@0.service: Job op-reset-set-power-on@0.service/start failed with result 'dependency'.
Feb 22 06:06:42 witherspoon systemd[1]: op-reset-pgood-check@0.service: Unit entered failed state.
Feb 22 06:06:42 witherspoon systemd[1]: op-reset-pgood-check@0.service: Failed with result 'exit-code'.
Feb 22 06:06:42 witherspoon systemd[1]: Reached target Chassis0 power on after reset.
```

amboar, 2017-03-20:
Note: This is the same boot/journal as #1336

geissonator, 2017-03-20:
Yeah, I'm getting this one a lot.  This is actually working as designed.  In order to have the entire bmc reset design be controlled via targets, I need to fail a the target that checks whether pgood is on after the bmc reset (if pgood is not on).  If pgood is on then this targets goes on to run the services that create the appropriate pgood file and get the chassis start target into the running state (without actually executing any of it's services).  The target failing here has no affect on the overall system.

amboar, 2017-03-20:
Can we rethink that approach? I think we're going to get this report a lot if that's the case. Can't we drive this like we drive other system state transitions? That is, have a service start as part of the initial target, which probes pgood and requests a state transition to one target or another as part of its execution, then exit successfully. That way we shouldn't have "expected" dependency failures in the logs.

geissonator, 2017-03-20:
Yeah, no one seems to like my design here :)  Let me reopen and see if I can come up with something better, since my plan was to add another one of these to support the host (so we'd have 2 targets failing with that).

rfrandse, 2017-05-18:
https://gerrit.openbmc-project.xyz/4258 Combined services to create chassis-running service
Resolves #1337 op-reset-pgood-check@0.service: Main process exited, code=exited, status=1/FAILURE

closed 2017-05-24 by williamspatrick