openbmc/openbmc #1341: Build is failing on Ubuntu 17.04 ppc64le OS

PavamanSubramaniyam, 2017-03-20:
I am trying to build the openbmc images for witherspoon platform on Ubuntu 17.04 OS installed on P8 Hardware.

ubuntu@ltc-garri2:~/openbmc$ cd build/
ubuntu@ltc-garri2:~/openbmc/build$ bitbake obmc-phosphor-image

| /bin/bash ./ppc64le-linux-libtool --tag=CC   --mode=compile gcc  -DHAVE_CONFIG_H -I. -I../../bfd -I. -I../../bfd -I../../bfd/../include  -DHAVE_all_vecs  -DBINDIR='"/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/bin"' -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -I../../bfd/../zlib -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -O2 -pipe -c -o i386linux.lo ../../bfd/i386linux.c
| ppc64le-linux-libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../bfd -I. -I../../bfd -I../../bfd/../include -DHAVE_all_vecs -DBINDIR=\"/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/bin\" -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -I../../bfd/../zlib -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -O2 -pipe -c ../../bfd/hp300bsd.c -o hp300bsd.o
| ppc64le-linux-libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../bfd -I. -I../../bfd -I../../bfd/../include -DHAVE_all_vecs -DBINDIR=\"/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/bin\" -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -I../../bfd/../zlib -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -O2 -pipe -c ../../bfd/hp300hpux.c -o hp300hpux.o
| cc1: all warnings being treated as errors
| /bin/bash ./ppc64le-linux-libtool --tag=CC   --mode=compile gcc  -DHAVE_CONFIG_H -I. -I../../bfd -I. -I../../bfd -I../../bfd/../include  -DHAVE_all_vecs  -DBINDIR='"/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/bin"' -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -I../../bfd/../zlib -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -O2 -pipe -c -o i386lynx.lo ../../bfd/i386lynx.c
| Makefile:1635: recipe for target 'elf32-arc.lo' failed
| make[4]: *** [elf32-arc.lo] Error 1
| make[4]: *** Waiting for unfinished jobs....
| ppc64le-linux-libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../bfd -I. -I../../bfd -I../../bfd/../include -DHAVE_all_vecs -DBINDIR=\"/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/bin\" -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -I../../bfd/../zlib -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -O2 -pipe -c ../../bfd/i386bsd.c -o i386bsd.o
| ppc64le-linux-libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../bfd -I. -I../../bfd -I../../bfd/../include -DHAVE_all_vecs -DBINDIR=\"/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/bin\" -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -I../../bfd/../zlib -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -O2 -pipe -c ../../bfd/i386aout.c -o i386aout.o
| ppc64le-linux-libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../bfd -I. -I../../bfd -I../../bfd/../include -DHAVE_all_vecs -DBINDIR=\"/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/bin\" -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -I../../bfd/../zlib -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -O2 -pipe -c ../../bfd/i386linux.c -o i386linux.o
| ppc64le-linux-libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../bfd -I. -I../../bfd -I../../bfd/../include -DHAVE_all_vecs -DBINDIR=\"/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/bin\" -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -I../../bfd/../zlib -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -O2 -pipe -c ../../bfd/i386dynix.c -o i386dynix.o
| ppc64le-linux-libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../bfd -I. -I../../bfd -I../../bfd/../include -DHAVE_all_vecs -DBINDIR=\"/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/bin\" -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -I../../bfd/../zlib -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -O2 -pipe -c ../../bfd/i386freebsd.c -o i386freebsd.o
| ppc64le-linux-libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../bfd -I. -I../../bfd -I../../bfd/../include -DHAVE_all_vecs -DBINDIR=\"/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/bin\" -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -I../../bfd/../zlib -isystem/home/ubuntu/openbmc/build/tmp/sysroots/ppc64le-linux/usr/include -O2 -pipe -c ../../bfd/i386lynx.c -o i386lynx.o
| ../../bfd/elf32-nds32.c: In function 'nds32_elf_relax_section':
| ../../bfd/elf32-nds32.c:12347:45: error: left shift of negative value [-Werror=shift-negative-value]
|        sec_size_align = (sec->size + (~((-1) << sec->alignment_power)))
|                                              ^~
| ../../bfd/elf32-nds32.c:12348:18: error: left shift of negative value [-Werror=shift-negative-value]
|           & ((-1) << sec->alignment_power);
|                   ^~
| cc1: all warnings being treated as errors
| Makefile:1635: recipe for target 'elf64-s390.lo' failed
| make[4]: *** [elf64-s390.lo] Error 1
| cc1: all warnings being treated as errors
| Makefile:1635: recipe for target 'elf32-nds32.lo' failed
| make[4]: *** [elf32-nds32.lo] Error 1
| make[4]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/binutils-native/2.26-r0/git/build.ppc64le-linux.ppc64le-linux/bfd'
| Makefile:1673: recipe for target 'all-recursive' failed
| make[3]: *** [all-recursive] Error 1
| make[3]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/binutils-native/2.26-r0/git/build.ppc64le-linux.ppc64le-linux/bfd'
| Makefile:1152: recipe for target 'all' failed
| make[2]: *** [all] Error 2
| make[2]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/binutils-native/2.26-r0/git/build.ppc64le-linux.ppc64le-linux/bfd'
| Makefile:2711: recipe for target 'all-bfd' failed
| make[1]: *** [all-bfd] Error 2
| make[1]: Leaving directory '/home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/binutils-native/2.26-r0/git/build.ppc64le-linux.ppc64le-linux'
| Makefile:845: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: oe_runmake failed
| ERROR: Function failed: do_compile (log file is located at /home/ubuntu/openbmc/build/tmp/work/ppc64le-linux/binutils-native/2.26-r0/temp/log.do_compile.41526)
ERROR: Task 958 (virtual:native:/home/ubuntu/openbmc/meta/recipes-devtools/binutils/binutils_2.26.bb, do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1480 tasks of which 1435 didn't need to be rerun and 1 failed.
Waiting for 0 running tasks to finish:

Summary: 1 task failed:
  virtual:native:/home/ubuntu/openbmc/meta/recipes-devtools/binutils/binutils_2.26.bb, do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

PavamanSubramaniyam, 2017-03-20:
I have added the below two lines into the file import-layers/yocto-poky/meta/recipes-devtools/binutils/binutils_2.26.bb as a workaround.

CFLAGS_class-native += " -Wno-error=shift-negative-value "
CFLAGS_class-nativesdk += " -Wno-error=shift-negative-value "

Once we add the above mentioned two lines, the issue gets resolved.
Please let me know if I can send a PULL request with the above fixes as a patch.

williamspatrick, 2017-04-19:
I believe this is fixed upstream now via Yocto 2.2.  Please feel free to reopen if it still is an issue.

closed 2017-04-19 by williamspatrick