openbmc/openbmc #1342: Watchdog support

bradbishop, 2017-03-20:
One of the watchdog IP blocks on the AST2500 is wired to a 'max fans' input on the fan control chip on the Witherspoon system.  As the various pieces of the overall fan / thermal support are starting or restarted the state of this input must be maintained appropriately.

Much more definition is required here; suggestions welcome.

bradbishop, 2017-03-20:
@spinler @msbarth

geissonator, 2017-05-12:
queue up for design discussion

spinler, 2017-05-17:
The direction I'm going is to try to use the watchdog application (https://linux.die.net/man/8/watchdog, import-layers/yocto-poky/meta/recipes-extended/watchdog) to ping the watchdog hardware, and see if it can be configured to only ping when a set of processes are all running (fan control, monitor, hwmon, presence-detect, maybe others).

We'd set the timeout to ~5s, and ping it maybe every 1s.

Additionally, there is a BMC GPIO called WATCHDOG_CLEAR_PULSE_N that needs to be toggled on each power on.  Still investigating the details behind that.

Finally, the fan controller chip needs a bit set in each page to tell it to actually monitor the fault pin and sets the PWMs to 100 when it is activated.  This will need to go into the device driver somehow.

spinler, 2017-05-17:
Some other open issues:

* The fan monitor also checks for fans running too fast.  So if the watchdog trips, but the Target property in hwmon isn't updated, it will flag it as a bad fan
* Fan control needs to know the watchdog tripped, so it doesn't keep trying to set its own fan speeds

These could be solved by having a way to know the watchdog actually tripped.  It wouldn't be able to just read the MAX chip via I2C, as I2C could be broken.  If we could read a BMC WDT1 register directly that might work, though then we have to see if that CLEAR_PULSE gpio could cause WDT1 not actually reflect the FAULT pin value on the MAX chip.

spinler, 2017-05-17:
CLEAR_PULSE Function:

If the watchdog has tripped, and then is pinged again, the fans will still stay at high until this GPIO goes 1 -> 0 -> 1. (the rising edge is the trigger)

spinler, 2017-05-18:
10,000 foot design:

- modify watchdog driver to enable setting external gpio on timeout instead of reset (another story)
- modify fan control drive to obey watchdog signal (another story)
- use wd_keepalive from watchdog package to ping watchdog, start it on a power on
- if phosphor-hwmon (any instance, for now), fan-monitor, fan-presence, or fan-control crash continuously and exceed the threshold, start a 'watchdog triggered' target
- this target kills wd_keepalive and fan-monitor
- the watchdog is then tripped permanently until a power off or BMC reset (too many corner cases trying to recover it, adds lots of work)
- also on a power on, toggle the CLEAR_PULSE GPIO to enable watchdog

Task list coming soon

spinler, 2017-05-18:
- [ ] pull `import-layers/yocto-poky/meta/recipes-extended/watchdog_5.15.bb` into build (all systems)  only install wd_keepalive if possible
- [ ] in witherspoon layer, create a watchdog0.conf, install into /etc/default/obmc/watchdog/ (it tells wd_keepalive to use /dev/watchdog0)
- [ ] Create a `wd_keepalive@.service`, install it in `obmc-chassis-poweron@.target.requires` (only starts wd_keepalive if the conf file is there, and takes the conf as an argument) Starts before fan-control --init.
- [ ] Create a 'phosphor-watchdog-reset' exe under phosphor-fan that will bring the CLEAR_PULSE GPIO low then high again (a few us delay in between).  It takes the GPIO as an arg.  Use /dev/gpiochip?
- [ ] Update phosphor-fan.bb to get phosphor-watchdog-reset installed.
- [ ] Create `phosphor-watchdog-reset@.service` to run it.  install in power on target to run right before wd_keepalive starts
- [ ] Create new obmc-stop-fan-watchdog.target and install it
- [ ] When stop-watchdog target is started, it will stop wd_keepalive and fan-monitor (using conflicts= in their respective service files)
- [ ] Update service files of watchdog-interested apps to start the stop-watchdog target after fails more than 5 times in 1 30s (final numbers can be negotiated)

mdmillerii, 2017-05-18:
Why are you trying to pull a dedicated watchdog utility ? 

1.  busybox has a watchdog
2. systemd has a watchdog built into it.  Systemd can be configured for units to do behaviour.
3. the kernel can create multiple devices, but normally only the first is used
4. the hardware has 2 (2400) or 3 (2500) that can be configured to reset the chip, switch the byte length of the address to the spi flash auto-read, or drive an interrupt or specific gpio on expiration.

It would seem your "drive gpio instead of reset" would be better described as "configure secondary watchdog to drive gpio" and use the ioctl there, let systemd run the primary watchdog?  Not sure.

spinler, 2017-05-19:
@mdmillerii I'm not really sure what you are asking.  We want a watchdog only over the programs that are involved in fan control.   systemd can still monitor a watchdog for its own health (though it isn't configured to do so now, and it can't use WDT1 as that must be for fans) and that has nothing to do with this.

bradbishop, 2017-05-19:
@mdmillerii 

The high level need is we need a watchdog that doesn't reset the chip and does pulse the net that goes to the max chip.

1 - I didn't know that, thanks.  We will take a look.

2 - We looked at this.  systemd does not support pinging multiple watchdog devices.  I assumed we'd want to reserve the one it can ping for the usual watchdog use case (reset the chip if the kernel or systemd is hung).

3 - I think we need two, one for the fans and one to reset the chip.  That seem reasonable?

4 - Pretty sure I agree with you here.  Is there something we've posted that led you to make this assertion?

I didn't find an IOCTL for configuring this...in either the driver or the watchdog api itself.  Is there one?

spinler, 2017-05-22:
Updated task list.

mdmillerii, 2017-05-24:
@bradbishop, @spinler, and myself had a discussion Friday and covered the following topics:

1. The watchdog command is `busybox` is useful for this purpose
2. There are multiple watchdog monitor pins, but each pin is assigned a specific watchdog so they are not interchangeable.
3.  The kernel needs a watchdog to reset the system for reboot and for panic. 
4.  When the chip is fully reset watchdog 0 is restart boot, 1 is swap flash CS1 to CS0 if hardware strap enabled, and 2 is change address size for read command (2500, strap enabled).  
5. Ideally, dual flash / alternate flash will use the alternate chip select when the primary boot fails.  This would imply that the watchdog be left running in u-boot.  There was a recent mailing list post to configure u-boot that adjusted this timer.
6. Ideally the normal watchdog expiration would apply to raise the fans or not for a controlled reboot
7.  The fans must be configured when 

I raise the following concerns:
1. The both u-boot and the kernel "know" this default watchdog assignment.
2. `systemd` will open the primary watchdog.  The kernel will need to reboot with the "reset system" watchdog.   
3.  The use of the watchdog external pin is a property of wiring and I would argue device-tree.
4.  The need for an alternate configuration for system watchdog would argue that 3 drives the reset to be device-tree driven, but this is weaker.
5.  u-boot likely  needs to know these alternate configurations, at least when it fails to load and wants to do a reset.

I proposed an alternate implementation of these requirements.  Currently this alternative is not being pursued:

Considering these requirements:
1. The fan needs to ramp / watchdog output asserted if the system panics
2.  The fans are wired to watchdog 0 assertion, which is the current panic / default system watchdog
3.  The pin can be controlled as a gpio
4.  The fan control and monitoring is distributed among several processes and software will trigger the fault ramp by killing a daemon

An alternate proposal would be:
1.  By default the system reset watchdog is left as watchdog 0
2.  An additional pin control setting is added for two states:  The default state configures the output as watchdog,  An additional named state configures the output as asserted (driven low) via gpio. 
3.  A `sysfs` file, likely on the watchdog driver, would trigger the driver to request the alternate output state.  The device-tree exposing the alternate pin configuration would trigger the creation of the attribute.  The driver would enumerate the configurations and would assert the alternate when requested via the `pinconf` system.
4.  Instead of killing the watchdog ping daemon, the code would write to the above mentioned attribute to ramp the fans on software detected error.

@amboar this may provide background for your patch review.

spinler, 2017-05-26:
Updated the task list with the addition of a new phosphor-unit-failure-monitor application that will be run via the OnFailure directive in unit files.  It will make a dbus call to query the ActiveState property of the unit to see if it says 'failed', which means it exceeded it's allowed restarts and will stay dead.  It will be able to start or stop an arbitrary unit (the stop-fan-watchdog target in our case) based on command line parameters.

We need this to work around the fact that the unit specified in OnFailure is called every single restart, even if the app will restart again OK, and we don't want to trip the watchdog for that.

I don't know what code repository this should go in, will need to discuss in a design call.

spinler, 2017-05-30:
Going to descope this issue to only use it to install and run the busybox watchdog program. Other issues will track the other work.

spinler, 2017-05-30:
- [ ] Enable the busybox watchdog in busybox.cfg
- [ ] Create a service to start it at power on
- [ ] use an environment file for the command line arguments of the device to use, the timeout value, and ping intervals
- [ ] Create and install a obmc-stop-fan-watchdog.target that will conflict with this service as well as the fan-monitor service

bradbishop, 2017-06-27:
Removed from 6.14 committed.  Not sure why it was added.

spinler, 2017-07-19:
This is done.  I must have forgotten a resolves tag.

closed 2017-07-19 by spinler