openbmc/openbmc #1343: Sensor groups

bradbishop, 2017-03-20:
Develop a general purpose application that does runtime monitoring of build time defined groups of sensors for a condition.  Asserts an action when the condition evaluates.

- [ ]  Define YAML format
- [ ]  Write application build skeleton (hwmon repository?)
- [ ]  Write parser with mako and generate runtime code / data model.
- [ ]  Implement 'if any in group > hardcoded number' condition.
- [ ]  Implement 'create error log' action.

bradbishop, 2017-03-20:
@spinler @msbarth

msbarth, 2017-03-23:
- [ ] Define YAML with group of sensors, sensor details, filters/triggers(algorithm), and actions
- [ ] Write application build skeleton (hwmon repository?)
- [ ] Write parser with mako and generate runtime code / data model.

Incremental testing will take into account the following specifics regarding thermal condition & action:
- Implement 'if any in group > hardcoded number' condition.
- Implement 'create error log' action.

geissonator, 2017-03-23:
Brad/Matt to break down a bit more, need to be 3 or less points of work

geissonator, 2017-04-20:
Reducing from 3->2 points.  Matt will create new story (1pt) to track the parser work.

rfrandse, 2017-05-10:
https://gerrit.openbmc-project.xyz/#/c/4116/

closed 2017-05-19 by williamspatrick