openbmc/openbmc #1346: tach feedback functional state monitoring

bradbishop, 2017-03-20:
Develop an application that monitors Sensor.Value and FanSpeed.Target to update functional state based on fan responsiveness.

bradbishop, 2017-03-20:
@spinler @msbarth

spinler, 2017-04-13:
Requirements:
- create a new application phosphor-fan-monitor
- on app startup, read and store all fan sensor Target values on a per fan enclosure basis
- then register for Target properties changed signals, and store each new value
- register for fan sensor properties changed signals, and on a signal:
  - compare the value to the latest Target value
  - if not within some deviation percentage (configurable from yaml), start a timer (value from yaml)
  - if it is within the deviation, cancel the timer if one is running
  - also if within deviation, need to set inventory back to functional if it previously wasn't
- if timer expires:  set the 'not functional' inventory property for the fan
- on startup, will need a longer timeout since fans have a bigger delta
- if app is restarted at runtime, how to know a bigger timeout isn't needed?

spinler, 2017-04-20:
- [ ] Create a Timer utility class that calls an arbitrary function on timeout
- [ ] Write code to keep track of tach input and target values based on properties changed signals
- [ ] Write code to start timers when the input values are below 85% of the Target
- [ ] Create an Operation Status property/interface definition on the fan inventory if one doesn't already exist
- [ ] If the timer expires, set the non-functional/non-operational state on the fan in the inventory

rfrandse, 2017-05-10:
https://gerrit.openbmc-project.xyz/#/c/4014/

closed 2017-05-12 by williamspatrick